mattanger/ckb-next #2: CKB - New Device Support

mattanger, 2016-11-20:
If a Corsair device isn't supported, leave a comment with the device details, and it may get added eventually.

dan-santana, 2016-12-06:
Corsair MM800 RGB Polaris... which device details are needed?

mattanger, 2016-12-07:
Device IDs are a good place to start. Relevant output from lsub, any relevant kernel messages. That kind of thing, and a willingness to test hardware that I don't have!

fleischie, 2016-12-07:
I suggest we write a (platform-independent) script, that outputs the necessary information, that we need to update the device-driver (as long as it has the current structure).

This should enable users without technical background to help us find the correct information.

tatokis, 2016-12-11:
@fleischie how about https://github.com/mattanger/ckb/blob/master/ckb-dev-detect ?

dan-santana, 2016-12-11:
I'm available to test the hardware, but probably only on weekends. Here's my output of `ckb-dev-detect` (uncompressed):
```
ckb-dev-detect - Sun Dec 11 12:44:27 BRST 2016

Kernel: Linux green 4.8.12-3-ARCH #1 SMP PREEMPT Thu Dec 8 16:10:23 CET 2016 x86_64 GNU/Linux

OS Version:
LSB Version:	1.4
Distributor ID:	Arch
Description:	Arch Linux
Release:	rolling
Codename:	n/a

USB Devices:
Bus 001 Device 008: ID 1b1c:1b33 Corsair 
Bus 001 Device 006: ID 1b1c:1b2e Corsair 
Bus 001 Device 007: ID 1b1c:1b3b Corsair 
Bus 001 Device 004: ID 1b1c:1b2c Corsair 

Dmesg:
[    8.520992] input: Corsair Corsair Gaming M65 Pro RGB Mouse as /devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2.2/1-2.2:1.0/0003:1B1C:1B2E.0001/input/input13
[    8.521087] hid-generic 0003:1B1C:1B2E.0001: input,hidraw0: USB HID v1.11 Mouse [Corsair Corsair Gaming M65 Pro RGB Mouse] on usb-0000:00:14.0-2.2/input0
[   11.121384] input: ckb2: Corsair Gaming K70 LUX RGB Keyboar as /devices/virtual/input/input21
[   11.174676] input: ckb2: Corsair Gaming K70 LUX RGB Keyboar as /devices/virtual/input/input22
[   18.625340] hid-generic 0003:1B1C:1B2E.0002: usb_submit_urb(ctrl) failed: -1
[   18.625362] hid-generic 0003:1B1C:1B2E.0002: timeout initializing reports
[   18.625598] input: Corsair Corsair Gaming M65 Pro RGB Mouse as /devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2.2/1-2.2:1.1/0003:1B1C:1B2E.0002/input/input23
[   18.681122] hid-generic 0003:1B1C:1B2E.0002: input,hiddev0,hidraw0: USB HID v1.11 Mouse [Corsair Corsair Gaming M65 Pro RGB Mouse] on usb-0000:00:14.0-2.2/input1
[   18.681960] hid-generic 0003:1B1C:1B3B.0003: hiddev0,hidraw2: USB HID v1.11 Device [Corsair Components, Inc. Corsair Gaming MM800 RGB POLARIS] on usb-0000:00:14.0-2.1.1/input0
[   18.682209] input: Corsair Components, Inc. Corsair Gaming MM800 RGB POLARIS as /devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2.1/1-2.1.1/1-2.1.1:1.1/0003:1B1C:1B3B.0004/input/input24
[   18.737298] hid-generic 0003:1B1C:1B3B.0004: input,hidraw3: USB HID v1.11 Keyboard [Corsair Components, Inc. Corsair Gaming MM800 RGB POLARIS] on usb-0000:00:14.0-2.1.1/input1
[   18.838016] input: ckb1: Corsair Gaming M65 Pro RGB Mouse as /devices/virtual/input/input25
[   18.839732] input: ckb1: Corsair Gaming M65 Pro RGB Mouse as /devices/virtual/input/input26
```

tatokis, 2016-12-11:
I've modified the script slightly to give more detailed info under USB devices on Linux.

@dan-santana Yours is an interesting case, as it isn't really a keyboard, nor a mouse.

From what I can see so far, it is a mousepad with LEDs on it.
What you could do for now, is try adding its dev ID (1B3B) in src/ckb-daemon/usb.h , under different device classes, and see if you can make it do anything at all.

If not, we'll have to figure out how CUE talks to it.

Also, quick suggestion after looking at your dmesg,
```
[   18.625362] hid-generic 0003:1B1C:1B2E.0002: timeout initializing reports
```
you should probably add the appropriate USB quirk for your mouse.
```usbhid.quirks=0x1B1C:1B2E:0x20000408``` at the end of your kernel cmdline
and if you also want it to take effect without rebooting,
```sudo modprobe -r usbhid && sudo modprobe usbhid quirks=0x1B1C:1B2E:0x20000408```
It'll reduce your boot time and the mouse will be recognised instantly if you unplug it and plug it back in.

dan-santana, 2016-12-19:
Indeed, adding the quirks for  the mouse (and keyboard too) improved the boot time a lot!  Thank you :) 

Yeah, it's a RGB mousepad, but with a USB passthrough port too (like K70 Lux). I tried the dev id in all device classes, `ckb` always showed the right name of the device on the tab, but the settings had no effect on the device.

The new script gave me some errors (running with `sudo`):
```
This script will collect information about the Corsair devices in your system.
Please make sure they are plugged in and press Enter.

can't get device qualifier: Resource temporarily unavailable
can't get debug descriptor: Resource temporarily unavailable
can't get debug descriptor: Resource temporarily unavailable
can't get device qualifier: Resource temporarily unavailable
can't get debug descriptor: Resource temporarily unavailable
can't get device qualifier: Resource temporarily unavailable
can't get debug descriptor: Resource temporarily unavailable
Done! Please send the following report to the developers.
Location: /home/dan/ckb-dev-detect-report.gz
```

Here's the output (uncompressed):
```
ckb-dev-detect - Sun Dec 18 22:55:33 BRST 2016

Kernel: Linux green 4.8.13-1-ARCH #1 SMP PREEMPT Fri Dec 9 07:24:34 CET 2016 x86_64 GNU/Linux

OS Version:
LSB Version:	1.4
Distributor ID:	Arch
Description:	Arch Linux
Release:	rolling
Codename:	n/a

USB Devices:

Bus 001 Device 009: ID 1b1c:1b3b Corsair 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1b1c Corsair
  idProduct          0x1b3b 
  bcdDevice            1.02
  iManufacturer           1 Corsair Components, Inc.
  iProduct                2 Corsair Gaming MM800 RGB POLARIS
  iSerial                 3 1B4002146904B8AE3C05A355401900F5
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           66
    bNumInterfaces          2
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
[... 394 lines cut ...]
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
Device Status:     0x0000
  (Bus Powered)

Dmesg:
[    0.000000] Command line: BOOT_IMAGE=/vmlinuz-linux root=UUID=189a3bd6-8024-4778-97d0-32b00c31b295 rw quiet usbhid.quirks=0x1B1C:0x1B33:0x20000408,0x1B1C:0x1B2E:0x20000408
[    0.000000] Kernel command line: BOOT_IMAGE=/vmlinuz-linux root=UUID=189a3bd6-8024-4778-97d0-32b00c31b295 rw quiet usbhid.quirks=0x1B1C:0x1B33:0x20000408,0x1B1C:0x1B2E:0x20000408
[    2.486204] input: Corsair Corsair Gaming M65 Pro RGB Mouse as /devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2.1/1-2.1:1.0/0003:1B1C:1B2E.0001/input/input2
[    2.486244] hid-generic 0003:1B1C:1B2E.0001: input,hidraw0: USB HID v1.11 Mouse [Corsair Corsair Gaming M65 Pro RGB Mouse] on usb-0000:00:14.0-2.1/input0
[    2.486362] input: Corsair Corsair Gaming M65 Pro RGB Mouse as /devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2.1/1-2.1:1.1/0003:1B1C:1B2E.0002/input/input3
[    2.486481] hid-generic 0003:1B1C:1B2E.0002: input,hiddev0,hidraw1: USB HID v1.11 Mouse [Corsair Corsair Gaming M65 Pro RGB Mouse] on usb-0000:00:14.0-2.1/input1
[    2.983296] input: Corsair Corsair Gaming K70 LUX RGB Keyboard  as /devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2.2/1-2.2:1.0/0003:1B1C:1B33.0003/input/input4
[    2.983580] hid-generic 0003:1B1C:1B33.0003: input,hidraw2: USB HID v1.11 Keyboard [Corsair Corsair Gaming K70 LUX RGB Keyboard ] on usb-0000:00:14.0-2.2/input0
[    2.985812] input: Corsair Corsair Gaming K70 LUX RGB Keyboard  as /devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2.2/1-2.2:1.1/0003:1B1C:1B33.0004/input/input5
[    2.986151] hid-generic 0003:1B1C:1B33.0004: input,hiddev0,hidraw3: USB HID v1.11 Keyboard [Corsair Corsair Gaming K70 LUX RGB Keyboard ] on usb-0000:00:14.0-2.2/input1
[    3.943609] hid-generic 0003:1B1C:1B3B.0005: hiddev0,hidraw4: USB HID v1.11 Device [Corsair Components, Inc. Corsair Gaming MM800 RGB POLARIS] on usb-0000:00:14.0-2.4.1.1/input0
[    3.944945] input: Corsair Components, Inc. Corsair Gaming MM800 RGB POLARIS as /devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2.4/1-2.4.1/1-2.4.1.1/1-2.4.1.1:1.1/0003:1B1C:1B3B.0006/input/input6
[    3.998994] hid-generic 0003:1B1C:1B3B.0006: input,hidraw5: USB HID v1.11 Keyboard [Corsair Components, Inc. Corsair Gaming MM800 RGB POLARIS] on usb-0000:00:14.0-2.4.1.1/input1
[    8.196143] input: ckb1: Corsair Gaming M65 Pro RGB Mouse as /devices/virtual/input/input25
[    8.255829] input: ckb2: Corsair Gaming K70 LUX RGB Keyboar as /devices/virtual/input/input26
[    8.255935] input: ckb1: Corsair Gaming M65 Pro RGB Mouse as /devices/virtual/input/input27
[    8.256265] input: ckb2: Corsair Gaming K70 LUX RGB Keyboar as /devices/virtual/input/input28
```

mmolad, 2017-01-06:
K65 Non-RGB

tatokis, 2017-01-06:
@mmolad Please download ckb-next from https://github.com/mattanger/ckb-next/archive/master.zip and run "ckb-dev-detect"

mmolad, 2017-01-06:
@tatokis updated comment with log.

cmawhinney, 2017-01-13:
Vengeance® K65 Compact Mechanical Gaming Keyboard

[image]

mmolad, 2017-01-16:
@tatokis as a short term solution, couldn't the K65 Non-RGB just be treated as a K70 Non-RGB?  Would this be a quick fix?

tatokis, 2017-01-16:
Apologies, got distracted by other issues. 

I can add it later when I get home as its own device, but keep in mind there might be issues with Sierra.

mmolad, 2017-01-16:
No problem - thanks for taking a look!  I just switched over to macOS
professionally so finding out my favorite keyboard is incompatible is a
real bummer.

dtrang, 2017-01-17:
Is there a plan to support K70 RGB Rapidfire?

tatokis, 2017-01-17:
@mmolad @cmawhinney on the testing branch, https://github.com/mattanger/ckb-next/commit/86118b35092474062508ca895075176c6e12d387

@dtrang it's already supported.

mmolad, 2017-01-17:
> @mmolad <https://github.com/mmolad> @cmawhinney
> <https://github.com/cmawhinney> on the testing branch, 86118b3
>
> @dtrang <https://github.com/dtrang> it's already supported.

Whats the best way to update to the testing branch?

tatokis, 2017-01-17:
@mmolad 
https://github.com/mattanger/ckb-next/archive/testing.zip
https://github.com/mattanger/ckb-next#building-from-source

mmolad, 2017-01-17:
I updated via rebuilding from source and still get "no devices connected"
after rebooting. The keyboard still only functions in BIOS mode, with the
other settings giving partial support (most keys don't work, like
backspace).

Shall we move this to a different issue?

tatokis, 2017-01-17:
Interesting, it should at least recognise it. 

Please open a new issue in order to better keep track of that and I'll give you something to test in a couple of hours.

guenniman, 2017-01-19:
Hello, I would like to have support for the VOID USB Headset. Feel free to ask me for testing.

[ckb-dev-detect-report.txt](https://github.com/mattanger/ckb-next/files/717714/ckb-dev-detect-report.txt)

Camoleopard, 2017-01-27:
Corsair Harpoon Gaming Mouse: (Uncompressed)

ckb-dev-detect - Fri Jan 27 12:17:42 AWST 2017

Kernel: Darwin 181095-SCT11938.local 15.5.0 Darwin Kernel Version 15.5.0: Tue Apr 19 18:36:36 PDT 2016; root:xnu-3248.50.21~8/RELEASE_X86_64 x86_64

OS Version:
ProductName:	Mac OS X
ProductVersion:	10.11.5
BuildVersion:	15F34

USB Devices:
        Corsair Gaming HARPOON RGB Mouse:

          Product ID: 0x1b3c
          Vendor ID: 0x1b1c  (CORSAIR MEMORY INC.)
          Version: 0.13
          Serial Number: 19024017AEC78440568182BBF5001940
          Speed: Up to 12 Mb/sec
          Manufacturer: Corsair
          Location ID: 0x14200000 / 17
          Current Available (mA): 1000
          Current Required (mA): 300
          Extra Operating Current (mA): 0

tatokis, 2017-01-30:
@Camoleopard Could you please try https://github.com/mattanger/ckb-next/commit/84e195caba1ec3fa0f523106040a880c6c8c03f9 (testing branch) ?

Camoleopard, 2017-01-30:
@tatokis Is it possible I can get a compiled version? I don't have QT and it's too big to install.

Camoleopard, 2017-01-31:
@tatokis I tested this branch and it did not detect the device at all.

tatokis, 2017-01-31:
@Camoleopard please open a new issue and I'll see what can be done.

dodedodo33, 2017-01-31:
Hi all, last week I purchased the K95 Platinum with brown switches. It works a treat on windows, very happy with the keyboard. However it comes as no real surprise that it doesn't get detected by ckb-next. I'm interested in looking into it in some spare time. 

The lsusb -v output:
```
Bus 003 Device 014: ID 1b1c:1b2d Corsair 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1b1c Corsair
  idProduct          0x1b2d 
  bcdDevice            1.07
  iManufacturer           1 
  iProduct                2 
  iSerial                 3 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           84
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      67
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               8
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     172
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      25
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x03  EP 3 OUT
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
``` 
The ckb-dev-detect script is not much of a help for this keyboard, as it won't detect the keyboard both as regular or super user. I'm happy to help with the development for this new device, really like the work that has been done so far!

tatokis, 2017-01-31:
@dodedodo33 I don't see why the dev detect script wouldn't detect it. It just runs lsusb and filters out only devices with the corsair vendor id.
https://github.com/mattanger/ckb-next/blob/master/ckb-dev-detect#L33

I've added your device ID in the testing branch, please test if it works, and if it doesn't open a new issue with more details.
https://github.com/mattanger/ckb-next/tree/testing

dodedodo33, 2017-02-01:
Thanks for the quick reply, I probably have something misconfigured with permissions or groups or something. I have tried the testing branch, results:

The keyboard is detected and connected to by ckb-daemon. 

The issues so far:

- The keyboard layout in ckb is incorrect, as it is the regular k95 layout where the k95 Platinum has a slightly different key-set (6 G-keys and the top rgb bar).
- More drastically: when starting the ckb client application the keyboard stops working. The animation that was already playing on the keyboard (HW profile) stays on but no key presses are registered until I quit the client and re-plug the keyboard.

radean0909, 2017-06-07:
Hello! Just got the corsair glaive rgb mouse. The following is the report from the ckb-dev-detect:

`ckb-dev-detect - Wed Jun  7 09:47:05 PDT 2017

Kernel: Linux lappy-tappy 4.10.0-21-generic #23-Ubuntu SMP Fri Apr 28 16:14:22 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux

OS Version:
Distributor ID:	Ubuntu
Description:	Ubuntu 17.04
Release:	17.04
Codename:	zesty

USB Devices:
Couldn't open device, some information will be missing

Bus 001 Device 004: ID 1b1c:1b34 Corsair 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1b1c Corsair
  idProduct          0x1b34 
  bcdDevice            1.03
  iManufacturer           1 
  iProduct                2 
  iSerial                 3 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           84
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              300mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      58
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               8
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 No Subclass
      bInterfaceProtocol      0 None
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     172
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 No Subclass
      bInterfaceProtocol      0 None
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      25
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x03  EP 3 OUT
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1

Dmesg:
[18254.088871] usb 1-1: New USB device found, idVendor=1b1c, idProduct=1b34
[18254.088882] usb 1-1: Product: Corsair Gaming GLAIVE RGB Mouse
[18254.088886] usb 1-1: Manufacturer: Corsair
[18254.133783] input: Corsair Corsair Gaming GLAIVE RGB Mouse as /devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.0/0003:1B1C:1B34.0004/input/input18
[18254.190929] hid-generic 0003:1B1C:1B34.0004: input,hidraw2: USB HID v1.11 Mouse [Corsair Corsair Gaming GLAIVE RGB Mouse] on usb-0000:00:14.0-1/input0
[18254.191185] hid-generic 0003:1B1C:1B34.0005: item 0 1 0 8 parsing failed
[18254.191205] hid-generic: probe of 0003:1B1C:1B34.0005 failed with error -22`

jasonlin07, 2017-06-11:
Is there a plan to support Glauber RGB?

reefland, 2017-07-13:
Corsair MM800 RGB Polaris (RGB Game Pad)..  Arch Linux.  The default demo rainbow mode on the Polaris is driving me crazy.  Be glad to help test support for this.

BTW - I also have a CORSAIR - Strafe RGB MX Silent Gaming Keyboard, ckb-next worked nearly flawless, updated firmware, programmed key colors, saved to hardware.   Wasn't able to set color of the upper right lock key, everything else perfect.

```
ckb-dev-detect - Thu Jul 13 00:03:11 EDT 2017

Kernel: Linux dldsk01 4.11.9-1-ARCH #1 SMP PREEMPT Wed Jul 5 18:23:08 CEST 2017 x86_64 GNU/Linux

OS Version:

USB Devices:
can't get device qualifier: Resource temporarily unavailable
can't get debug descriptor: Resource temporarily unavailable

Bus 001 Device 006: ID 1b1c:1b3b Corsair 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1b1c Corsair
  idProduct          0x1b3b 
  bcdDevice            1.01
  iManufacturer           1 Corsair Components, Inc.
  iProduct                2 Corsair Gaming MM800 RGB POLARIS
  iSerial                 3 3BF002172184C5AECE995756411900F5
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           66
    bNumInterfaces          2
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              2 Corsair Gaming MM800 RGB POLARIS
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      33
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
[... 214 lines cut ...]
Jul 12 23:29:37.542882 dldsk01 ckb-daemon[6424]: Output (Request): ff 01 3c 00
Jul 12 23:29:37.542882 dldsk01 ckb-daemon[6424]: Input(Reply): 0e ff 01 3c 00 00 00 00
Jul 12 23:29:42.703879 dldsk01 ckb-daemon[6424]: [E] os_usbrecv (via led_keyboard.c:242): Connection timed out
Jul 12 23:29:42.703879 dldsk01 ckb-daemon[6424]: [W] _start_dev (device.c:67): Unable to load hardware profile
Jul 12 23:29:42.704334 dldsk01 ckb-daemon[6424]: [I] Setup finished for /dev/input/ckb1
Jul 12 23:30:06.158150 dldsk01 ckb-daemon[6424]: [I] Loading firmware version 0205 from /tmp/ckb-6586-fwblob.bin
Jul 12 23:30:19.707517 dldsk01 ckb-daemon[6424]: [I] Firmware update complete
Jul 12 23:30:19.714496 dldsk01 ckb-daemon[6424]: [I] Stopping input thread for /dev/input/ckb1
Jul 12 23:30:19.717951 dldsk01 ckb-daemon[6424]: [E] os_usbsend (via led_keyboard.c:119): No such device
Jul 12 23:30:19.717951 dldsk01 ckb-daemon[6424]: [I] Attempting reset...
Jul 12 23:30:19.818069 dldsk01 ckb-daemon[6424]: [E] os_resetusb (via usb.c:472): resetusb failed: No such device
Jul 12 23:30:19.818069 dldsk01 ckb-daemon[6424]: [E] usb_tryreset (usb.c:480): Reset failed. Disconnecting.
Jul 12 23:30:19.818069 dldsk01 ckb-daemon[6424]: [I] Disconnecting /dev/input/ckb1
Jul 12 23:30:20.014090 dldsk01 ckb-daemon[6424]: [I] Removed device path /dev/input/ckb1
Jul 12 23:30:36.323448 dldsk01 ckb-daemon[6424]: [I] >>>vendor = 0x1b1c, product = 0x1b20, path = /dev/bus/usb/003/003, syspath = /sys/devices/pci0000:00/0000:00:14.0/usb3/3-3
Jul 12 23:30:36.423707 dldsk01 ckb-daemon[6424]: [I] Connecting Corsair STRAFE RGB Gaming Keyboard at /dev/input/ckb1
Jul 12 23:30:36.423707 dldsk01 ckb-daemon[6424]: [I] claiming interfaces. name=Corsair STRAFE RGB Gaming Keyboard, firmware=0205; Got >> 3<< as ep_str
Jul 12 23:30:36.423707 dldsk01 ckb-daemon[6424]: [I] claiming 3 endpoints
Jul 12 23:30:36.574823 dldsk01 ckb-daemon[6424]: [I] Starting input thread for /dev/input/ckb1
Jul 12 23:30:38.685613 dldsk01 ckb-daemon[6424]: [I] Setup finished for /dev/input/ckb1

Daemon on last boot:
-- No entries --

```

francoism90, 2017-07-18:
K55 RGB:
```
ckb-dev-detect - Tue Jul 18 20:31:07 CEST 2017

Kernel: Linux desktop1 4.11.11-1-ck-skylake #1 SMP PREEMPT Sat Jul 15 13:14:40 EDT 2017 x86_64 GNU/Linux

OS Version:

USB Devices:

Bus 001 Device 003: ID 1b1c:1b3d Corsair 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1b1c Corsair
  idProduct          0x1b3d 
  bcdDevice           10.18
  iManufacturer           1 Corsair
  iProduct                2 Corsair Gaming K55 RGB Keyboard
  iSerial                 3 AECDA409577543F0F500194040048000
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           84
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      67
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               8
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
[... 78 lines cut ...]
        bInterval               1
Device Status:     0x0000
  (Bus Powered)

Dmesg on current boot:
Jul 18 20:22:59.438387 localhost kernel: input: Corsair Corsair Gaming K55 RGB Keyboard as /devices/pci0000:00/0000:00:14.0/usb1/1-10/1-10:1.0/0003:1B1C:1B3D.0005/input/input3
Jul 18 20:22:59.513270 localhost kernel: hid-generic 0003:1B1C:1B3D.0005: input,hidraw2: USB HID v1.11 Keyboard [Corsair Corsair Gaming K55 RGB Keyboard] on usb-0000:00:14.0-10/input0
Jul 18 20:23:10.023285 localhost kernel: hid-generic 0003:1B1C:1B3D.0006: usb_submit_urb(ctrl) failed: -1
Jul 18 20:23:10.023621 localhost kernel: hid-generic 0003:1B1C:1B3D.0006: timeout initializing reports
Jul 18 20:23:10.023854 localhost kernel: input: Corsair Corsair Gaming K55 RGB Keyboard as /devices/pci0000:00/0000:00:14.0/usb1/1-10/1-10:1.1/0003:1B1C:1B3D.0006/input/input4
Jul 18 20:23:10.093285 localhost kernel: hid-generic 0003:1B1C:1B3D.0006: input,hiddev0,hidraw3: USB HID v1.11 Keyboard [Corsair Corsair Gaming K55 RGB Keyboard] on usb-0000:00:14.0-10/input1

Dmesg on last boot:

Daemon on current boot:
-- Logs begin at Fri 2017-05-12 17:04:50 CEST, end at Tue 2017-07-18 20:31:06 CEST. --
Jul 18 20:26:37.495558 desktop1 systemd[1]: Started Corsair Keyboard Daemon.
Jul 18 20:26:37.497236 desktop1 ckb-daemon[22268]:     ckb: Corsair RGB driver v0.2.8
Jul 18 20:26:37.497481 desktop1 ckb-daemon[22268]: [I] Root controller ready at /dev/input/ckb0

Daemon on last boot:
-- No entries --
```

Thanks :)

SethBarberee, 2017-07-18:
After glancing at the Corsair website:
K55 RGB works with 3 distinct "zones" so that's the major hurdle that I see in implementing support. Another hurdle seems to be the macro record button since we don't have macros fully implemented yet.

Media keys and g-keys seem to borrow off the K95 Platinum after initial inspection so coding wise is a copy and paste.

Is this the only keyboard that does this kind of zoning? @frickler24 @light2yellow

francoism90, 2017-07-20:
@SethBarberee Thanks for trying to add support for the K55. :)
Unfortunately I have returned the keyboard because the quality wasn't good and I've switched to a mechanical keyboard.

Pract1cal, 2017-08-04:
Hi! Yesterday, I purchased an M65 Pro RGB mouse from Best Buy and downloaded CKB after seeing the CUE files open not in a program, but in a folder full of readme's and .exe's. Seeing as I have an iMac, I didn't see it fit to try and open those files as I have done this before and gotten many lines of code in a text document. After downloading CKB (using my new mouse to navigate around Google and GitHub), I opened the program and immediately noticed at the top of the page that my mouse was not being detected. I have a really ugly RGB default setup right now and I would love, love, love to change it, but the program doesn't seem to detect my mouse even after countless plugins. I checked the download page to see if my mouse is supported and, surprisingly, it is! However, it's been 1 night and my mouse is still not being seen by the program. Should I just keep the colors the way they are and just use the otherwise fully functional mouse?

akkbng, 2017-08-31:
Hi! Just got the corsair K70 Rapidfire non-rgb. The following is the report from the ckb-dev-detect:

Kernel: Darwin Bengisus-MacBook-Pro.local 16.5.0 Darwin Kernel Version 16.5.0: Fri Mar  3 16:52:33 PST 2017; root:xnu-3789.51.2~3/RELEASE_X86_64 x86_64

OS Version:
ProductName:	Mac OS X
ProductVersion:	10.12.4
BuildVersion:	16E195

USB Devices:
        Corsair Gaming K70 RAPIDFIRE Keyboard :

          Product ID: 0x1b3a
          Vendor ID: 0x1b1c  (CORSAIR MEMORY INC.)
          Version: 2.05
          Serial Number: 0301F020AEB8942955B0B4AFF5001946
          Speed: Up to 12 Mb/sec
          Manufacturer: Corsair
          Location ID: 0x14200000 / 2
          Current Available (mA): 500
          Current Required (mA): 500
          Extra Operating Current (mA): 0

light2yellow, 2017-09-01:
@akkbng it is already supported: https://github.com/mattanger/ckb-next/blob/6af27735f70436fc4535ce0b844ed4af0f96a69d/src/ckb-daemon/usb.h#L61

azacock, 2017-09-10:
Third-ing a request for support for the MM800 mouse pad.

sylvaindeloux, 2017-10-14:
Hi, I've a Corsair Gaming MM800 RGB POLARIS. Here is the related `ckb-dev-detect-report` : 

```
ckb-dev-detect - Sat Oct 14 19:03:38 CEST 2017

Kernel: Linux pc-mint 4.4.0-97-generic #120-Ubuntu SMP Tue Sep 19 17:28:18 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux

OS Version:
Distributor ID:	LinuxMint
Description:	Linux Mint 18.1 Serena
Release:	18.1
Codename:	serena

USB Devices:
Couldn't open device, some information will be missing
Couldn't open device, some information will be missing

Bus 001 Device 005: ID 1b1c:0c0a Corsair 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.10
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1b1c Corsair
  idProduct          0x0c0a 
  bcdDevice            1.00
  iManufacturer           1 
  iProduct                2 
  iSerial                 3 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           32
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower               50mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
[... 306 lines cut ...]
Oct 14 18:57:55.033160 pc-mint kernel: hid-generic 0003:1B1C:1B2F.0006: timeout initializing reports
Oct 14 18:57:55.033166 pc-mint kernel: input: Corsair Corsair Gaming SABRE RGB Mouse as /devices/pci0000:00/0000:00:14.0/usb1/1-10/1-10.2/1-10.2:1.1/0003:1B1C:1B2F.0006/input/input8
Oct 14 18:57:55.033198 pc-mint kernel: hid-generic 0003:1B1C:1B2F.0006: input,hiddev0,hidraw5: USB HID v1.11 Mouse [Corsair Corsair Gaming SABRE RGB Mouse] on usb-0000:00:14.0-10.2/input1
Oct 14 18:57:56.807281 pc-mint kernel: input: ckb1: Corsair Gaming SABRE RGB Mouse as /devices/virtual/input/input21
Oct 14 18:57:56.807293 pc-mint kernel: input: ckb1: Corsair Gaming SABRE RGB Mouse as /devices/virtual/input/input22

Dmesg on last boot:

Daemon on current boot:
-- Logs begin at Sat 2017-10-14 18:57:55 CEST, end at Sat 2017-10-14 18:58:55 CEST. --
Oct 14 18:57:56.497567 pc-mint systemd[1]: Started Corsair Keyboard Daemon.
Oct 14 18:57:56.498171 pc-mint ckb-daemon[1069]:     ckb: Corsair RGB driver v0.2.8
Oct 14 18:57:56.498217 pc-mint ckb-daemon[1069]: [I] Root controller ready at /dev/input/ckb0
Oct 14 18:57:56.502011 pc-mint ckb-daemon[1069]: [I] >>>vendor = 0x1b1c, product = 0x1b2f, path = /dev/bus/usb/001/007, syspath = /sys/devices/pci0000:00/0000:00:14.0/usb1/1-10/1-10.2
Oct 14 18:57:56.602484 pc-mint ckb-daemon[1069]: [I] Connecting Corsair Gaming SABRE RGB Mouse at /dev/input/ckb1
Oct 14 18:57:56.602539 pc-mint ckb-daemon[1069]: [I] claiming interfaces. name=Corsair Gaming SABRE RGB Mouse, firmware=0204; Got >> 3<< as ep_str
Oct 14 18:57:56.602586 pc-mint ckb-daemon[1069]: [I] claiming 3 endpoints
Oct 14 18:57:56.805128 pc-mint ckb-daemon[1069]: [I] Starting input thread for /dev/input/ckb1
Oct 14 18:57:58.998706 pc-mint ckb-daemon[1069]: [I] Setup finished for /dev/input/ckb1

Daemon on last boot:
Specifying boot ID has no effect, no persistent journal was found
```

Do you need more information?

danilosaft, 2017-12-08:
I have a MM800 RGB Polaris mousepad as well, along with a K95 RGB Platinum. The Platinum is very well supported, but the MM800 doesn't list as a tab and constantly changes colors at the moment. Here is my skb-dev-detect report. Also, I will try and support this project with the limited hardware but more sophisticated gui programming skills I have, if needed.

`
ckb-dev-detect - Thu Dec  7 22:57:43 CET 2017

Kernel: Linux ANSARI 4.9.0-3-amd64 #1 SMP Debian 4.9.30-2+deb9u3 (2017-08-06) x86_64 GNU/Linux

OS Version:
Distributor ID:	Debian
Description:	Debian GNU/Linux 9.1 (stretch)
Release:	9.1
Codename:	stretch

USB Devices:

Bus 010 Device 015: ID 1b1c:1b3b Corsair 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1b1c Corsair
  idProduct          0x1b3b 
  bcdDevice            1.02
  iManufacturer           1 Corsair Components, Inc.
  iProduct                2 Corsair Gaming MM800 RGB POLARIS
  iSerial                 3 260002076184BEAECA353956471900F5
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           66
    bNumInterfaces          2
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 No Subclass
      bInterfaceProtocol      0 None
      iInterface              2 Corsair Gaming MM800 RGB POLARIS
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      33
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
[... 273 lines cut ...]
Dec 07 21:54:14.478558 ANSARI kernel: hid-generic 0003:1B1C:1B3B.0011: hiddev0,hidraw3: USB HID v1.11 Device [Corsair Components, Inc. Corsair Gaming MM800 RGB POLARIS] on usb-0000:00:1d.0-1.5.1/input0
Dec 07 21:54:14.479613 ANSARI kernel: input: Corsair Components, Inc. Corsair Gaming MM800 RGB POLARIS as /devices/pci0000:00/0000:00:1d.0/usb10/10-1/10-1.5/10-1.5.1/10-1.5.1:1.1/0003:1B1C:1B3B.0012/input/input31
Dec 07 21:54:14.539900 ANSARI kernel: hid-generic 0003:1B1C:1B3B.0012: input,hidraw4: USB HID v1.11 Keyboard [Corsair Components, Inc. Corsair Gaming MM800 RGB POLARIS] on usb-0000:00:1d.0-1.5.1/input1
Dec 07 22:08:11.193922 ANSARI kernel: input: ckb1: Corsair Gaming K95 RGB PLATINUM Keyboard as /devices/virtual/input/input34
Dec 07 22:08:11.194022 ANSARI kernel: input: ckb1: Corsair Gaming K95 RGB PLATINUM Keyboard as /devices/virtual/input/input35

Dmesg on last boot:

Daemon on current boot:
-- Logs begin at Mon 2017-08-28 18:35:04 CEST, end at Thu 2017-12-07 22:57:41 CET. --
Dec 07 22:08:10.751357 ANSARI systemd[1]: Started Corsair Keyboard Daemon.
Dec 07 22:08:10.752848 ANSARI ckb-daemon[8075]:     ckb: Corsair RGB driver v0.2.8
Dec 07 22:08:10.753132 ANSARI ckb-daemon[8075]: [I] Root controller ready at /dev/input/ckb0
Dec 07 22:08:10.767252 ANSARI ckb-daemon[8075]: [I] >>>vendor = 0x1b1c, product = 0x1b2d, path = /dev/bus/usb/010/013, syspath = /sys/devices/pci0000:00/0000:00:1d.0/usb10/10-1/10-1.4
Dec 07 22:08:10.867602 ANSARI ckb-daemon[8075]: [I] Connecting Corsair Gaming K95 RGB PLATINUM Keyboard at /dev/input/ckb1
Dec 07 22:08:10.867602 ANSARI ckb-daemon[8075]: [I] claiming interfaces. name=Corsair Gaming K95 RGB PLATINUM Keyboard, firmware=0107; Got >> 3<< as ep_str
Dec 07 22:08:10.867602 ANSARI ckb-daemon[8075]: [I] claiming 3 endpoints
Dec 07 22:08:11.191016 ANSARI ckb-daemon[8075]: [I] Starting input thread for /dev/input/ckb1
Dec 07 22:08:11.306647 ANSARI ckb-daemon[8075]: [E] getfwversion (firmware.c:18): Bad input header
Dec 07 22:08:11.306647 ANSARI ckb-daemon[8075]: [W] _start_dev (device.c:41): Unable to load firmware version/poll rate
Dec 07 22:08:11.307097 ANSARI ckb-daemon[8075]: [I] Setup finished for /dev/input/ckb1

Daemon on last boot:
Specifying boot ID has no effect, no persistent journal was found
`

ZirconiumX, 2017-12-09:
Please (join us in IRC)[http://webchat.freenode.net/?channels=%23ckb-next&uio=d4] so we can help you better. It's hard to communicate for this kind of thing on GitHub.

closed 2018-01-21 by tatokis

KenEucker, 2019-02-16:
Did we ever get the MM800 Polaris? It's the only device in my corsair lineup that isn't recognized by ckb-next.

tatokis, 2019-02-16:
Yes, https://github.com/ckb-next/ckb-next/