mattanger/ckb-next #3: Road Map

mattanger, 2016-12-06:
Ideas for outlining the new feature/plan roadmap.

mattanger, 2016-12-13:
@tatokis @fleischie Either of you have any desire to move to a different build system? I've been considering getting CMake going. My motivation completely selfish because my IDE of choice doesn't support qmake.

fleischie, 2016-12-20:
@mattanger what IDE are you using?

I am all in favor of porting to another, more generic build system. I only did a shallow search for comparisons (having only basic knowledge of both `q-` and `cmake`) but it seems `qmake` is somewhat the ready-made solution for Qt-projects. `cmake` is supposedly a little harder to setup, as many steps need to be reproduced manually.

Did you already start with this or are you just contemplating it? I would say, if you want to and manage to procude some usable results, why not change. :stuck_out_tongue_winking_eye:

light2yellow, 2016-12-20:
@fleischie I bet he uses Clion (which I use as well :)) which currently supports cmake only projects (and probably will stick to such architecture for a couple of years because other features need to be implemented). So I'd be happy to see a transition on cmake too.

fleischie, 2016-12-20:
@light2yellow OK, thanks. I only heard good about IntelliJ-derivatives so far. :smile:

There seem to be scripts available online, which convert qmake projects automatically to cmake https://github.com/Lnd-stoL/qmake2cmake/ 

Other advantages are:
* CMake is available on different platforms as well (Unix, Windows, OSX),
* CMake has arguably a better console output than qmake/make if done right and
* maybe it helps to make the project more independent in general (for example allowing to simply remove Qt, if it's not necessary anymore).

Also maybe we can start with smaller sub-packages of `ckb`, which do not rely heavily on Qt (like for example the daemon). If this works there, we can incrementally port the project to `cmake`. :smile: :+1:

light2yellow, 2016-12-20:
@fleischie with all the respect to the author of this script it doesn't seem to me any good project should rely on such black box. Somebody just needs to invest his time into it and create a hand-written one.

fleischie, 2016-12-20:
@light2yellow You're right. :wink: The script should merely act as an example. I prefer manual conversion as well. :smile:

fleischie, 2016-12-20:
Apart from that, is anyone interested in having a chat somewhere? (gitter/discord/etc?) Maybe for synchronous conversions, when developing simultaneously or troubleshoot inoperative devices etc.?

tatokis, 2016-12-20:
While I don't think we are all available at common times, for now at least, I'd like to request that we use IRC. 

The reasons for that being that some of the aforementioned platforms do not support my operating system of choice, meaning that I have to either use browser clients, or unofficial ones. 

 Additionally they are quite resource heavy and don't run on some of the devices I use. 

I will host public logs if needed.

tatokis, 2016-12-20:
Regarding the build system, I am neutral to which one we use. 
All the development I've been doing so far was done with SciTE and gdb/make on the terminal.

mattanger, 2016-12-20:
@light2yellow right on the money, and @fleischie CLion is a pretty damn pleasant experience compared to some of the other IDEs out there. Bit memory hungry though. Anyway, I have started working on a switch to cmake, and it's actually not very hard to use Qt with it. On that note, I've only been targeting Linux and I may need some help doing the OSX part. 

I'm down for getting chat going to, anything is good with me. @tatokis IRC is cool with me if you wanna set a channel up somewhere.

tatokis, 2016-12-22:
I was planning to set up an IRC room, but once I attempted to do it, I 
realised that we don't have a name for the project, and I don't think 
it would be appropriate to name the room "ckb".

I propose we use "ckb-next". Any other suggestions?

zimmertr, 2016-12-22:
ckb-ULTRA

mattanger, 2016-12-23:
I'll make a new thread for suggestions, and we can have a vote discussion.

zimmertr, 2016-12-23:
Hey guys, I've never collaborated with other people on a Github project aside from a little bit at work. I'd really like to be involved in the revival of this project. How can I help? 

I'm exceptionally good with Linux and automation. Fair with programming in JAVA, C#, and Ruby.

mattanger, 2016-12-29:
@zimmertr Thanks for your offer! It's possible we may need some scripting done in the future, but I think the main thing we'll need is testers for as many systems as possible. Specifically Apple products and not Arch Linux.

zimmertr, 2016-12-29:
Awesome, well I have a Macbook Pro running Sierra too.

light2yellow, 2016-12-29:
I can test (another day) as well on mac 10.12.2 (hackintosh).

mattanger, 2016-12-29:
For anyone interested. I've pushed a new branch, cmake. Right now it's linux only and barebones, but it's building for me on linux. Be nice if anyone could give it a build, and lemme know if there are any issues on linux so far. Ignore -lz linking warning, the flags need a little cleaning up. 

Out of source build: 
mkdir build/ && cd build
cmake .. 
make

light2yellow, 2016-12-31:
Builds for me too @mattanger.

tatokis, 2017-01-02:
I've set up #ckb-next on Freenode.
ircs://chat.freenode.net:6697/#ckb-next (SSL/TLS)
Otherwise, here's a webchat link https://webchat.freenode.net?channels=%23ckb-next

mattanger, 2017-01-02:
Thanks @light2yellow and @tatokis

tatokis, 2017-01-02:
@mattanger  I think you should lower the minimum Qt5 version, as cmake refuses to compile ckb for me and I am on Ubuntu 16.04.
```
CMake Error at src/ckb/CMakeLists.txt:30 (message):
  Minimum supported Qt5 version is 5.6!
```
Ubuntu-wise, last I checked it compiled on 14.04, and that has Qt 5.2.1

mattanger, 2017-01-02:
Ok. I'll do that.

tatokis, 2017-01-02:
So, I've been editing the UI, renaming ckb to ckb-next.
I came across the copyright in the settings page.

I changed it to this, 
[image]
however I am not sure what to have "the ckb development team" point to.
It is currently set to https://github.com/mattanger/ckb-next/graphs/contributors but I am not sure if it is appropriate.

Suggestions?

ksexton, 2017-01-13:
Thoughts on splitting into two different projects?  One for the daemon and one for the GUI interface?  With the daemon running the GUI is optional and other programs can control the keyboard.

mattanger, 2017-01-15:
I think the flag is definitely a better option right now.

batchpert, 2017-01-15:
Hi
Really pleased this project seems to be developing quickly at the moment. Just wondering if since the original ckb with its timeline of whats happening in each version whether or not that is the same as your 'road map'. Just as a fan of the program interested in whats coming next!!!

light2yellow, 2017-01-15:
@mattanger regarding CMake building system on macOS:
* it doesn't build in the current `cmake` branch state
* it requires `homebrew` as a dependency and both `cmake` and `qt5` installed with `homebrew` (and not the official ones)
* it requires `CMAKE_PREFIX_PATH` to be set, for example:
```cmake
if(${CMAKE_SYSTEM_NAME} MATCHES "Darwin")
    message(STATUS "Building for macOS")
    execute_process(COMMAND brew --prefix qt5  # wrapping it in quotes is wrong
            OUTPUT_VARIABLE CMAKE_PREFIX_PATH OUTPUT_STRIP_TRAILING_WHITESPACE)
    message(STATUS "CMAKE_PREFIX_PATH: " ${CMAKE_PREFIX_PATH}) # prints /usr/local/opt/qt5
endif()
```

closed 2018-01-21 by tatokis