mattanger/ckb-next #4: Pull Requests merge progress.

tatokis, 2016-12-09:
This is a quick list to be used as a reference for which PRs have been merged from the original repository and which haven't.

* https://github.com/ccMSC/ckb/pull/530
* https://github.com/ccMSC/ckb/pull/516
* https://github.com/ccMSC/ckb/pull/514
* https://github.com/ccMSC/ckb/pull/511
* https://github.com/ccMSC/ckb/pull/508
* https://github.com/ccMSC/ckb/pull/507
* https://github.com/ccMSC/ckb/pull/467
* https://github.com/ccMSC/ckb/pull/455
* https://github.com/ccMSC/ckb/pull/437
* https://github.com/ccMSC/ckb/pull/403 
* https://github.com/ccMSC/ckb/pull/367

You might notice that some pull requests are not listed here, despite being open.
That is because they were merged to the testing branched and then pulled into this fork.

mattanger, 2016-12-11:
>    - ccMSC#530 <https://github.com/ccMSC/ckb/pull/530>
>    - ccMSC#516 <https://github.com/ccMSC/ckb/pull/516>
>    - ccMSC#514 <https://github.com/ccMSC/ckb/pull/514>
>    - ccMSC#511 <https://github.com/ccMSC/ckb/pull/511>
>    - ccMSC#508 <https://github.com/ccMSC/ckb/pull/508>
>    - ccMSC#507 <https://github.com/ccMSC/ckb/pull/507>
>    - ccMSC#467 <https://github.com/ccMSC/ckb/pull/467>
>    - ccMSC#455 <https://github.com/ccMSC/ckb/pull/455>
>    - ccMSC#437 <https://github.com/ccMSC/ckb/pull/437>
>    - ccMSC#403 <https://github.com/ccMSC/ckb/pull/403>
>    - ccMSC#367 <https://github.com/ccMSC/ckb/pull/367>

Gotcha. Thanks for the heads up.

batchpert, 2016-12-11:
Hi its nice to see some progress finally being made on this :D however i don't know how you add these individual bits into the full ckb application. Any chance you could make a quick guide on what to do? I would be very grateful!!
Thanks

tatokis, 2016-12-27:
I've brought newfx up to date, and I made some changes so that it compiles on linux.
Could someone test it (ckb-mviz) on linux?

Just want to be sure that it isn't a case of "it only works on my machine".

fleischie, 2016-12-27:
@tatokis what needs toll bei tested? I can check, whether it builds and the daemon/GUI starts without crashing. But I don't have my keyboard with me over the holidays.

mattanger, 2016-12-27:
Yeah I can later today.

tatokis, 2016-12-27:
Basically need to see if it compiles properly on other systems. And if 
possible, test it on a system that does not have pulseaudio.
It's supposed to check for libpulse simple and if it doesn't exist, 
ignore ckb-mviz completely.

fleischie, 2016-12-27:
@tatokis Compiles and daemon + GUI runs (only basic settings, because of no keyboard).

I couldn't run `ckb-mviz` though, because (again) of no keyboard.

tatokis, 2016-12-27:
Alright, thanks

mattanger, 2016-12-28:
Builds find and works fine with my keyboard. I do have pulseaudio. I'll try to build it on a vm without pulse sometime today.

tatokis, 2017-04-03:
Regarding ccMSC/ckb#508 do we merge it or not?
Otherwise this issue should be closed.

tatokis, 2017-04-03:
> <l2y>	-f is a good thing, but that line should've changed since the PR was made and it's easier to just add it manually

Fair enough point

closed 2017-04-03 by tatokis