mattanger/ckb-next #6: The new name!

mattanger, 2016-12-23:
Comment with your vote, or suggest another and I'll add it to the list. 

Suggestions so far: 
ckb-next
ckb-ULTRA
ckb-reloaded 
????

fgblomqvist, 2016-12-23:
While I'm not a contributing developer, I'm an avid fan (and very happy you guys are reviving this project).

I vote on ckb-reloaded.

tbcrawford, 2016-12-23:
I as well am not a contributing developer. I can contribute problems I see and how I interpret them since I understand C and C++. I am very happy to help in any way I can with this project and will see what I can learn about it as I go. If you care about my vote, I vote for ckb-next.

fleischie, 2016-12-24:
I am in favor of `ckb-next` because it's still short and hasn't too much co-notations to it.

light2yellow, 2016-12-25:
`ckb-%smth%` should suffice if the bus factor won't be 1 and we won't have the single owner rule it all (that's what happened with `ckb`). Otherwise `ckb2` or `ckb-second` should be concerned in case this project gets abandoned as well.

HeroCC, 2016-12-25:
+1 for ckb2, it is short and has a nice ring to it, while still being simple, and defines the project in an easy way.

tatokis, 2016-12-27:
I'm going to go with ckb-next as I feel it is short and on point. It's not really common to have numbers in project/application names.

@light2yellow considering there are three developers right now, I don't really see it happening. I guess as long as @mattanger doesn't randomly disappear, we should be fine.

mattanger, 2016-12-28:
I'll do my best not to!

fleischie, 2016-12-29:
@mattanger @tatokis how long should we keep the poll open?

@mattanger the `ckb-ULTRA` name was mentioned in the origin ckb project's issues - if I recall correctly - so I say it should be allowed it's vote as well. 😉

HeroCC, 2016-12-29:
When the new name is decided, I don't think it would be a bad idea to set up a team. That way there can be multiple ckb oriented repos, @tatokis and others can be added, removed, and can commit to them as needed. Plus it makes it seem more 'official' and organised. :smile:

mattanger, 2016-12-29:
@fleischie haha well. I'll allocate "it's" vote accordingly, though I'm not sure that the vote will put it ahead...... :stuck_out_tongue: 

And I'm ok with closing this poll immediately. I was planning on casting my vote with the leading name, +1 ckb-next. AND since we, @fleischie and @tatokis, are the only ones who really matter :wink: we can just make that decision!

tatokis, 2016-12-29:
Welp, ckb-next it is!

fleischie, 2016-12-29:
Yay for oligarchy. 😂

tatokis, 2016-12-29:
Well, jokes aside, I think it was pretty democratic :p

mattanger, 2016-12-29:
The Glorious Republic of ckb-next's central planning committee for democratic names has reached a decision.

closed 2016-12-29 by mattanger