mattanger/ckb-next #10: Release packages for macOS

tbcrawford, 2016-12-28:
I cannot do this I do not believe even through a PR, but could we get a `ckb beta 0.2.7` pkg release file for easy download and installation for users being directed here? I apologize if this is in the works and you guys are waiting on a few other items before doing a 0.2.7 release but I thought it might be nice to have. Of course, feel free to close this if it is in the works. I have no problem building from source now.

tatokis, 2016-12-28:
We are mostly waiting for the rebranding to be finished, and make sure 
everything is stable.
Once that is done, we are going to provide official binaries.

Until then, you can possibly link your binary to people asking for help.

tbcrawford, 2016-12-28:
That's fine. I wasn't sure what the exact plans were. Thanks

closed 2016-12-28 by mattanger

light2yellow, 2017-05-06:
The package will be released today.

chicks-net, 2017-05-08:
Congratulations on your first release!