mattanger/ckb-next #11: Random hang on Linux.

tatokis, 2016-12-29:
CKB Randomly hangs on Ubuntu 16.04 with a simple rainbow effect running.
Attaching strace to the main process reveals that is waiting for a child to finish.
```
strace: Process 17818 attached
futex(0x2472d74, FUTEX_WAIT_PRIVATE, 1, NULL
```
In this case, the only child is ckb-wave, attaching strace to it shows
```
strace: Process 19971 attached
read(0, 
```
meaning it's waiting endlessly.

Lsof for the child pid:
```
# lsof -p 19971
COMMAND    PID    USER   FD   TYPE DEVICE SIZE/OFF     NODE NAME
ckb-wave 19971 tatokis  cwd    DIR   8,18    32768   901581 /home/tatokis
ckb-wave 19971 tatokis  rtd    DIR   8,18     4096        2 /
ckb-wave 19971 tatokis  txt    REG   8,18    22584  7275854 /usr/bin/ckb-animations/ckb-wave
ckb-wave 19971 tatokis  mem    REG   8,18  1864888  5970818 /lib/x86_64-linux-gnu/libc-2.23.so
ckb-wave 19971 tatokis  mem    REG   8,18  1088952  5970828 /lib/x86_64-linux-gnu/libm-2.23.so
ckb-wave 19971 tatokis  mem    REG   8,18   162632  5970609 /lib/x86_64-linux-gnu/ld-2.23.so
ckb-wave 19971 tatokis    0r  FIFO   0,10      0t0 95761765 pipe
ckb-wave 19971 tatokis    1w  FIFO   0,10      0t0 95761766 pipe
ckb-wave 19971 tatokis    2w  FIFO   0,10      0t0 95761767 pipe
ckb-wave 19971 tatokis   20w  FIFO    0,6      0t0    21002 /dev/input/ckb1/cmd
-----
# lsof -n -P | grep 95761765
ckb       17818                   tatokis   34w     FIFO               0,10          0t0   95761765 pipe
QXcbEvent 17818 17819             tatokis   34w     FIFO               0,10          0t0   95761765 pipe
gmain     17818 17820             tatokis   34w     FIFO               0,10          0t0   95761765 pipe
gdbus     17818 17821             tatokis   34w     FIFO               0,10          0t0   95761765 pipe
dconf\x20 17818 17822             tatokis   34w     FIFO               0,10          0t0   95761765 pipe
QThread   17818 17827             tatokis   34w     FIFO               0,10          0t0   95761765 pipe
Kb        17818 17837             tatokis   34w     FIFO               0,10          0t0   95761765 pipe
Qt\x20bea 17818 17839             tatokis   34w     FIFO               0,10          0t0   95761765 pipe
Qt\x20HTT 17818 17840             tatokis   34w     FIFO               0,10          0t0   95761765 pipe
ckb-wave  19971                   tatokis    0r     FIFO               0,10          0t0   95761765 pipe
```

mattanger, 2016-12-29:
Hmm. Any ideas/steps to reproduce? Or just let ckb-wave run until it hangs?

tatokis, 2016-12-29:
Sometimes I have the lights on my keyboard off. After some time, I try to turn them back on, and the brightness button does not respond.
Unminimizing the GUI shows that it has been greyed out by my window manager due to it being locked up.

Keyboard input and ckb daemon continue to work normally with the ckb GUI locked up.

It basically just happens randomly, haven't been able to figure out any specific triggers.

mattanger, 2016-12-29:
Ok. I'll see if I can make make it happen. So maybe try turn on ckb-wave, turn off the lights for a while, and turn them back on and see if it's crashed? 

Are you using Gnome/Unity shell?

tatokis, 2016-12-29:
Yeah, Unity.

mattanger, 2016-12-29:
Aight. I'm on Gnome 3. Maybe it'll happen for me to.

tatokis, 2017-01-16:
Looks like this was a case of PEBCAK. Closing.

closed 2017-01-16 by tatokis