mattanger/ckb-next #12: Explicit QT Version

mattanger, 2016-12-29:
@tatokis I saw that in the issue https://github.com/ccMSC/ckb/issues/549 you were talking to someone about the required Qt version. Do you or @fleischie have any opposition to making the Qt 5.6 minimum version an explicit requirement in the documentation? I've finally made some time to work on the CMake version, and I've made 5.6 the min there. 

Also, @tatokis what did you mean by not 5.7? Not as in one doesn't need it?

tatokis, 2016-12-29:
Qt 5.6 was the last Qt version to support Carbon on Mac.
Qt 5.7 does not support carbon, which means ckb-daemon does not compile 
and fails with an error about not being able to find Carbon.h

I had a quick look to see if we can convert ckb-daemon to use Cocoa 
instead, however it seems like it needs quite some work and I have 
absolutely zero knowledge regarding mac development.

So basically the latest Qt version to work with ckb is 5.6. 5.7 will 
fail to compile.

The real question is why, on Mac, does ckb-daemon depend on Qt in the 
first place.

tatokis, 2016-12-29:
Qt 5.7 _should_ be fine on Linux, however Ubuntu 16.04 does not come with it, so I can't test anything.

mattanger, 2016-12-29:
I see. Well, that's annoying. It could be that the daemon depends on some Qt core lib or something, I dunno. I have no clue about Mac development either. I'll see what I can figure out. 

I'm compiling with 5.7 and everything is fine for me, on Linux. That's why I was asking.

fleischie, 2016-12-29:
@mattanger I am currently working through the daemon to fully understand the workflow and to potentially fix some mac-related issues there.

This will however take some time still, as it's a new platform for me either. I will keep you updated on this issue, if I find out.

mattanger, 2016-12-29:
@fleischie Awesome. I've got access to OS X right now, and I'm going to start testing out some builds on it. Since you're already going in depth on the daemon, I'm not going to think to deeply about it, but I am gonna mess with this issue some to see if I uncover any useful info.

tbcrawford, 2016-12-29:
@tatokis have you tried building and running with Qt 5.7 lately on macOS? I have 5.7 installed and can compile it fine it seems. I previously had a problem where Carbon couldn't be found but I do believe I fixed that by ensuring that I had Xcode command line tools properly installed. They were not properly installed previously to my surprise.

fruityten, 2017-01-13:
Just installed on Sierra using Qt 5.7 and the only issues were with Xcode. I had to go to into Xcode -> Preferences -> Locations -> Command Line Tools and select 8.2.1 and then all worked like a charm.

Might save someone some faffing around. Also great work on the fork to everyone involved.

light2yellow, 2017-01-13:
@fruityten what were the issues? Did you have CLT installed previously? Can you test your configuration against new `quickinstall` [from here](https://github.com/light2yellow/ckb-next/blob/quickinstall-improvement/quickinstall)? Just execute `./quickinstall`.

fruityten, 2017-01-13:
@light2yellow Thinking it over CLT will have been installed but not enabled, I'd only installed Xcode via the App store with the licence agreed to on this machine.

At this point, running the quickinstall script failed because it couldn't find the macosx SDK. After I changed the settings in the Xcode prefs (which enables the CLT) all ran smoothly and the quickinstall script ran fine.

light2yellow, 2017-01-13:
@fruityten I'm not sure I understand the Apple's "think different" situation around XCode and CLT. From what I've experienced so far CLT are now just standalone package without any connection to XCode and installing XCode from the Store but not installing CLT using terminal does nothing. Anyway, now the script checks for CLT being installed somehow somewhere and this must work in most situations.

light2yellow, 2017-03-04:
In ccMSC/ckb#549 it ended up installing CLT and I see no reason introducing 5.6 as a required minimum. Can somebody show me why we should do this? Because by default we shouldn't do this. It compiles and works as supposed with 5.5, 5.6, 5.7 and 5.8, without any problems (except for 5.7 which has some vague segfaults).

tatokis, 2017-06-15:
This seems to have been sorted out by now. Closing

closed 2017-06-15 by tatokis