mattanger/ckb-next #15: K70 LUX RGB won't be recognised

daniel-law, 2016-12-30:
Hi there, been pulling my hair out for hours trying to get the CKB software to pick up my K70 LUX RGB. It's new, I've force updated the firmware on Windows and I'm running macOS Sierra 10.12.2. I've installed Xcode and the command line tools for Xcode as well. Ensured I'm running the correct version of Qt which is 5.6.2. Attached my log which has tons of errors, hopefully that helps.
[log.txt](https://github.com/mattanger/ckb-next/files/678998/log.txt)

tbcrawford, 2016-12-30:
@daniel-law Did you clone from this project recently? Or pull any new updates? I believe this should be fixed from everything I'm understanding so far. Your log shows this error: 
```
./os.h:69:13: error: typedef redefinition with different types ('int' vs 'enum clockid_t')
typedef int clockid_t;
```

But when looking at the current [source in os.h line 69](https://github.com/mattanger/ckb-next/blob/master/src/ckb-daemon/os.h#L69), `typedef int clockid_t;` is not defined. This is not defined until [line 74](https://github.com/mattanger/ckb-next/blob/master/src/ckb-daemon/os.h#L74).

Try pulling to see if you can update to the latest source and then running `./quickinstall` again. If that does not work, I will link a package that I have built from the latest source for you to try.

daniel-law, 2016-12-30:
Thanks so much, was completely my fault. Not sure whether it's me just being unfamiliar with GitHub, but I installed the latest release before instead of cloning the project.

closed 2016-12-30 by daniel-law

tbcrawford, 2016-12-30:
Yeah no problem! Glad I could help.