mattanger/ckb-next #17: Rewrite README

light2yellow, 2016-12-31:
So, as the new project is launched I guess we need to partially rewrite README. For example, what do you think about donation buttons? I suggest that for now we should at least introduce the situation around the original `ckb` and then mark the donation buttons as the donations to the original author. Then packaging: at least for Arch Linux the links are valid for `ckb` and so we need to explicitly say this. And the general structure is rather cluttered and the file is too big to quickly find something. Maybe we should split it into TROUBLESHOOTING.md, CONTRIBUTON.md etc.? And other cosmetic changes that will make the text shorter and more clear. As a side note, there are a lot of people with corsair keyboards on mac knowing nothing about github and yet finding `ckb` repo (and maybe `ckb-next`) without any clue where to start with because the instructions are too long and not a bit unclear.

fleischie, 2016-12-31:
Yes, this is a valid concern and I was meaning to update this for a few weeks.

If you want to start with this and open a PR, it will be very welcome. Otherwise I'll start with this by the end of next week I guess.

light2yellow, 2016-12-31:
Great, I'll edit it a bit. We'll also need a native speaker to blue-pencil it.

mattanger, 2016-12-31:
Agreed. I don't mind taking up some editing. I could some more editing, I can either work on it sometime tomorrow or the next day, unless you get moving on it first @light2yellow

mattanger, 2017-01-23:
I've merged in these changes, and we can open a new issue for any future changes.

closed 2017-01-23 by mattanger