mattanger/ckb-next #21: Compiler version

sh4d0w28, 2017-01-09:
Need a help. 

I`m trying to build newfx branch using Ubuntu 16.04 x64 with gcc 5.4 / g++ 5.4.

Can you try to build newfx branch and provide me version of gcc and g++ you using?
Also, i will need "uname -a" information.  Thanks.

mattanger, 2017-01-09:
Builds fine for me. 
gcc version 6.2.1
kernel: 4.8.13-1-ARCH

Where is the build failing? Errors?

sh4d0w28, 2017-01-10:
> gcc -c -m64 -pipe -std=c99 -g -fPIC  -I../../../ckb/src/ckb-mviz -I. -I/usr/lib/x86_64-linux-gnu/qt5/mkspecs/linux-g++-64 -o main.o ../../../ckb/src/ckb-mviz/main.c
> gcc -c -m64 -pipe -std=c99 -g -fPIC  -I../../../ckb/src/ckb-mviz -I. -I/usr/lib/x86_64-linux-gnu/qt5/mkspecs/linux-g++-64 -o kiss_fft.o ../../../ckb/src/ckb-mviz/kiss_fft.c
> gcc -c -m64 -pipe -std=c99 -g -fPIC  -I../../../ckb/src/ckb-mviz -I. -I/usr/lib/x86_64-linux-gnu/qt5/mkspecs/linux-g++-64 -o kiss_fftr.o ../../../ckb/src/ckb-mviz/kiss_fftr.c
> g++ -m64 -lpulse-simple -lpulse -o ../../../ckb/bin/ckb-animations/ckb-mviz main.o kiss_fft.o kiss_fftr.o    
> main.o: In function `ckb_init':
> /home/mvladimirovich/qt/build-ckb-Desktop-Debug/src/ckb-mviz/../../../ckb/src/ckb-mviz/main.c:51: undefined reference to `pa_simple_new'
> main.o: In function `getFreqDec':
> Makefile:133: recipe for target '../../../ckb/bin/ckb-animations/ckb-mviz' failed
> make[1]: Leaving directory '/home/mvladimirovich/qt/build-ckb-Desktop-Debug/src/ckb-mviz'
> Makefile:201: recipe for target 'sub-src-ckb-mviz-make_first' failed
> /home/mvladimirovich/qt/build-ckb-Desktop-Debug/src/ckb-mviz/../../../ckb/src/ckb-mviz/main.c:93: undefined reference to `pa_simple_read'
> collect2: error: ld returned 1 exit status
> make[1]: *** [../../../ckb/bin/ckb-animations/ckb-mviz] Error 1
> make: *** [sub-src-ckb-mviz-make_first] Error 2
> 18:33:35: The process "/usr/bin/make" exited with code 2.
> Error while building/deploying project ckb (kit: Desktop)
> When executing step "Make"

tatokis, 2017-01-10:
You don't have the libpulse-simple dev headers installed.
Qmake was supposed to check if it was installed or not.
You used newfx from ckb-next, right?

sh4d0w28, 2017-01-10:
No. Its still ckb. But you correct about libpulse-simple. How can i check it installed?

sh4d0w28, 2017-01-10:
I have this command:
[~/qt/ckb]: [master] pkg-config --libs libpulse-simple
-lpulse-simple -lpulse

tatokis, 2017-01-10:
There's your problem, then.

First of all, switch to ckb-next, second it should be ```sudo apt-get install libpulse-dev```

sh4d0w28, 2017-01-10:
Building dependency tree       
Reading state information... Done
libpulse-dev is already the newest version (1:8.0-0ubuntu3.1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

tatokis, 2017-01-10:
Interesting. Can you check if this file exists in your system? ```/usr/lib/x86_64-linux-gnu/libpulse-simple.so```

tatokis, 2017-01-10:
Actually, no, you *HAVE* to switch to ckb-next.
I pushed a fix to actually compile, as it wouldn't before.

https://github.com/mattanger/ckb-next/commit/ed02527b9987e00dacb01690be697ac573b78a42

sh4d0w28, 2017-01-10:
Cool. Not saw your commit last time.

closed 2017-01-10 by sh4d0w28