mattanger/ckb-next #23: Caps lock freezes the whole system (k70 lux) [macOS Sierra)

ahmadfiroz, 2017-01-13:
Tried some of the ways found in google but nothing works actually. Uninstalling CKB works good except the indicator lights ( which was also not working correctly with ckb)

light2yellow, 2017-01-13:
I experience the same problem with Strafe RGB. Is it possible to debug it/compile with debug logs?

light2yellow, 2017-01-15:
Here's my log using `dtruss` (this is `strace` alternative for macos) on 10.12.2. What I do here:
1. Launch it with `sudo dtruss -f -n ckb-daemon &> ~/ckb-daemon.dtruss`.
2. Press `Caps Lock`, `Enter`, `a` (capitalised), `Enter`, `Caps Lock`, `a`, `Enter`, switch with `Cmd + Tab` to another app (this freezes the whole system on a few seconds), switch back to terminal (freezes the same way), `Enter`, `Ctrl + C` (to kill the process).
The whole log is in the file, I cannot use pastebin because it's too long.
[ckb-daemon.dtruss.txt](https://github.com/mattanger/ckb-next/files/706573/ckb-daemon.dtruss.txt)

tbcrawford, 2017-01-16:
Just thought I might chime in and add that I too have this problem with the Strafe RGB on Sierra.

smachi, 2017-01-17:
Try to change the Caps Lock lights from Normal to RGB indicator and restart your computer

guenniman, 2017-01-17:
Same Problem with STRAFE RGB on Linux Mint 18 (4.8 Kernel)

hyzenthlayed, 2017-01-19:
Having the same issue with a K65 LUX RGB on an iMac (2010) running macOS Sierra.

frickler24, 2017-01-21:
@light2yellow @tbcrawford @guenniman pls can one of you send me the output from ckb-daemon?
I assume that in the "freezing-time" the daemon is restarting because of an exception (maybe an SEGV or similar). If we can reproduce a restarting sequence, I have an idea to look further...
You can either start the daemon as root manually in a terminal and copy/paste the output or you have a look into you system logs (filter for ckb). Because you are mac-users and I do not have a mac, I cannot give hint to you how to manage this on MacOs. you may also have a look at the PIDs. If the PIDs of the ckb-daemon threads change after a "freezing period", it is also an indicator for a crash.

light2yellow, 2017-01-26:
@frickler24 unfortunately I don't have much time these days but here's the info for everyone else (and for me to return here later) - [here's how you can debug a daemon on new launchd versions](http://serverfault.com/a/803146). Also @tatokis [found a function](https://github.com/mattanger/ckb-next/blob/master/src/ckb-daemon/input_mac.c#L280) which might be the culprit.

JarrenJ, 2017-01-28:
I might have found possible fix. It works for me anyways. 

In the CKB Interface/GUI:

Click on your keyboard, then Performance, at the bottom use the dropdown menu next to CAPS LOCK and make it say: ALWAYS OFF. so far its been working for me. Hope it helps. :D

clodal, 2017-02-07:
Having the same issue with a K95 RGB on an iMac (2015) running macOS Sierra 10.12.2. Implemented the workaround suggested by @JarrenJ to get things going for now.

klaus385, 2017-02-25:
@JarrenJ thanks for the fix seems to allow my computer to function again :)

light2yellow, 2017-03-07:
@auibrian could you look at this issue as well?

auibrian, 2017-03-09:
@light2yellow Sure I can look into this. I've only been able to reproduce it sporadically. It actually seems to be somewhat related to what order the GUI/daemon are started in. (While developing I typically would fire up the daemon first, but if I started the GUI first I usually could produce this problem.) It also seems to be fixed by unplugging the keyboard and plugging it back in again. I'll update as I get more info.

bmsimons, 2017-03-21:
I am on macOS 10.12.3 and am using a Corsair Strafe keyboard (non-RGB). Same issue over here, as soon as I press caps lock, my system completely freezes. When I press caps lock again, it doesn't turn off caps lock and my system continues to freeze at random events like you described earlier. 

However, when I hold the caps lock button and press another button (for example the A key) it unfreezes my system and turns off caps lock. I hope that this information is useful for debugging purposes.

The solution mentioned by @JarrenJ works for me though. Thanks for that!

FkBoyLeader, 2017-03-22:
I'm also having this issue on macOS Sierra on my RGB strafe, the workaround JarrenJ mentioned has worked beautifully! Hopefully it gets patched.

contreras2004, 2017-05-04:
@JarrenJ Saved my life, this was so annoying... Had to restart my mac every time I accidentally press the caps lock button. It it's useful for anybody, it also works if you change it to "RGB Indicator". At least with K70 LUX under MAC OS Sierra 10.12.3 beta (16D17a)

[image]

tatokis, 2017-05-04:
Odd, if it works with "RGB indicator", then it's probably not the code that probes the OS for the caps lock status.
@contreras2004 Does the actual caps lock key (in RGB Indicator mode), light on and off when you press it?

auibrian, 2017-05-04:
I have not looked at this extensively but as far as I had gotten I was pretty sure the problem lay in ckb and not ckb-daemon.

contreras2004, 2017-05-04:
@tatokis Yep, it actually does.

tatokis, 2017-05-04:
I propose we do an ifdef mac and force RGB Indicator by default until we can figure out what breaks this.

schteeben95, 2017-05-06:
I tried to switch it to "Always Off" but it still freezes the computer. K95 here

insideouting2, 2017-05-11:
I tried to switch it to "Always Off" or  " RGB indicator" the system starts to work fine until restarting the computer again fails.
Entering in CKB I see that it continues to be "always on", but the system continues to freeze.

miagg, 2017-05-15:
Same problem here. K95 RGB with v0.2.8 on Sierra 10.12.4. Tried both "Always off" and "RGB Indicator" fixes and it still freezes my system whenever I hit caps-lock for 2-3 seconds. It seems happen randomly after each restart.

Mordegay, 2017-07-30:
I have this issue with my K65 LUX RGB (UK layout) and macOS Sierra 10.12.6 (driver version 0.2.8, both pre-compiled .pkg and self-compiled). I also use a SABRE RGB mouse.
Setting the Caps Lock indicator to always off or LED indicator does not help - freezes still occur in these two modes. Whenever Caps Lock is toggled _**on**_, the freezing occurs and the system gets unstable. Even _clicking something_ while caps lock was toggled on causes the system to freeze. I also used a wireless Logitech keyboard, and toggling Caps Lock on that would also cause the system to freeze while ckb is running.

Mordegay, 2017-08-01:
So here's my findings on this issue:
1. When pressing the middle mouse button (below the mouse wheel) on a Sabre RGB and loading ckb-next _without pressing Caps Lock_, things seem to work fine. Possibly as long as I don't press Caps Lock while getting ckb-next's GUI running, things will work fine?
2. Quitting ckb-next's GUI _after_ getting a working Caps Lock indicator will cause the usual crashing. Does this mean it's a bug in the daemon?

I'LL HAVE TO CUT THIS COMMENT SHORT, SINCE SETTING IT TO NORMAL WILL CAUSE THE OS TO FREEZE. SET IT TO ALWAYS ON/OFF IN CKB-NEXT AND RESTART.

corpnewt, 2017-09-14:
Same issue here on 10.12.6 with a Corsair K70 RGB - running on:

* Asus X79 Sabertooth
* i7-4930k
* 32GB Corsair Vengeance
* GTX 1080ti

Any touching of the caps-lock key causes intermittent freezing until a reboot.  If anyone's got any leads or made any progress on this and needs testers, I can recreate the problem very consistently.

Handsomfreddy, 2017-11-10:
Had the same problem.  Computer locking up intermittently upon press of caps lock.  Thanks to JarrenJ for tip - switched to "RGB Indicator" apparently has done the trick.  Will update if the problem shows up again! Thanks!

tatokis, 2018-01-21:
https://github.com/ckb-next/ckb-next/issues/1

closed 2018-01-21 by tatokis