mattanger/ckb-next #26: Version file points to ccMSC github

frickler24, 2017-01-15:
If new versions may be downloaded and installed, src/ckb/kbfirmware.cpp references a file in github which belongs to ccMSC (line 35):
`(tableDownload = networkManager->get(QNetworkRequest(QUrl("https://raw.githubusercontent.com/ccMSC/ckb/master/FIRMWARE")));`
That file is signed by him via GnuPG and needs an update for the M65RGB at least.

The incidents looks as follows:
My current FW is 1.19 for the M65RGB, but corsair has a new one (2.04).
[image]
So I get a hint in ckb to upgrade, but that hint is wrong: Because in the file mentioned the latest version is 2.02, it tries to update to this version, but gets 2.04 from corsair.

The manual activation is faulty also, but I don't know why yet.

mattanger, 2017-01-18:
I'm aware of this issue, and I just haven't had a chance to do anything about it yet. I'm going to do work on addressing this shortly.

frickler24, 2017-01-18:
Can I help you? 
If  we create a new pub/priv Key pair just for this purpose, more than one person is able to update the info.

light2yellow, 2017-01-26:
@frickler24 this is definitely an important issue, if you come up with an extensible solution (like the proposed one), I guess @mattanger won't be against.

P. S. thanks for the hardware profile fixes, much appreciated.

mattanger, 2017-01-28:
I'm not opposed to it.

frickler24, 2017-01-29:
Does anyone have the following HW + firmware running with ckb 0.2.7? 
Firmware for Scimitar is a 2.4 available, for Strafe 2.05 (non-RGB)?

The last entries for these two are less than the current downloads.
Problem is, if I put the current links into the FW-Checker and that combination is not running with ckb, we could stop working some equipment.

light2yellow, 2017-01-30:
@frickler24 I have a Strafe but an RGB one.

frickler24, 2017-02-02:
ToDo: Refer the K95 Platinum as a new HW in the file

mattanger, 2017-02-04:
@frickler24 What are you working on in regards to resolving this issue? I was planning on signing a new firmware file for now, but if you've got a nice solution in the works I don't want cause you an issue.

frickler24, 2017-02-04:
@mattanger  I'm just preparing the pullreq.

mattanger, 2017-02-04:
Awesome. Haven't had a chance to check it yet, but I will tomorrow.

mattanger, 2017-02-08:
Merged #60

closed 2017-02-08 by mattanger