mattanger/ckb-next #27: All files should end with a newline

light2yellow, 2017-01-16:
I suggest that all files in the repository should end with a newline. Rationale: [here](https://stackoverflow.com/questions/729692/why-should-text-files-end-with-a-newline).

mattanger, 2017-01-18:
I'm checking everything in master right now.

light2yellow, 2017-01-18:
I've written a python script to fix this:
```python
#!/usr/bin/env python3

from pathlib import Path

def fix_newline(p):
    for entry in p.iterdir():
        if entry.is_dir():
            fix_newline(entry)
        else:
            try:
                with open(str(entry), 'r') as f:
                    lines = f.read()
                    if lines and lines[-1] != '\n':
                        with open(str(entry), 'a') as fi:
                            fi.write("\n")
            except UnicodeDecodeError:
                pass

if __name__ == '__main__':
    fix_newline(Path())
```

fleischie, 2017-01-18:
@light2yellow This is a valid one-time fix maybe. I guess the more prominent problem is, that each editor can decide whether to append new-lines at the end of a file.

This can be an OS- or editor-specific behaviour (I only use vim and it does this by default) and has to be checked each time changes are made to the sources. So this must be checked by each developer on each commit, they make. (or we install a git-bot, that does this with each push, but let's not get carried away here.) :wink:

mattanger, 2017-01-18:
I did some bashing, and the only file I found missing a newline was VERSION. We make a define with the contents of that file. I haven't checked to see if adding a new line to it would do anything.

light2yellow, 2017-01-18:
Yep, you're right. If i do this:
```python
#!/usr/bin/env python3

from pathlib import Path

def fix_newline(p):
    for entry in p.iterdir():
        if entry.is_dir():
            fix_newline(entry)
        else:
            try:
                with open(str(entry), 'r') as f:
                    lines = f.read()
                    if lines and lines[-1] != '\n':
                        print("Missing newline at {}".format(str(entry)))
                        # with open(str(entry), 'a') as fi:
                        #     fi.write("\n")
            except UnicodeDecodeError:
                pass

if __name__ == '__main__':
    counter = 0
    fix_newline(Path())
```
I get this:
```
Missing newline at osx/distribution.xml
Missing newline at VERSION
```
So it can be fixed manually and that'll probably be it.

> I haven't checked to see if adding a new line to it would do anything.

Then it'll be possible to strip it. By the way with cmake I'd rather define the version in it and then generate the VERSION file.

mattanger, 2017-01-18:
Yep. Agreed on cmake version definition. I've got another version in the
pipe that I'll hopefully have ready for testing today. I'm going to remove
the version file in favor of cmake.

light2yellow, 2017-01-18:
@mattanger I'm also working on cmake in macos direction. Maybe open an issue to track the progress?

mattanger, 2017-01-18:
I like it. Opened one: https://github.com/mattanger/ckb-next/issues/32

mattanger, 2017-01-19:
Closing since we're all agreed. Note will be made in #32 about moving the version definition into the cmake build scripts.

closed 2017-01-19 by mattanger

tatokis, 2017-01-19:
I'm still confused, but wouldn't moving the version definition to cmake break qmake.

mattanger, 2017-01-19:
Sorry, see #32 about generating with CMake. I got a little hasty closing, according to @light2yellow you've discussed leaving it/generating for qmake support.