mattanger/ckb-next #29: Shift Select or Cmd Select not Working on macOS Sierra

smachi, 2017-01-17:
This issue comes from the original CKB driver but still present on this fork:

https://github.com/ccMSC/ckb/issues/534

sinetek, 2017-01-18:
bump pressing issue

rag3trey, 2017-01-19:
I would also like this bug fixed. from the looks of it, only Corsair brand mice are synergizing well with the Corsair keyboard. So, until this bug is fixed I am having to use my Corsair Sabre instead of my trusty Razer Deathadder.

frickler24, 2017-01-19:
sorry that I cannot help on this issue. I don't have mac and on linux your scenario is running normally.

sinetek, 2017-01-20:
It works with corsair mice?
Would like to confirm,  I'll think about picking one up in that case.

dtrang, 2017-01-21:
I am having this problem too running macOS Sierra. As an avid IntelliJ and Chrome user not being able to use Cmd + Click or Shift Click bugs the hell out of me. The reasons for me to buy this keyboard were:

- To test drive the Cherry MX Speed switch: it felt a bit weird at first since I've been using my Das 4 Ultimate Cherry Brown for the past few years. It took a few hours to get used to but now everything is honky dory and I actually enjoy using this keyboard. The build quality is indisputable. 
- For my kid to play with the color effects! They absolutely loves them so it actually worths the trade-off of not having Cmd + Click working properly. A big big thank you to all the awesome people that made 'ckb' and 'ckb-next' (I had to build ckb-next for my K70 Rapidfire RGB since I couldn't find the binary build).

However my kid only plays with this for maybe 15' max every day and I need to do some side projects at home for 2+ hours everyday on even more hours on the weekend! So again having Cmd + Click while retaining all the animated color effect is a must for me. I think I just found out a work around for the best of both requirements. 

- Enable the driver when my kid wants to play: sudo launchctl load -w /Library/LaunchDaemons/com.ckb.daemon.plist
- Disable the driver when I need to work: sudo launchctl unload /Library/LaunchDaemons/com.ckb.daemon.plist. All the modifier keys are back to work normally. 

I've tested this a few times and it's worked perfectly without having to restart my Mac.

Btw for those who want a quick instruction to build ckb-next from command line, this is what I did:
- Ensure you have Xcode and Xcode CLI tools installed
- Get brew
- Install qt5 from brew: brew install qt5
- Run ./quickinstall from ckb-next
- Choose Yes to install ckb system wide and driver system wide

AlexanderXu2008, 2017-01-22:
I have the same issue with K65 RGB && Sierra
I don't konw how to fix it.
It will go bad with load /Library/LaunchDaemons/com.ckb.daemon.plist ,If I load this file ,my k65 can not use cmd and shift to select muti files well. That's so ......strange..
Please tell me What should I do ?

adal-io, 2017-01-22:
Have been watching this issue as I am experiencing the same bug (as well as the "hit caps lock too many times and everything crashes" bug). However this morning, it appears to have gotten _worse_ - previously, I could not use the modifier hold key combos (shift/cmd + select, cmd multi-line in Atom, etc.) but could still use them for basic modifier functions (cmd + a, cmd + w, cmd + n, etc). Now, the modifier keys don't have any effect at all.

mattanger, 2017-01-23:
Just to make everyone aware, Sierra support is, at best, very poor. We're working on a solution, but Apple deprecated APIs used by ckb-next in Sierra and we have not found a good solution yet. This post: http://dae.me/blog/2144/on-macos-sierra-support-and-the-future-of-smoothmouse/ has some relevant information regarding some of these issues. 

MacOS hackers/experts/wizards are invited to contribute. Those of use who are working on this project currently are not MacOS experts.

fleischie, 2017-01-24:
I just tested my K70 RGB out on the latest master version. I bound the keys in the following way:

- Ctrl -> Control
- Windows -> Option
- Alt -> Command

In the terminal it works as expected, a non-corsair and a corsair-but-non-ckb-trapped keyboard works the default way (ctrl -> control, windows -> command, alt -> option).
In the finder the shift/cmd + select does not work with the K70 but as expected with the non-corsair and corsair-but-not-ckb-trapped keyboard.

I researched a little and discovered, that we cannot naively replace the Carbon API with the Cocoa API, the former is the legacy C API, the latter is the newer Objective-C API. Thus, if we want to update the Daemon process, we need to split the source code into two OS-specific sources ... OR ... use a more high-level abstraction library of which I don't know yet, which. ... OR ... we remove the rebinding of the keys, which should be handled by other software (maybe the Karabiner-rewrite).

I personally am in favor of removing this, as it adds another level of maintaining- as well as OS-specific problems. If you have differing opinions/reasoning, I would like to hear it.

ssvenn, 2017-01-28:
If the modifier key remapping functionality is what breaks ctrl-clicking in Sierra I'm in favour of just stripping out that functionality too on the macOS platform. I think most people installing ckb are just interested in the RGB light effects anyway.

sinetek, 2017-01-29:
, we can remap the control and option keys in the settings already.
+1 to that

fleischie, 2017-01-30:
After some more research I found out, that the problem with mac-specific keyboard drivers is more severe, that at first anticipated:

- Carbon is the C Mac OS API and
- Cocoa is an Objective-C Mac OS API.

This means, that for utilizing the Cocoa API, we have to write a separate daemon process to handle the keyboard functionality.

Apart from that, I will try to remove the Mac OS-specifc remapping-functionality for Sierra and above in a separate branch in the next few weeks. (Disclaimer: I say the next weeks, as I am not sure, how long I can work on this each day.)

tatokis, 2017-01-30:
@fleischie You probably didn't see my messages on IRC, but to sum it up 
I said that there are keyboards that do not work out of the box on 
macOS, so by removing ckb-next's ability to do this, then these 
keyboards will still not work on anything other than BIOS mode. #31 is 
a good example.

sinetek, 2017-01-30:
I'm willing to work on this if someone points me in the right direction with what's needed.
AFAIK my keyboard is not affected by #31 (non RGB STRAFE), so we should keep this and not merge the issues together.

fleischie, 2017-01-30:
@sinetek Yes, I discussed this further with @tatokis and @light2yellow in the IRC chat, you're welcome to chat with us there, as well.

The main problem it seems is that Mac OS is awaiting an explicit setting of the appropriate IO Option Bit/Flag, when using a command modifier (Cmd/Shift/Alt/Option/etc.). Thus your best bet is to check out [input_mac.c](https://github.com/mattanger/ckb-next/blob/master/src/ckb-daemon/input_mac.c).

More specifically, `os_keypress` is used to handle all queued keypresses, which calls `postevent_kp`, which also calls `postevent`. Somewhere here there must be checked, where the key event is generated and therefore correctly initialized with the appropriate keyboard option flags. (The most obvious ones should be `NX_SHIFTMASK` and `NX_CONTROLMASK`.)

Then I am puzzled as how to fix the additional selection functionality. Maybe also check [Karabiner-Elements](https://github.com/tekezo/Karabiner-Elements/blob/master/src/core/grabber/include/manipulator/modifier_flag_manager.hpp) how they are implementing this (and whether this works as intended).

If you need further information and/or discussion, you can always ask or come into the IRC chat and chat with us. :smile:

daveedjackson, 2017-01-30:
My issue with this is fairly recent. Not had the issue before with the modifier and click. But can't use shift/control to highlight files, or layers in software like photoshop, after effects etc or system wide for that matter.

BTW i'm using a standard wired Apple keyboard. UK and US version - both have the same problem. Shift to rescale items (keeping proportions) anything requiring a shift/control shortcut no longer works. Wonder if anyone has any thoughts. There have been multiple updates but still present... Seems odd that this wouldn't be fixes as a matter of urgency?
Thanks in advance.
D

light2yellow, 2017-01-30:
@daveedjackson this is strange, Apple keyboard must work. The only issue related to a few keys was [found and fixed](http://blog.edovia.com/2016/10/24/macos-sierra-and-broken-keyboard-shortcuts/) in sierra .2. Do you use Karabiner? If yes, it is no longer suitable, use Karabiner-elements instead.

daveedjackson, 2017-01-30:
@light2yellow Two keyboards don't - have literally tried everything - it's very annoying - and very strange. No i don't use Karabiner...

light2yellow, 2017-01-30:
@daveedjackson [that's what google tells me](https://apple.stackexchange.com/questions/254140/my-usb-apple-keyboard-stopped-working-after-upgrading-to-sierra).

mattanger, 2017-01-30:
@daveedjackson Not that we don't want to help you out, but the issues in the this thread are related to corsair keyboards and ckb-next.

fleischie, 2017-02-05:
Just for reference, I am currently trying to read the apple system's documentation:

The specified issue is referenced as [Keyboard Interface Control](https://developer.apple.com/library/content/documentation/Cocoa/Conceptual/EventOverview/HandlingKeyEvents/HandlingKeyEvents.html#//apple_ref/doc/uid/10000060i-CH7-SW9), which - if I understand correctly - has something to do with how Apple's Windowing System handles keyboard input delegation.

The main problem still seems to be, how the daemon is handling the low-level usb-input-events and how Cocoa-based applications handle high-level framework specific input-events.

I am starting to get the feeling, that Apple is expecting the low-level functionality, which is happening in the daemon to be implemented in the high-level (Cocoa) portion of the app (which is used as the GUI as well as user input handling). I am contemplating a possibility to write the daemon separately as a C++ program, that is linked against the Cocoa framework, utilizing the "high-level" portion of the key-inputs. But I am still trying to figure out, how the daemon does and the Apple expects to work. 🤷

@sinetek are you still interested in this issue?

sinetek, 2017-02-06:
@fleischie Yes, I am, yes. I'll need to read the documentation first and get a handle on what you said.

max-l-weaver, 2017-02-06:
Just FYI shift-select works fine when ckb-next is removed. Not sure if anyone has tried this yet? As nice as it is to have the RGB functionality of this keyboard, and as much as I appreciate everything the developers are doing, it's gonna have to remain uninstalled until this vital feature is repaired.

clodal, 2017-02-07:
Hi guys, could we label this issue with the macOs label and also include this as a known issue in the README for macOS users? This would be really helpful. Thanks!

AlexanderXu2008, 2017-02-16:
After a period of use, I found that, once loaded the com.ckb.daemon.plist file, I can't use CMD and shift - select, even if I don't start CKB. App, if delete the com.ckb.daemon.plist files, my keyboard can use as normal, but the default configuration of lights I can use only.(K65 RGB LUX,MacOS 10.12)

auibrian, 2017-03-06:
I've been digging into this a little, here's what I've found so far:
* The modifier keys are being passed "correctly" to the system
* The modifier keys "work" under certain circumstances.
* One of those circumstances is not moving the pointer.
* The issue seems to be one of the flags being lost when events come in from different devices.

You can test what I've said above in an app like Xcode like so:
1. Connect an apple kb (or your corsair kb w/ the ckb daemon not running)
2. On the apple kb hold down the alt-key and observe the cursor changes to a cross hair (if you're over a symbol the cursor changes to a '?')
3. Move the mouse, observe the cursor stays as a cross hair.

Repeat the above steps with the corsair daemon running and a corsair kb attached. When you get to step 3 you will find that the moment the mouse moves, the cursor reverts to the text caret and the modifier key is lost.

Similarly if you use finder with keyboard only and make sure not to move the mouse you'll see you can shift-select files.

If @rag3trey would be able to confirm that ckb also controls corsair mice then that would help me to confirm what I'm seeing here because it would mean that input from both kb and mouse were being routed through the same USB handler (ckb/ckb-next) and that would result in the flags being set properly again.

Unfortunately the documentation is a bit light so my current status is: trying to find out how not to lose the modifiers on pointer movement.

So far I've tried not resetting last key when dealing with modifiers but that just hoses the whole system.

Anyway, I'll keep plugging at it when I have time, I can't really use my keyboard w/ the daemon without this fixed so I have a personal investment. Happy to team up with anyone who's got more experience with IOKit.

light2yellow, 2017-03-06:
@auibrian thanks for the investigation! You can also look at Karabiner-elements, how they solved this. They have `DEVELOPMENT.md` there where there are some of the changes in a nutshell.

auibrian, 2017-03-07:
@light2yellow Good pointer. Thanks.

I have what appears to be a fix at the moment, but the code is not phenomenal. I'll see what I can do about cleaning it up and then PR.

sinetek, 2017-03-07:
awesome, thanks to you i might be able to upgrade to 10.12.

rag3trey, 2017-03-07:
@auibrian I can confirm with both my Corsair Sabre and Scimitar that they will both work in ckb.
I have even put together a video for you all. https://youtu.be/rTQAYg8qLyo

mrgaolei, 2017-03-13:
mark

sinetek, 2017-03-13:
Yep, that works for me. STRAFE on Sierra. Thank you

andreasgrill, 2017-03-13:
My K65 RGB Lux is now working on Sierra with bells and whistles! Thanks a lot!

rag3trey, 2017-03-19:
@auibrian I really appreciate all the hard work put in to this. Unfortunately, I just updated CKB-Next and this issue still resides. Not sure if I updated incorrectly or anything, but my Strafe does not want to recognize CMD/Shift clicks from my Steelseries Rival 300 and itself. Switching to a Corsair product seems to partially work though. I'll help in any way that I can. I'm on Mac OS X Sierra 10.12.3.

auibrian, 2017-03-19:
Hi @rag3trey, you're running the branch for my PR? Or develop / testing? My work has yet to be merged so unless you're running my PR branch the problem will persist.

rag3trey, 2017-03-19:
@auibrian i just realized where i goofed. let me try running the PR branch.

juaganmu, 2017-03-27:
@auibrian Hi, I installed CKB from your PR branch, and problems with cmd/shit+select where solved, but I restarted my Mac and I'm having the same problem once more. I'm running on Sierra 10.12.3.

mrgaolei, 2017-03-27:
@rag3trey Where is PR branch? Why can't I found it?

smachi, 2017-03-27:
I'm experiencing the same issue as mentioned @juaganmu. It worked just after the installation (a week ago) and the day after it stopped working and not worked anymore. With latest version of Sierra too.

auibrian, 2017-03-27:
@juaganmu @smachi Can I ask if you both installed the PR build or if you just ran the daemon after building it? The problem your describing sounds very much like you ran the daemon from the command line after building but never installed it. In this case the old release daemon would be what gets started after a reboot.

smachi, 2017-03-27:
@auibrian I downloaded your PR and removed manually the app and the daemon and restarted  before installing it.

I was very happy when I saw that cmd + click was working again after installing your PR (thing that didn't happen since I upgraded to Sierra), but at some point stopped working.

I didn't try to re-install the PR again.

auibrian, 2017-03-27:
@smachi Thanks. Sorry to be a pain but can you just confirm the output of `ls -l /Applications/ckb.app/Contents/Resources`? And also confirm the daemon is running?

smachi, 2017-03-27:
@auibrian The output is:

`total 480
drwxr-xr-x  9 root  admin     306 19 mar 09:37 ckb-animations
-rwxr-xr-x  1 root  admin  212156 19 mar 09:37 ckb-daemon
-rw-r--r--  1 root  admin   26827 19 mar 09:37 ckb-logo.icns
-rw-r--r--  1 root  admin       0 19 mar 09:37 empty.lproj
-rw-r--r--  1 root  admin      78 19 mar 09:37 qt.conf`

And

`root               128   2,3  0,1  2516660   9308   ??  Ss    8:28AM   0:39.95 /Applications/ckb.app/Contents/Resources/ckb-daemon
Salva             4073   0,0  0,0  2442020   2060 s000  S+    8:58AM   0:00.00 grep ckb-daemon`

All the other CKB features (light, binding, etc) are working normally, just the cmd/shift + click is not working now

auibrian, 2017-03-27:
@smachi Thanks. Very odd. I'm running 10.12.2. Let me see what happens if I upgrade. I'll need a day or two to explore this fully.

smachi, 2017-03-27:
@auibrian No problem. Thanks a lot for your help.

light2yellow, 2017-03-27:
@mrgaolei you can't find it because this is an issue, the PR itself is #106.

When you visit that page, on the top of it you will see something like

[image]

You are interested in the __from__ part of it. For some reason GitHub doesn't allow easy navigation by click here, so here's what I would have done:

* go to __aubrian__'s fork page: https://github.com/auibrian/ckb-next
* download the branch you saw in the __from__ section:
```sh
git clone https://github.com/auibrian/ckb-next.git --branch issue29 --single-branch aubrian-ckb-next
```
* `cd` and build (like README suggests):
```sh
cd aubrian-ckb-next
./quickinstall
```

SeNeO, 2017-03-27:
I've got the same issue that @smachi  and @juaganmu mentions. After each reboot of my iMac the modifier keys doesn't work anymore. I need to go to the activity monitor to kill the daemon, and after it restart the modifier keys works again.

auibrian, 2017-03-27:
I'm seeing something similar now that I've upgraded to 10.12.3 from 10.12.2. I will report back when I have found more.

rag3trey, 2017-03-27:
@auibrian it is seeming like there is some sort of modifier inside ckb-daemon causing it to revert. My iPad is about to die or I would go through the code and see if I could find it.

auibrian, 2017-03-28:
I think I've gotten this figured out. I'm hoping to have a fix w/in 24 hours. Thanks for the feedback everyone!

mrgaolei, 2017-03-29:
@auibrian I have the same problem with @juaganmu @smachi
Shift + Click will failed when I restart computer.

rag3trey, 2017-04-02:
So, is there an update to the fix @auibrian ?

rag3trey, 2017-04-02:
@auibrian I seem to have discovered something. by running 
`sudo launchctl unload /Library/LaunchDaemons/com.ckb.daemon.plist`
and then 
`sudo launchctl load /Library/LaunchDaemons/com.ckb.daemon.plist`
the functionality comes back. 
This is the only sort of temporary fix I can find.

auibrian, 2017-04-02:
Hi @rag3trey,

Yes, I made a fix the other night for it but it's not ready to release yet. I'd like to explore another option. The short version of the issue is: when a user isn't logged on we can't create the CGEventTap. If you waited until after login to plugin your keyboard after a restart you would. Or see this problem. Also as you have discovered a simple restart of the daemon will also route around the problem for the time being. I'm hoping to finish my second version of the fix this weekend.

rag3trey, 2017-04-02:
ah good news @auibrian 
Looking forward to it.

rag3trey, 2017-04-02:
also @auibrian  shouldn't it not really matter if it doesn't work on the login screen? You aren't trying to shift click or anything there. It may just need to start it with the client, rather than with the computer itself.

smachi, 2017-04-28:
Hi @auibrian,

Any news about the new fix you were working on?

fleischie, 2017-04-28:
I closed the appropriate PR right now and the changes are in the testing branch now.

Regarding the supplemental issue, please open a new separate issue. Thank you! :slightly_smiling_face:

closed 2017-04-28 by fleischie

porkserver, 2017-12-27:
Compiling from source works for me on High Sierra 10.13.2  The pre built binary is from May and from what I can surmise from comments does not implement the fix.  I just did the quickinstall as I already had the other prereqs installed or setup already for my own coding.