mattanger/ckb-next #30: Single Spot Heat Map Issue

slugger7, 2017-01-17:
Where as I type sometimes some keys will keep on doing the heat map effect on certain keys.
I had this issue with the newdev branch on the original CKB as well (i have the K70 LUX).
Is my settings by any chance wrong for the heatmap to work properly?
[Here](https://drive.google.com/open?id=0B2hUGYneQmLuTElGSHJGZWtHdXc) is a video demonstrating what is happening.
It does it on random keys at random intervals

tatokis, 2017-01-21:
If I understood correctly, that is the intended use of a heat map.
The point of it is to keep the most used keys lit.

If you just want the keys to light up as you type and then fade out, you can use Gradient -> Type lighting.

tatokis, 2017-01-21:
Apologies, I set a black background and I can see the keys randomly starting to light up for no reason.
I'll have a look in the code and see what is going on.

tatokis, 2017-01-22:
Please try the latest HEAD from the testing branch.
The issue should be fixed.

slugger7, 2017-01-24:
Just got time to get to testing it. Will keep you posted if I find any issues with the type lighting

slugger7, 2017-01-29:
@tatokis it seems to have fixed the problem. Thank you very much

tatokis, 2017-01-29:
Awesome!

commit d297c476 (2017-01-22): ckb-heat: Usage count can not be negative. Fixes #30
closed 2017-01-29 by tatokis