mattanger/ckb-next #31: K65 Non-RGB is not detected

mmolad, 2017-01-17:
Using beta .27, I am unable to detect my K65 Non RGB in macOS.

tatokis, 2017-01-20:
Alright, make sure your keyboard is plugged in and the switch is not in the BIOS mode.
On a terminal, run
```
sudo launchctl unload /Library/LaunchDaemons/com.ckb.daemon.plist && sudo killall -9 ckb-daemon && sudo /Applications/ckb.app/Contents/Resources/ckb-daemon
```
all in a single line, and then paste the output of the command here.

mmolad, 2017-01-20:
`No matching processes were found`

tatokis, 2017-01-20:
Oops, try this
```
sudo launchctl unload /Library/LaunchDaemons/com.ckb.daemon.plist && sudo killall -9 ckb-daemon ; sudo /Applications/ckb.app/Contents/Resources/ckb-daemon
```

mmolad, 2017-01-20:
`/Library/LaunchDaemons/com.ckb.daemon.plist: Could not find specified service
    ckb: Corsair RGB driver beta-v0.2.7
[I] Root controller ready at /var/run/ckb0
[I] Connecting Corsair K65 Gaming Keyboard at /var/run/ckb1
[W] add_hid (usb_mac.c:679): Got unknown handle (I: 2, O: 0, F: 0)
[W] add_hid (usb_mac.c:679): Got unknown handle (I: 8, O: 1, F: 8)`

Changed from non-BIOS to BIOS...
`[I] Disconnecting /var/run/ckb1
[I] Removed device path /var/run/ckb1
[I] Removed device path /var/run/ckb1
[W] add_hid (usb_mac.c:705): Tried to set up ifhid[2] for device ckb1, but it was already set up. Skipping...
[W] add_hid (usb_mac.c:679): Got unknown handle (I: 2, O: 0, F: 0)
[W] add_hid (usb_mac.c:679): Got unknown handle (I: 8, O: 1, F: 8)`

tatokis, 2017-01-20:
Interesting, what firmware version is it running?

mmolad, 2017-01-20:
It's fresh out of the box.

tatokis, 2017-01-20:
Since I don't have the keyboard, you'll have to modify the daemon to provide more info.
Open up src/ckb-daemon/usb_mac.c and on line 682 add 
```ckb_info("Got handle (I: %d, O: %d, F: %d)\n", (int)input, (int)output, (int)feature);```
save and re-run quickinstall.
Once it installs, run
```sudo launchctl unload /Library/LaunchDaemons/com.ckb.daemon.plist && sudo killall -9 ckb-daemon ; sudo /Applications/ckb.app/Contents/Resources/ckb-daemon```
and attach the output again.

mmolad, 2017-01-23:
```
    ckb: Corsair RGB driver beta-v0.2.7
[I] Root controller ready at /var/run/ckb0
[I] Connecting Corsair K65 Gaming Keyboard at /var/run/ckb1
[I] Got handle (I: 15, O: 0, F: 0)
[W] add_hid (usb_mac.c:679): Got unknown handle (I: 2, O: 0, F: 0)
[W] add_hid (usb_mac.c:679): Got unknown handle (I: 8, O: 1, F: 8)
```

Switching back to BIOS...

```
[I] Disconnecting /var/run/ckb1
[I] Removed device path /var/run/ckb1
[I] Removed device path /var/run/ckb1
[I] Got handle (I: 15, O: 0, F: 0)
[W] add_hid (usb_mac.c:706): Tried to set up ifhid[2] for device ckb1, but it was already set up. Skipping...
[W] add_hid (usb_mac.c:679): Got unknown handle (I: 2, O: 0, F: 0)
[W] add_hid (usb_mac.c:679): Got unknown handle (I: 8, O: 1, F: 8)
```

tatokis, 2017-01-26:
Interesting, it seems to expose a handle that is used by RGB keyboards. (At least according to comments in the code, anyway.)

I made a patch, by attempting to guess which handle does what.
Please apply it, recompile, and if it is detected test both the normal and the media keys.

Should be as easy as downloading it to the root of the repo (where quickinstall is) and then running from a terminal ```patch -p1 < k65nrgbtest.patch```

https://vps.tasossah.com/bugs/k65nrgbtest.patch

mmolad, 2017-01-26:
Patch reports `patch: **** strip count l is not a number`

tatokis, 2017-01-26:
After -p, It's the number one, not the letter l.

mmolad, 2017-01-26:
Yup, just caught that.  Retrying now!

mmolad, 2017-01-26:
It looks like itis dedteecint . igg t now but with major keyboard/keyprpess issuess andd duplicate keypreesses.

mmolad, 2017-01-26:
How do I undo? 😢

tatokis, 2017-01-26:
```sudo killall -9 ckb-daemon```
should do it

mmolad, 2017-01-26:
That doesn't appear to resolve the issue.

Now that I'm on my backup keyboard... in BIOS mode the keys have very erratic behavior (demonstrated above).  In normal modes, the keys are completely nonfunctional (before only some keys didn't work, like Enter).

tatokis, 2017-01-26:
Well, since I don't know how to kill ckb-daemon on macOS, what you can (and should do, in order to do further testing) is run ```patch -p1 -R < k65nrgbtest.patch``` and recompile.

That should revert the patch and return the file to its previous state.

mmolad, 2017-01-26:
Will revert for now.  Was hopeful we had it figured out. :(

tatokis, 2017-01-26:
Another guess, how about https://vps.tasossah.com/bugs/ckb-next/k65nrgbtestv2.patch
Same procedure but different filename on the patch command.

light2yellow, 2017-01-26:
To stop the daemon you write:
```
sudo launchctl unload /Library/LaunchDaemons/com.ckb.daemon.plist
```
To start the daemon you write:
```
sudo launchctl load /Library/LaunchDaemons/com.ckb.daemon.plist
```

mmolad, 2017-01-27:
With patch v2, the keyboard continues to function in BIOS mode with no apparent issues (outside some latency issues as before), but fails to be detected.

In standard modes, it is detected but non functional or the key appears to be stuck (I saw no response and then the spacebar was infinitely repeated).

Wertuoz, 2017-01-30:
Hi 2all!
I have the same problem, my k65 rgb is No devices connected.
Sometimes it works fine, but after reboot - i spend a lot of time to make it work(

frickler24, 2017-01-30:
@Wertuoz ist anything more than the line ending with /var/run/ckb0?
can you pls look into that directory: There should be 3 files: connected, pid, version.
Interesting is the content of connectet. You can cat or open it in a viewer / texteditor.

tatokis, 2017-01-30:
@Wertuoz @frickler24 this thread is about the Non RGB version of the keyboard.
As this is clearly a different problem, please run the ckb-dev-detect script from this repository and open a new issue.

mmolad, 2017-02-06:
@tatokis any update on developing a new solution?

tatokis, 2017-02-06:
Currently waiting for macports to fix lsusb, so that I can gather some necessary info. 

Haven't forgotten about this :)

light2yellow, 2017-02-19:
@mmload
As soon as this https://github.com/macports/macports-ports/pull/284 is merged we can see if it is possible to gather required information.

tatokis, 2017-02-26:
@mmolad Can you please try the latest code from testing?

light2yellow, 2017-03-09:
@mmolad any news?

Unfortunately, the ported `lsusb` on Mac just uses the default Mac utility to examine usb devices. In particular, `lsusb -v` is an alias for `system_profiler SPUSBDataType`.

However, @tatokis & @fleischie, I got back to the USB Prober. And it actually worked for me on 10.12.3 (by using [this](https://superuser.com/a/814339) guide).

On the screenshot below one can clearly see the endpoints:

[image]

Interface # 2 HID Descriptor unwrapped:

[image]

And the full report:
[USB Bus Probe.txt](https://github.com/mattanger/ckb-next/files/831416/USB.Bus.Probe.txt)

Is that what we were looking for on Mac?

mmolad, 2017-03-09:
@light2yellow sorry for the delay!

Inexplicably, without any update that I'm aware of to any software, the standard (non-BIOS) mode on the keyboard started working without any apparent issues.

I can update the CKB-testing program I have and see if there's anything obvious or if it isn't detected properly, but I'm not sure functionality is an accurate test anymore.

tatokis, 2017-03-11:
@mmolad Would you be willing to try the latest code from testing and see if it works? If not, we can close this issue.

@light2yellow Yes, thank you. This is exactly what I wanted in the first place, however I read that USB Prober and its kernel modules broke computers, so I didn't actually try it.

Glad to see that it actually works though, meaning we can easily gather detailed information as needed in the future.

commit 9b234127 (2017-02-26): Now this is a PEBCAK, forgot to mark the keyboard as NRGB. Chances are, this fixes #31
closed 2017-04-03 by tatokis