mattanger/ckb-next #33: Standalone operation

chrylis, 2017-01-18:
As far as I can tell, commands sent to the `/dev/input/ckb1/cmd` pipe are executed only if the CKB desktop application is actually running--they're ignored at all other times. Even if I open up the CKB GUI and suspend the process, CLI commands suddenly start working when they were ignored before.

It seems the daemon is performing some overeager check and not proceeding unless it has what it considers a stable connection to the GUI. Instead, it should dumbly accept line input from the pipe and execute the command provided.

I am using an M65 Pro mouse on Arch.

frickler24, 2017-01-19:
@chrylis sorry but that's not correct (at least on linux). 
But you must read the protocol information in daemon.md carefully. 
If you open ckb client, then most of opening protocol handling is done for you. Adding a cmd statement is more easy then.
Please have a look at the phpCorsair thread, with a code example to program the daemon w/o the client. First cmd-statement in the script were
`echo exec("echo hwsave > /dev/input/ckb1/cmd");
echo exec("echo idle > /dev/input/ckb1/cmd");
usleep(20000);
echo exec("echo hwload > /dev/input/ckb1/cmd");
echo exec("echo active > /dev/input/ckb1/cmd");
echo exec("echo fps 60 > /dev/input/ckb1/cmd");
`
And so on, all running w/o ckb.

frickler24, 2017-01-22:
In the last days I did some debugging on the daemon and I found the reason, how ckb closes the connection: When ckb client is terminating, it send an "idle" cmd to the pipe.
You must reactivate it by sending an "activa" again (you may see it in the perl-example above). Afterwards it is interpreting your commands as you want. Plz have a deeper look at the DAEMON.md description.

closed 2017-01-23 by mattanger