mattanger/ckb-next #34: K65 RGB no device connected on 0.2.4 Mac OS 10.12.2

drk7891, 2017-01-20:
It seemed everything installed correctly as I had the application created however I get a no device detected error. I have tried all of the troubleshooting tips along with trying different USB ports but none have worked.

If there is any information I can provide to troubleshoot this please let me know and I will reply back with it.

Thanks!

Using version 0.2.4 on Mac OS 10.12.2 (Sierra)

tatokis, 2017-01-20:
You have to compile from source.
The binaries are completely out of date.

drk7891, 2017-01-20:
By compile from source you mean to manually build it using Qt Creator, correct?

When I do that I receive errors:

`In file included from device.c:1:
In file included from ./command.h:4:
In file included from ./includes.h:4:
./os.h:67:9: warning: 'CLOCK_MONOTONIC' macro redefined [-Wmacro-redefined]
#define CLOCK_MONOTONIC 1
        ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.12.sdk/usr/include/time.h:156:9: note: previous definition is here
#define CLOCK_MONOTONIC _CLOCK_MONOTONIC
        ^
In file included from device.c:1:
In file included from ./command.h:4:
In file included from ./includes.h:4:
./os.h:69:13: error: typedef redefinition with different types ('int' vs 'enum clockid_t')
typedef int clockid_t;
            ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.12.sdk/usr/include/time.h:171:3: note: previous definition is here
} clockid_t;
  ^
1 warning and 1 error generated.
make[2]: *** [debug/device.o] Error 1
make[1]: *** [debug] Error 2
make: *** [sub-src-ckb-daemon-make_first] Error 2
15:36:18: The process "/usr/bin/make" exited with code 2.
Error while building/deploying project ckb (kit: Desktop Qt 5.6.2 clang 64bit)
When executing step "Make"
15:36:18: Elapsed time: 00:01.`

tatokis, 2017-01-20:
Just run the quickinstall script that's bundled.

drk7891, 2017-01-20:
Yes, I run that but still come up with devices not found.

Alternatively, when I try to build from cmd I get:

> ./qmake-auto && make
cd src/ckb-daemon/ && ( test -e Makefile || /Users/derek/Qt/5.6/clang_64/bin/qmake /Users/derek/Desktop/ckb-next-0.2.4/src/ckb-daemon/ckb-daemon.pro -o Makefile ) && /Applications/Xcode.app/Contents/Developer/usr/bin/make -f Makefile
/Applications/Xcode.app/Contents/Developer/usr/bin/make -f Makefile.Debug
clang -c -std=gnu99 -Wno-unused-parameter -Werror=all -g -fPIC -DCKB_VERSION_STR=\"`cat /Users/derek/Desktop/ckb-next-0.2.4/src/ckb-daemon/../../VERSION`\" -I. -I../../../../Qt/5.6/clang_64/mkspecs/macx-clang -o debug/device.o device.c
In file included from device.c:1:
In file included from ./command.h:4:
In file included from ./includes.h:4:
./os.h:67:9: warning: 'CLOCK_MONOTONIC' macro redefined [-Wmacro-redefined]
#define CLOCK_MONOTONIC 1
        ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.12.sdk/usr/include/time.h:156:9: note:
      previous definition is here
#define CLOCK_MONOTONIC _CLOCK_MONOTONIC
        ^
In file included from device.c:1:
In file included from ./command.h:4:
In file included from ./includes.h:4:
./os.h:69:13: error: typedef redefinition with different types ('int' vs
      'enum clockid_t')
typedef int clockid_t;
            ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.12.sdk/usr/include/time.h:171:3: note:
      previous definition is here
} clockid_t;
  ^
1 warning and 1 error generated.
make[2]: *** [debug/device.o] Error 1
make[1]: *** [debug] Error 2
make: *** [sub-src-ckb-daemon-make_first] Error 2

tatokis, 2017-01-20:
"/os.h:69:13: error: typedef redefinition with different types ('int' 
vs 'enum clockid_t') typedef int clockid_t; ^ "
Let me guess, you are using the source from ckb and not ckb-next.

drk7891, 2017-01-20:
No I have downloaded it straight from the ckb-next repo

tatokis, 2017-01-20:
I thought this was fixed, interesting.
Can you check if you have the appropriate changes?
https://github.com/mattanger/ckb-next/commit/fa2b5b3293178ace1f02a3713d63bf665afa67d6

drk7891, 2017-01-20:
Interesting, I don't have those changes.

tatokis, 2017-01-20:
I see your problem, you downloaded the release source code.
What I meant is that you need to grab the latest git HEAD.
You can do so from here, https://github.com/mattanger/ckb-next/archive/master.zip

drk7891, 2017-01-20:
Ah thank you very much! That did the trick and is working great now :)

closed 2017-01-20 by drk7891

tatokis, 2017-01-20:
Glad you got it sorted out :)