mattanger/ckb-next #35: Keyboard Stopped Working when CKB is on

batchpert, 2017-01-21:
I connected my laptop up to my usb hub this morning and keyboard lit up as usual. However, i couldn't type, nothing was being registered. So I quit CKB keyboard went white like usual and reloaded it but still keys wouldn't be registered when typing. At that point i was running the ckb-beta v0.2.7 from ccMSC. I deleted that CKB and downloaded the master version from here. Once it installed keyboard lit up with original animations and seemed to work for a minute but then the key bindings stopped registering and i believe the whole keyboard stopped being registered then. So i quit CKB again restored it and now it doesn't register the keyboard as being connected.

Using Strafe RGB on firmware 2.05 and ckb-next beta-v0.2.7

Any help would be appreciated,

James

frickler24, 2017-01-21:
@batchpert two questions:
What OS do you use?
Did you unplug/plug the KB instead of terminating/restarting the ckb client? 

Currently I'm debugging some effects similar to yours. They come up sometimes when resuming from standby. When you unplug/replug the KB, ckb-daemon restarts the corresponding channel.

batchpert, 2017-01-21:
I am running  OS X El Capitan
I can't remember what i did first to be honest but i will play around with it in a bit and see if i can get it working again it functions as a keyboard but its not being detected anymore

batchpert, 2017-01-21:
Ok so i deleted and re-installed the same version of CKB. After maybe 10 seconds keyboard was detected and lit up with correct colours. But now the lighting effects are all working and 'it' or 'ckb' (I don't know enough about this stuff to give you a technical name) knows i am pressing a key because type lighting is working. But its not being registered on the computer. none of the keys actually  produce letters on the screen and the key bindings do not register other than caps lock.

batchpert, 2017-01-24:
Update

Restarted my laptop today because of a separate issue. restarted and ckb launched on start up and BOOM its all working fine with key bindings and everything :)

closed 2017-01-24 by batchpert

frickler24, 2017-01-25:
OK, then pls let us know in a few days if we can close the issue.
Good luck!