mattanger/ckb-next #38: SIGSEGV when deleting copied profile

frickler24, 2017-01-21:
This Issue is for me to remember. But if someone is faster than me :-)
Scenario is as follows:
1. open ckb
2. duplicate one of the profiles on the left (tested with KB K95RGB)
3. delete the **upper** one of the two siblings
4. crash.
Crash-protocol from gdb ist here:
[Absturz-beim-Profillöschen.txt](https://github.com/mattanger/ckb-next/files/721552/Absturz-beim-Profilloschen.txt)
It is reproducable an error (nil-ptr) somewhere in the button-handler (action).

tatokis, 2017-01-21:
While trying to reproduce yours, I stumbled upon another SIGSEGV.

All I did was go to "manage profiles" and then right click -> duplicate the first profile.
```
(gdb) bt
#0  0x00007ffff6a50de0 in qHash(QUuid const&, unsigned int) () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#1  0x0000000000451fdb in QHash<QUuid, AnimScript*>::findNode(QUuid const&, unsigned int*) const ()
#2  0x000000000044a7fb in AnimScript::copy(QObject*, QUuid const&) ()
#3  0x00000000004550b4 in KbAnim::KbAnim(QObject*, KeyMap const&, KbAnim const&) ()
#4  0x000000000044010e in KbLight::KbLight(KbMode*, KeyMap const&, KbLight const&) ()
[...]
```

frickler24, 2017-01-22:
@tatokis I tried a bit to produce your error but I couldn't produce it.
Is the first profile you try to duplicate marked as "saved to HW"? How many profiles do you have?
Anyhow, I'll have a look at it when debugging. Thanks!
@mattanger could you plz mark this issue as a bug?

frickler24, 2017-02-11:
@tatokis I've found the reason for the bug as I described it. 
Despite a lot of tests, I could not reproduce the behavior your mentioned above.
Do you have any more hints?

light2yellow, 2017-02-26:
@tatokis I wasn't able to reproduce it on Ubuntu 16.10 with different combinations (saved into HW, not saved etc.).
However, I managed to reproduce this one by @frickler24. Will try a patch from a new PR.

light2yellow, 2017-03-04:
I have no idea why it didn't close the issue, maybe it closes it only when the header contains a notice and not the body, or because I force pushed a few times (sorry).

commit 86d7cb34 (2017-03-04): merge #88

Merge branch 'frickler24-testing-issue-38-SIGSEGV_when_deleting_copied_profile' into testing
* fixes #38
PR #88 (2017-03-04): Fixes error #38 SIGSEGV when deleting copied profile
This error was caused by not completely copying a mode.
The KeyActions always referenced the original data and were not copied.
The commit fixes this and adds a few comments in the modules
where I passed while debugging.

This PR replaces #72, which was based on master.

Fixes #38.
commit f219fc89 (2017-03-04): merge #88

Merge branch
'frickler24-testing-issue-38-SIGSEGV_when_deleting_copied_profile'
into testing

* fixes #38
commit 7a13207d (2017-03-04): merge #88, fixes #38

Merge branch
'frickler24-testing-issue-38-SIGSEGV_when_deleting_copied_profile'
into testing
closed 2017-03-04 by light2yellow