mattanger/ckb-next #39: No USB Headset Mic when Deamon is runnning

guenniman, 2017-01-22:
Hallo there, 

I've stumbled upon a weird bug, whenever the ckb daemon is running Alsa/PulseAudio won't recognize my USB Headset Microphone correctly.
If i disable ckb and reboot the Microphone works just fine.

Headset: Corsair Void USB RGB
OS: Linux Mint 18 w/ kernel 4.9.0-040900-lowlatency

mattanger, 2017-01-23:
Strange indeed. It's possible the headset is somehow being caught by the daemon, but I can't really say for sure with out some more information. Maybe see what driver is being used when the ckb-daemon is running? If you've got usbutils package, `usb-devices` will tell you.

guenniman, 2017-01-27:
The output of "usb-devices" w/ daemon running is the following

`
T:  Bus=01 Lev=00 Prnt=00 Port=00 Cnt=00 Dev#=  1 Spd=480 MxCh= 2
D:  Ver= 2.00 Cls=09(hub  ) Sub=00 Prot=00 MxPS=64 #Cfgs=  1
P:  Vendor=1d6b ProdID=0002 Rev=04.09
S:  Manufacturer=Linux 4.9.0-040900-lowlatency ehci_hcd
S:  Product=EHCI Host Controller
S:  SerialNumber=0000:00:1a.0
C:  #Ifs= 1 Cfg#= 1 Atr=e0 MxPwr=0mA
I:  If#= 0 Alt= 0 #EPs= 1 Cls=09(hub  ) Sub=00 Prot=00 Driver=hub

T:  Bus=01 Lev=01 Prnt=01 Port=00 Cnt=01 Dev#=  2 Spd=480 MxCh= 6
D:  Ver= 2.00 Cls=09(hub  ) Sub=00 Prot=01 MxPS=64 #Cfgs=  1
P:  Vendor=8087 ProdID=0024 Rev=00.00
C:  #Ifs= 1 Cfg#= 1 Atr=e0 MxPwr=0mA
I:  If#= 0 Alt= 0 #EPs= 1 Cls=09(hub  ) Sub=00 Prot=00 Driver=hub

T:  Bus=02 Lev=00 Prnt=00 Port=00 Cnt=00 Dev#=  1 Spd=480 MxCh= 2
D:  Ver= 2.00 Cls=09(hub  ) Sub=00 Prot=00 MxPS=64 #Cfgs=  1
P:  Vendor=1d6b ProdID=0002 Rev=04.09
S:  Manufacturer=Linux 4.9.0-040900-lowlatency ehci_hcd
S:  Product=EHCI Host Controller
S:  SerialNumber=0000:00:1d.0
C:  #Ifs= 1 Cfg#= 1 Atr=e0 MxPwr=0mA
I:  If#= 0 Alt= 0 #EPs= 1 Cls=09(hub  ) Sub=00 Prot=00 Driver=hub

T:  Bus=02 Lev=01 Prnt=01 Port=00 Cnt=01 Dev#=  2 Spd=480 MxCh= 8
D:  Ver= 2.00 Cls=09(hub  ) Sub=00 Prot=01 MxPS=64 #Cfgs=  1
P:  Vendor=8087 ProdID=0024 Rev=00.00
C:  #Ifs= 1 Cfg#= 1 Atr=e0 MxPwr=0mA
I:  If#= 0 Alt= 0 #EPs= 1 Cls=09(hub  ) Sub=00 Prot=00 Driver=hub

T:  Bus=02 Lev=02 Prnt=02 Port=04 Cnt=01 Dev#=  3 Spd=12  MxCh= 0
D:  Ver= 2.00 Cls=00(>ifc ) Sub=00 Prot=00 MxPS=64 #Cfgs=  1
P:  Vendor=1b1c ProdID=1b1e Rev=02.04
S:  Manufacturer=Corsair
S:  Product=Corsair Gaming SCIMITAR RGB Mouse
S:  SerialNumber=07041020AEAA100255048DA3F5001943
C:  #Ifs= 3 Cfg#= 1 Atr=a0 MxPwr=300mA
I:  If#= 0 Alt= 0 #EPs= 1 Cls=03(HID  ) Sub=01 Prot=02 Driver=usbfs
I:  If#= 1 Alt= 0 #EPs= 1 Cls=03(HID  ) Sub=00 Prot=00 Driver=usbfs
I:  If#= 2 Alt= 0 #EPs= 1 Cls=03(HID  ) Sub=00 Prot=00 Driver=usbfs

T:  Bus=02 Lev=02 Prnt=02 Port=05 Cnt=02 Dev#=  4 Spd=12  MxCh= 0
D:  Ver= 2.00 Cls=00(>ifc ) Sub=00 Prot=00 MxPS=64 #Cfgs=  1
P:  Vendor=1b1c ProdID=1b20 Rev=02.05
S:  Manufacturer=Corsair
S:  Product=Corsair STRAFE RGB Gaming Keyboard
S:  SerialNumber=0202401EAEB8046755A7D4A8F5001947
C:  #Ifs= 3 Cfg#= 1 Atr=a0 MxPwr=500mA
I:  If#= 0 Alt= 0 #EPs= 1 Cls=03(HID  ) Sub=01 Prot=01 Driver=usbfs
I:  If#= 1 Alt= 0 #EPs= 1 Cls=03(HID  ) Sub=00 Prot=00 Driver=usbfs
I:  If#= 2 Alt= 0 #EPs= 1 Cls=03(HID  ) Sub=00 Prot=00 Driver=usbfs

T:  Bus=02 Lev=02 Prnt=02 Port=07 Cnt=03 Dev#=  5 Spd=12  MxCh= 0
D:  Ver= 2.00 Cls=00(>ifc ) Sub=00 Prot=00 MxPS=64 #Cfgs=  1
P:  Vendor=1b1c ProdID=1b2a Rev=00.06
S:  Manufacturer=Cosair
S:  Product=Corsair VOID RGB USB Gaming Headset 
S:  SerialNumber=00000000
C:  #Ifs= 4 Cfg#= 1 Atr=a0 MxPwr=100mA
I:  If#= 0 Alt= 0 #EPs= 0 Cls=01(audio) Sub=01 Prot=00 Driver=snd-usb-audio
I:  If#= 1 Alt= 1 #EPs= 1 Cls=01(audio) Sub=02 Prot=00 Driver=snd-usb-audio
I:  If#= 2 Alt= 1 #EPs= 1 Cls=01(audio) Sub=02 Prot=00 Driver=snd-usb-audio
I:  If#= 3 Alt= 0 #EPs= 1 Cls=03(HID  ) Sub=00 Prot=00 Driver=usbhid

`

mattanger, 2017-01-27:
That verifies that I am sane, and this is not a crazy world where the daemon magically grabs corsair devices. The drivers for your headset appear to be correct, and it just your mic that doesn't work? Sound works?

frickler24, 2017-01-29:
There is just one link between pulse audio and our daemon: 
In media_linux.cpp you can find the following line:
`muteProcess.start("sh", QStringList() << "-c" << "pacmd list-sinks|grep -A 15 '* index'|awk '/muted:/{ print $2 }'");
`
That code is used to check (ca. 30 times per second), whether sound is on or muted (BTW the code for the mac is completely different).
It is just reading the config and searches for the "muted" flag and should not give us trouble.
You can try the following: Copy the intersting part of the code in an shell window:
`pacmd list-sinks|grep -A 15 '* index'`

Output should be something like the following and you should not get in trouble with your mike:
```
lutz@Mainfrix ~ $ pacmd list-sinks|grep -A 15 '* index'
  * index: 1
        name: <alsa_output.pci-0000_00_1b.0.analog-stereo>
        driver: <module-alsa-card.c>
        flags: HARDWARE HW_MUTE_CTRL HW_VOLUME_CTRL DECIBEL_VOLUME LATENCY DYNAMIC_LATENCY
        state: RUNNING
        suspend cause: 
        priority: 9959
        volume: front-left: 29493 /  45% / -20,81 dB,   front-right: 29493 /  45% / -20,81 dB
                balance 0,00
        base volume: 65536 / 100% / 0,00 dB
        volume steps: 65537
        muted: no
        current latency: 36,39 ms
        max request: 7 KiB
        max rewind: 64 KiB
        monitor source: 1
```
Maybe that's a hint for this issue.

mattanger, 2017-02-04:
@guenniman do you have any updates on this?

mattanger, 2017-02-06:
Closing due to lack of a response, and the case that it's pretty unlikely to be an issue caused by the daemon. @guenniman If you have any more information or can reproduce a bug please open a new issue, and include as much detail as possible. Thanks!

closed 2017-02-06 by mattanger