mattanger/ckb-next #40: Slow driver loading with error messages

hevanaa, 2017-01-22:
The driver currently works fine, but I wonder if anyone has any idea about the slow loading. There are also some error messages. I'm just wondering if this is specific for only this keyboard model (K70 LUX non-RGB), or if the other keyboards suffer from this, too. Maybe there is still something missing in the driver for this specific model? (I was the one that added the initial support for it in [ccMSC/ckb#530](https://github.com/ccMSC/ckb/pull/530))

When loading the driver, it takes almost a minute to get the keyboard working, and this is on a fast i7 platform. The keyboard is a K70 LUX non-RGB keyboard with latest firmware. Linux kernel is 4.9.4. I'm now running ckb-next master from 2017-01-20.

During boot the following messages are logged:

jan 22 15:58:30 yggdrasil kernel: usb 3-9: unable to read config index 0 descriptor/all
jan 22 15:58:30 yggdrasil kernel: usb 3-9: can't read configurations, error -110
jan 22 15:58:30 yggdrasil kernel: usb 3-9: new full-speed USB device number 7 using xhci_hcd
jan 22 15:58:30 yggdrasil kernel: usb 3-9: New USB device found, idVendor=1b1c, idProduct=1b36
jan 22 15:58:30 yggdrasil kernel: usb 3-9: New USB device strings: Mfr=1, Product=2, SerialNumber=3
jan 22 15:58:30 yggdrasil kernel: usb 3-9: Product: Corsair Gaming K70 LUX Keyboard 
jan 22 15:58:30 yggdrasil kernel: usb 3-9: Manufacturer: Corsair
jan 22 15:58:30 yggdrasil kernel: usb 3-9: SerialNumber: 16030015AEB89C4755BB595DF5001940
jan 22 15:58:30 yggdrasil kernel: input: Corsair Corsair Gaming K70 LUX Keyboard  as /devices/pci0000:00/0000:00:14.0/usb3/3-9/3-9:1.0/0003:1B1C:1B36.0002/input/input7
jan 22 15:58:30 yggdrasil kernel: hid-generic 0003:1B1C:1B36.0002: input,hidraw1: USB HID v1.11 Keyboard [Corsair Corsair Gaming K70 LUX Keyboard ] on usb-0000:00:14.0-9/input0
jan 22 15:58:30 yggdrasil kernel: input: Corsair Corsair Gaming K70 LUX Keyboard  as /devices/pci0000:00/0000:00:14.0/usb3/3-9/3-9:1.1/0003:1B1C:1B36.0003/input/input8
jan 22 15:58:30 yggdrasil kernel: hid-generic 0003:1B1C:1B36.0003: input,hiddev0,hidraw2: USB HID v1.11 Keyboard [Corsair Corsair Gaming K70 LUX Keyboard ] on usb-0000:00:14.0-9/input1
jan 22 15:58:30 yggdrasil kernel: usbhid 3-9:1.2: couldn't find an input interrupt endpoint

When unloading and loading the driver again, the following messages are logged (see the attached file ckb_upgrading.txt).

[ckb_upgrading.txt](https://github.com/mattanger/ckb-next/files/722095/ckb_upgrading.txt)

At the point where "Connection timed out" and "usbfs: USBDEVFS_CONTROL failed cmd ckb-daemon..." is appearing at the end of the log, the keyboard starts working (possibly at the point "jan 22 18:40:19 yggdrasil systemd[1]: Reloading."), but the error messages are still logged a while after the keyboard is working. Then after about 20 seconds the messages stop and don't repeat again.

mattanger, 2017-01-23:
Have you tried adding the appropriate usb quirks? I believe: `usbhid.quirks=0x1B1C:0x1B33:0x20000408`

hevanaa, 2017-01-23:
I actually tried at the point before I upgraded keyboard firmware (and then removed when the new firmware worked). But you are of course right, I will try the usb quirks again.

hevanaa, 2017-01-23:
I was wrong, I still have the quirks configured: usbhid.quirks=0x1B1C:0x**1B36**:0x20000408

hevanaa, 2017-01-23:
I'm suspecting the quirks command is wrong or not working. I tried with usbhid.quirks=0x1B1C:0x1B36:0x20000000, but it didn't help. Still one minute boot (without keyboard around 10 s.).

light2yellow, 2017-01-23:
@hevanaa I've just recalled - I had these sort of issues too, of you have `quiet` parameter - remove it, let the logs flow into the framebuffer. I thought about some race conditions when something did not get set up because of a very quick boot. At least for me this made the broken keyboard cases just after the boot much rare.

hevanaa, 2017-01-26:
Thanks for the suggestions. I've tried without quiet and with and without quirks and different combinations. Boot time in all cases is around one minute because of the timeout.

I noticed another error also, that probably hasn't anything to do with the kernel timeout.

ckb[2538]: The program 'ckb' received an X Window System error.
                                     This probably reflects a bug in the program.
                                     The error was 'BadAccess (attempt to access private resource denied)'.
                                       (Details: serial 670 error_code 10 request_code 131 minor_code 46)
                                       (Note to programmers: normally, X errors are reported asynchronously;
                                        that is, you will receive the error a while after causing it.
                                        To debug your program, run it with the --sync command line
                                        option to change this behavior. You can then get a meaningful
                                        backtrace from your debugger if you break on the gdk_x_error() function.)

frickler24, 2017-01-29:
@hevanaa The X Window error shown is from ckb client, not from the daemon. I think you have found it in your ~/.xsession-errors file, correct?

hevanaa, 2017-01-29:
In Fedora, everything is logged to the journal. As shown, it is from "ckb" (client).

tatokis, 2017-01-29:
Which version of Qt are you using?

hevanaa, 2017-01-29:
Qt 5.7.1, default in Fedora.

tatokis, 2017-01-30:
While I do believe this actually happens, I am not sure if it is ckb-next's fault.

Regarding the slow loading, could you check if usbhid is built as a module.
```
tatokis@tatokis-PC:~$ lsmod | grep usbhid
usbhid                 53248  0
hid                   122880  6 hid_generic,usbhid
```
Also, it might be silly, but check your kernel's cmdline on boot, maybe usbquirks isn't being passed to it for some unknown reason.

hevanaa, 2017-01-31:
usbhid is built into the kernel in Fedora. 

~]$ lsmod |grep hid
i2c_hid                20480  0

And yes, I have checked and changed/edited the grub kernel command line on boot. This is also logged in the journal:

~]$ journalctl -S 2017-01-31 |grep 'Kernel command line'
Jan 31 14:37:42 yggdrasil /usr/libexec/gdm-x-session[32476]: Kernel command line: BOOT_IMAGE=/vmlinuz-4.9.5-200.fc25.x86_64 root=UUID=0c5250a8-fe53-477d-9925-eb469cafd416 ro rootflags=subvol=root rhgb LANG=sv_FI.UTF-8 nouveau.modeset=0 rd.driver.blacklist=nouveau nomodeset gfxpayload=vga=normal usbhid.quirks=0x1B1C:0x1B36:0x20000408

frickler24, 2017-02-14:
@hevanaa I had a longer view into your logfile. The messages beginning with 
```
 jan 22 18:40:16 yggdrasil ckb-daemon[6310]: [E] os_sendindicators (usb_linux.c:84): Connection timed out
jan 22 18:40:16 yggdrasil kernel: usb 3-9: usbfs: USBDEVFS_CONTROL failed cmd ckb-daemon rqt 33 rq 9 len 1 ret -110
  jan 22 18:40:19 yggdrasil systemd[1]: Reloading.
 jan 22 18:40:22 yggdrasil ckb-daemon[6310]: [E] os_usbsend (via firmware.c:15): Connection timed out
jan 22 18:40:22 yggdrasil kernel: usb 3-9: usbfs: USBDEVFS_CONTROL failed cmd ckb-daemon rqt 33 rq 9 len 64 ret -110
jan 22 18:40:27 yggdrasil ckb-daemon[6310]: [E] os_usbsend (via firmware.c:15): Connection timed out
```
I have seen in my logfiles from time  to time.
Especially if I do a lot of testing, the  usb communication get confused with the corsair firmware.
When this happened, closing the daemon and the client and disconnecting the KB for more than 10 seconds helped most of the time.
The reason for the long delay is, that the OS (mine is linux ming 18) needs some time to detect, that the HW is disconnected. If I got hte error mesage again, only power down and reboot helped.
Please have a look at #64. If you are able to install that pullreq for your own in a test branch, feel free to test if it helps you.

light2yellow, 2017-02-26:
@hevanaa did you test #89 as was proposed by @frickler24?

hevanaa, 2017-02-26:
Not yet...

hevanaa, 2017-03-01:
Plugging the keyboard in an USB3 port solves the slow boot issue.

closed 2017-03-01 by hevanaa

light2yellow, 2017-03-01:
Ha. Great that you resolved it! In the context of positive experience inferred from this issue we should henceforth consider no difference between built-in usbhid and a module-like one (@tatokis).

tatokis, 2017-03-01:
Indeed, good to hear it. 

Could you please tell us what your problematic usb2 controller is, for future reference? 

@l2y , it was just an assumption from my end though, as nothing else made sense. Glad to hear it was false, as it could cause a lot of trouble on quite a few systems.

hevanaa, 2017-03-01:
The USB controller seems to be an USB 2.0 controller (motherboard is Asrock Z97 Extreme 4)

Bus 003 Device 002: ID 0409:005a NEC Corp. HighSpeed Hub
  bDeviceClass            9 Hub
  bDeviceProtocol         1 Single TT
  iProduct                0 
  bDeviceClass            9 Hub
  bDeviceProtocol         0 Full speed (or root) hub