mattanger/ckb-next #42: CKB Software reduces wifi speed

drk7891, 2017-01-23:
I have done a lot of testing to why my wifi speeds and ping were so wonky and out of the blue I quit CKB and ran another speed test. This time, my speeds were normal and where they should be.

Is there a reason as to why this is occurring? Im not sure what kind of information I could provide but this is without a doubt the cause.

On MacOS Sierra 10.12.2

tatokis, 2017-01-23:
Do you happen to have the latest macbook pro?
Does this issue also occur with ckb running but with the keyboard unplugged?

What happens if you disable checking for new firmware on the settings tab?

drk7891, 2017-01-23:
I am running it on a mid 2015 MacBook pro. It still happens with ckb running and the keyboard unplugged.

I think it is checking for updates that is causing the issue. I disabled it last night and I haven't had a problem yet. Any other info I can provide let me know

mattanger, 2017-01-23:
So, when you disabled the check for updates your wifi issue stopped?

drk7891, 2017-01-23:
Yes, I had to restart the application though after disabling the check for updates option

light2yellow, 2017-01-26:
@drk7891 I've pushed a commit to one of the branches. Could you please try it out? Most probably I haven't done it right but at least it compiled for me and didn't crash after pressing few buttons. I assume you have the last Qt version installed (which is 5.7.x). You can do this with:
```
git clone -b checkupdate-destructor --single-branch https://github.com/mattanger/ckb-next.git checkupdate-destructor && cd checkupdate-destructor && ./quickinstall
```
It clones only that branch and then installs as usually.

frickler24, 2017-01-29:
@light2yellow I tried the code and had no problem. Class KbFirmware is used just with static method calls. The old code with the init-manager was aware of this hack: 
```
void KbFirmware::initManager(){
    if(networkManager)
        return;
    networkManager = new QNetworkAccessManager();
}
```
The documentation of QNetworkAccessManager gives a hint: one object of this class should be enough for an QApplication.
The best Idea would be to refactor the code: create a KbFirmware object when starting the client and just use it in the KbFirmware Class. Open the QNetworkAccessManager in the contructror KbFirmware::KbFirmware(..).

light2yellow, 2017-01-29:
@frickler24 did I understand you right? 77e530d
Anyway I thought that this network object can cause some issues when I saw the destructor being commented out. But it seems to me that `KbFirmware` instance has the same lifecycle as the whole program and it is singletoned, so probably won't make any difference. Although should look cleaner.

light2yellow, 2017-02-19:
While you're waiting for solution, try building from `checkupdate-destructor` branch and see if it changes something. Although I doubt it will, today everything is possible. You can find a guide how to build from a specific git branch on the Internet.

light2yellow, 2017-02-26:
@drk7891 any news on this?

joemulray, 2017-03-23:
`checkupdate-destructor` work? having this issues on MPB 2014 model, along with cpu issues when running ckb

light2yellow, 2017-03-23:
@joemulray in case you wonder whether the branch works, then yes, it did build for me and I didn't experience any issues. As earlier, I doubt it can help with this particular issue, because the network object was instantiated only once with everything else. Nevertheless, it had a comment that it may cause bugs (in Qt implementation, not in ckb), so you are encouraged to test anyway.

light2yellow, 2017-06-29:
In case your Wi-Fi traffic consists of a lot of video streaming or whatever, you might to lower the FPS rate or turn off animations completely to free some CPU time.

commit b9ed55fb (2017-09-05): Merge 'ckb-next-pjcv/checkupdate-destructor'. Closes #42
closed 2017-09-05 by tatokis