mattanger/ckb-next #44: Add a donation link

JasonRivers, 2017-01-26:
I understand why you removed the donation link that was going to the original author, Please add a donation link so that those of use that wish to can still support the on-going development with this project.

mattanger, 2017-01-28:
Well money is always nice, and it's comforting to know that at least one person wants to give us some. :smile:  Once we decide that the release is ready, we'll probably put a link up. Thing is, we've go to decide what we'd do with any donations. In the meantime any other support you can offer is much appreciated! If you're a web designer, we've been thinking about putting up a site for the project, and I'm not a designer.. :wink: :wink:

JarrenJ, 2017-01-28:
@mattanger , I am pretty decent at website designing. I can't promise anything because I'm pretty busy right now. But I can work on it in my spare time. If you have any specifics you would want on the site send me an email at: j.jarren@yahoo.com. -- I hope I can find time to make a decent website because I want to help support this software in any way possible.

mattanger, 2017-01-29:
No worries, and thanks for the offer! As nice as a site would be, we're more concerned about producing a stable release.

closed 2018-01-21 by tatokis