mattanger/ckb-next #45: Is K95 Platinum Supported?

kerryhall, 2017-01-26:
If so, might be a good idea to note it in the readme.

Thanks!

mattanger, 2017-01-28:
Not that I'm aware of, please post the output of the ckb-dev-detect to #2

kerryhall, 2017-01-28:
I do not own a K95 Platinum yet, I am considering purchasing one just to try and test here.

light2yellow, 2017-01-28:
@kerryhall you don't need ckb to work on Linux or mac, it just allows to customize the keyboard in a similar way as the CUE software on Windows does.

kerryhall, 2017-01-28:
Understood. That is the functionality I am interested in. (Lighting at the very least)

mattanger, 2017-01-28:
Get one and let's try it out! Are you on macOS or Linux?

tatokis, 2017-01-28:
This "LightEdge" stuff is definitely not going to work out of the box, 
and I am not sure what other changes corsair did to the protocol.
Maybe it will work, maybe it won't. We can't really tell, however 
someone needs to get it and send us packet captures as one day we would 
like to support it.

Also, if you're on Sierra, there are some issues with it currently due 
to Apple deprecating APIs, so be aware of that.

kerryhall, 2017-01-29:
I'm on Linux but I'm still on the fence about dumping $200 for something that might not work, haha. Let me consider the return policy first. Thanks!

mattanger, 2017-01-29:
All good, if you decide to go ahead with it post the requisite info on #2, and we'll go from there. Thanks!

closed 2017-01-29 by mattanger