mattanger/ckb-next #47: Random SIGSEGVs (QSocketNotifier::setEnabled())

fgblomqvist, 2017-01-29:
This error happened in the same way in the original ckb as well (the latest one available there). Upgraded to ckb-next last week (like Jan 26) but it didn't go away.

Let me know if there are any logs I can provide you with. It just happens randomly, but every day for sure.

tatokis, 2017-01-29:
Is it related to https://github.com/ccMSC/ckb/issues/419 ?

fgblomqvist, 2017-01-29:
Possibly, I tried setting an animation now (trippy). I'll have to see if it solves the problem. Before I always kept it on solid. But yeah, hopefully the problem can be tracked down so that an animation is not needed?

frickler24, 2017-01-29:
@fgblomqvist PLease support us with some debugging info.
```
cd <ckb-next-dir>/src/ckb 
make clean debug && ckb --close && sleep 3 && gdb ../../bin/ckb
(in gdb type) run

when you get the sigsegv you are back in gdb
(in gdb type) backtrace

```
copy and paste the complete info from gdb here.

fgblomqvist, 2017-01-31:
```
(gdb) run
Starting program: /home/fgblomqvist/Desktop/ckb-next-master/bin/ckb 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7fffe8123700 (LWP 11817)]
[New Thread 0x7fffe7922700 (LWP 11818)]
[New Thread 0x7fffd7fff700 (LWP 11820)]
[New Thread 0x7fffd77fe700 (LWP 11821)]
[New Thread 0x7fffd6ffd700 (LWP 11822)]
[New Thread 0x7fffd67fc700 (LWP 11823)]
libpng warning: iCCP: known incorrect sRGB profile
[New Thread 0x7fffd5df9700 (LWP 11824)]
Scanning  "/home/fgblomqvist/Desktop/ckb-next-master/bin/ckb-animations/ckb-gradient"
Scanning  "/home/fgblomqvist/Desktop/ckb-next-master/bin/ckb-animations/ckb-heat"
Scanning  "/home/fgblomqvist/Desktop/ckb-next-master/bin/ckb-animations/ckb-pinwheel"
Scanning  "/home/fgblomqvist/Desktop/ckb-next-master/bin/ckb-animations/ckb-rain"
Scanning  "/home/fgblomqvist/Desktop/ckb-next-master/bin/ckb-animations/ckb-random"
Scanning  "/home/fgblomqvist/Desktop/ckb-next-master/bin/ckb-animations/ckb-ripple"
Scanning  "/home/fgblomqvist/Desktop/ckb-next-master/bin/ckb-animations/ckb-wave"
[New Thread 0x7fffbfeff700 (LWP 11833)]
[New Thread 0x7fffbf645700 (LWP 11834)]
[New Thread 0x7fffbe797700 (LWP 11835)]
[New Thread 0x7fffbdf96700 (LWP 11836)]
[Thread 0x7fffd67fc700 (LWP 11823) exited]
Downloaded new firmware list. 6 entries found.
[Thread 0x7fffd6ffd700 (LWP 11822) exited]
[Thread 0x7fffbdf96700 (LWP 11836) exited]
[New Thread 0x7fffbdf96700 (LWP 12029)]
[Thread 0x7fffbdf96700 (LWP 12029) exited]
ckb: Fatal IO error 10 (No child processes) on X server :0.
[Thread 0x7fffe7922700 (LWP 11818) exited]
QProcess: Destroyed while process ("sh") is still running.

Thread 1 "ckb" received signal SIGSEGV, Segmentation fault.
0x00007ffff5fabc03 in QSocketNotifier::setEnabled(bool) ()
   from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
(gdb) 
```

I turned off the animation (switched to Solid) before running CKB in debug mode (since the bug doesn't seem to happen in animated mode). I'll leave the gdb window open for another few hours in case any of you respond before then and want me to type any commands.

tatokis, 2017-01-31:
Yup, that's exactly the same issue.
I can reproduce it as well (as indicated from the other thread). All one needs to do is set a static animation and wait for it to segfault.

I haven't been able to figure out what is going on, QProcess usage seems okay, and I know it's not a Qt bug because it happens across different versions.
Suggestions are welcome.

fgblomqvist, 2017-01-31:
Ah okay. If I ever have some time over I'll see if I can look into it. Right now I'm dealing with an SSD failure so it will have to be later. Thanks for the info though!

frickler24, 2017-01-31:
@fgblomqvist IF your gdb window is still open, please type 
backtrace
and send me the output.

fgblomqvist, 2017-01-31:
@frickler24 Unfortunately due to the SSD failure (on which I had my Ubuntu) I can't help with this at the moment (otherwise I would start it up again and do the backtrace). But I'm hoping to have it fixed by the end of the week. Unless someone else gets back here with a backtrace before then, I'll do it

frickler24, 2017-01-31:
OK! Good luck for your SSD issue.

MaxHaydenChiz, 2017-02-19:
I was able to replicate this issue with the main branch, but after making a debug build of the testing branch to get a backtrace, I haven't been able to reproduce the problem. I'll keep it running in gdb for a few more days in case testing just reduced the frequency of the issue instead of eliminating it entirely. But so far, it appears that this issue was fixed.

MaxHaydenChiz, 2017-02-20:
Is this error message I got right after `run` relevant:
`GPG couldn't verify firmware signature:
""
"gpg: Signature made Tue 07 Feb 2017 06:18:10 PM CST using RSA key ID DCEF2F5F\ngpg: failed to create temporary file `/home/mchiz/.gnupg/.#lk0x703840.max-ThinkPad-X220.14481': Permission denied\ngpg: fatal: can't create lock for `/home/mchiz/.gnupg/trustdb.gpg'\nsecmem usage: 1408/1408 bytes in 2/2 blocks of pool 1408/65536\n"`

FWIW, testing does still crash this way, it just seems to be much less common. The one time it happened for me so far, I wasn't in gdb at the time because I had to rebooted and forgot to restart gdb. But I'll try to get that backtrace next time it happens.

MaxHaydenChiz, 2017-02-20:
Here we go. Do you need me to do anything else while I've got it in gdb or is this everything you need?

ckb: Fatal IO error 10 (No child processes) on X server :0.
QProcess: Destroyed while process ("sh") is still running.

Thread 1 "ckb" received signal SIGSEGV, Segmentation fault.
0x00007ffff6a990c1 in QSocketNotifier::setEnabled(bool) ()
   from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
(gdb) backtrace
#0  0x00007ffff6a990c1 in QSocketNotifier::setEnabled(bool) ()
   from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#1  0x00007ffff69939eb in ?? () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#2  0x00007ffff69f3675 in ?? () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#3  0x00007ffff698e2bf in QProcess::waitForFinished(int) ()
   from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#4  0x00007ffff698eade in QProcess::~QProcess() ()
   from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#5  0x00007ffff598aff8 in __run_exit_handlers (status=1, 
    listp=0x7ffff5d145f8 <__exit_funcs>, 
    run_list_atexit=run_list_atexit@entry=true) at exit.c:82
#6  0x00007ffff598b045 in __GI_exit (status=<optimized out>) at exit.c:104
#7  0x00007fffea6eaea1 in ?? ()
   from /usr/lib/x86_64-linux-gnu/libgdk-x11-2.0.so.0
#8  0x00007ffff78c2cee in _XIOError ()
   from /usr/lib/x86_64-linux-gnu/libX11.so.6
#9  0x00007ffff78c057d in _XEventsQueued ()
   from /usr/lib/x86_64-linux-gnu/libX11.so.6
#10 0x00007ffff78b1f47 in XPending ()
   from /usr/lib/x86_64-linux-gnu/libX11.so.6
#11 0x00007fffea6e0e2e in ?? ()
   from /usr/lib/x86_64-linux-gnu/libgdk-x11-2.0.so.0
#12 0x00007ffff501091d in g_main_context_prepare ()
---Type <return> to continue, or q <return> to quit---
   from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#13 0x00007ffff50112bb in ?? () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#14 0x00007ffff501149c in g_main_context_iteration ()
   from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#15 0x00007ffff6ab47cf in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#16 0x00007ffff6a5bb4a in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#17 0x00007ffff6a63bec in QCoreApplication::exec() ()
   from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#18 0x0000000000420a19 in main (argc=1, argv=0x7fffffffdf68) at main.cpp:222

frickler24, 2017-02-20:
@MaxHaydenChiz To your GPG error: This msg comes up only while validating the signature for the firmware-list, which is signed with gnupg. Are you sure to have installed gnupg correctly? Is the path mentioned (home/mchiz/.gnupg/) writable and executable for you or your group (sometime the installation runs with SU privileges and the config file is owned by root)?

MaxHaydenChiz, 2017-02-20:
It was owned by root, I assume it isn't supposed to be and changed the ownership to my user. I'll confirm that this fixed the problem after I'm told that I no longer need to hold this gdb session open.

tatokis, 2017-02-20:
Please run in the gdb prompt "frame 18" and then "info locals" and 
"info args"

MaxHaydenChiz, 2017-02-20:
(gdb) frame 18
#18 0x0000000000420a19 in main (argc=1, argv=0x7fffffffdf68) at main.cpp:222
222	    return a.exec();
(gdb) info locals
a = <incomplete type>
parser = {d = 0x9ab3b0}
errorMessage = {static null = {<No data fields>}, 
  d = 0x7ffff6b313e0 <QArrayData::shared_null>}
background = false
w = {<QMainWindow> = {<No data fields>}, static staticMetaObject = {d = {
      superdata = 0x7ffff7874e80 <QMainWindow::staticMetaObject>, 
      stringdata = 0x52ff80 <qt_meta_stringdata_MainWindow>, 
      data = 0x5302a0 <qt_meta_data_MainWindow>, 
      static_metacall = 0x4f37e2 <MainWindow::qt_static_metacall(QObject*, QMetaObject::Call, int, void**)>, relatedMetaObjects = 0x0, extradata = 0x0}}, 
  static mainWindow = 0x7fffffffde00, settingsWidget = 0x9c6940, 
  kbWidgets = {<QListSpecialMethods<KbWidget*>> = {<No data fields>}, {p = {
        static shared_null = {ref = {atomic = {_q_value = -1}}, alloc = 0, 
          begin = 0, end = 0, array = {0x0}}, d = 0xf58da0}, d = 0xf58da0}}, 
  restoreAction = 0x9b8350, closeAction = 0x9b8370, trayIconMenu = 0x9b8a10, 
  trayIcon = 0x9b9480, ui = 0x9ab6d0}
(gdb) info args
argc = 1
argv = 0x7fffffffdf68

frickler24, 2017-02-20:
@tatokis just a hint: For me it looks like memory overrun / buffer overflow or similar. That's at least the second traceback I've seen where exiting of a QTProcess fails exactly at that position. One is only able to produce this with completely corrupted stack or heap.

tatokis, 2017-02-20:
@frickler24 Indeed, I just wanted to be sure that the problem isn't anywhere near main.c, as that's the only mention of ckb code in the backtrace.

```QProcess: Destroyed while process ("sh") is still running.```
Not sure why it would destroy an sh process in the first place. If I remember correctly, sh is used to spawn the animations, but in this case there are none.

tatokis, 2017-02-20:
Maybe it would help if we recorded the process' execution in gdb, and then started going backwards for any clues.

MaxHaydenChiz, 2017-02-20:
While it is running, it generally creates and destroys processes even when no animations are on. From earlier:
(gdb) run
Starting program: /home/mchiz/Downloads/ckb-next-testing/bin/ckb 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7fffec026700 (LWP 14354)]
[New Thread 0x7fffe1c77700 (LWP 14355)]
[New Thread 0x7fffe1476700 (LWP 14356)]
[New Thread 0x7fffe0c75700 (LWP 14357)]
[New Thread 0x7fffcfbba700 (LWP 14359)]
[New Thread 0x7fffcf3b9700 (LWP 14360)]
[New Thread 0x7fffce85e700 (LWP 14362)]
Scanning  "/home/mchiz/Downloads/ckb-next-testing/bin/ckb-animations/ckb-gradient"
Scanning  "/home/mchiz/Downloads/ckb-next-testing/bin/ckb-animations/ckb-heat"
Scanning  "/home/mchiz/Downloads/ckb-next-testing/bin/ckb-animations/ckb-pinwheel"
Scanning  "/home/mchiz/Downloads/ckb-next-testing/bin/ckb-animations/ckb-rain"
Scanning  "/home/mchiz/Downloads/ckb-next-testing/bin/ckb-animations/ckb-random"
Scanning  "/home/mchiz/Downloads/ckb-next-testing/bin/ckb-animations/ckb-ripple"
Scanning  "/home/mchiz/Downloads/ckb-next-testing/bin/ckb-animations/ckb-wave"
[New Thread 0x7fffccf6f700 (LWP 14370)]
[New Thread 0x7fffbffff700 (LWP 14372)]
[New Thread 0x7fffbf595700 (LWP 14373)]
[New Thread 0x7fffbed94700 (LWP 14374)]
[Thread 0x7fffcfbba700 (LWP 14359) exited]
GPG couldn't verify firmware signature:
""
"gpg: Signature made Tue 07 Feb 2017 06:18:10 PM CST using RSA key ID DCEF2F5F\ngpg: failed to create temporary file `/home/mchiz/.gnupg/.#lk0x703840.max-ThinkPad-X220.14481': Permission denied\ngpg: fatal: can't create lock for `/home/mchiz/.gnupg/trustdb.gpg'\nsecmem usage: 1408/1408 bytes in 2/2 blocks of pool 1408/65536\n"
[Thread 0x7fffcf3b9700 (LWP 14360) exited]
[Thread 0x7fffbed94700 (LWP 14374) exited]
ckb: Fatal IO error 10 (No child processes) on X server :0.

MaxHaydenChiz, 2017-02-20:
If there's anything you need me to do, let me know and I'll do it. (E.g. Is there a way to build it with stack smashing protection and a memory overflow robust malloc to try to make it crash at the "right" point in time?)

tatokis, 2017-02-20:
Well turns out I was wrong, according to htop, it appears that "sh" isn't running even with animations enabled.
Other than the pulseaudio volume adjustment code, it is only called here https://github.com/mattanger/ckb-next/blob/testing/src/ckb/keyaction.cpp#L418
and here https://github.com/mattanger/ckb-next/blob/testing/src/ckb/keyaction.cpp#L421

tatokis, 2017-02-20:
@MaxHaydenChiz Not 100% sure, but I guess you can try something like this
```
diff --git a/src/ckb/ckb.pro b/src/ckb/ckb.pro
index 91ae7a3..351ac24 100644
--- a/src/ckb/ckb.pro
+++ b/src/ckb/ckb.pro
@@ -8,8 +8,8 @@ TEMPLATE = app
 # GL isn't needed
 QMAKE_LIBS_OPENGL =
 
-QMAKE_CFLAGS += -Wno-unused-parameter
-QMAKE_CXXFLAGS += -Wno-unused-parameter
+QMAKE_CFLAGS += -Wno-unused-parameter -fstack-protector-all
+QMAKE_CXXFLAGS += -Wno-unused-parameter -fstack-protector-all
 
 # Output path
 macx {
```
I'll turn off animations again to make it crash and see if I can figure anything out.

tatokis, 2017-02-20:
You can also try and use valgrind, as it will possibly detect failures in such cases, however I haven't used it before and unfortunately I don't have much time right now.

MaxHaydenChiz, 2017-02-20:
I tried compiling it with -fsanitize=address to try that before breaking out valgrind, but I got a load of compilation errors. Do you want me to post the output here, or you want to just run it yourself and see?

tatokis, 2017-02-20:
pass ```-lasan``` to the linker, if that's what you are on about.

tatokis, 2017-02-20:
It also appears that no GUI is drawn when libasan is used...

MaxHaydenChiz, 2017-02-20:
Well, I can't figure out why it won't let me link libasan (maybe ubuntu doesn't install it by default). But if it doesn't work with that option, I'll stick to the stack smasher and wait for a crash. If you need me to try anything else, let me know.

MaxHaydenChiz, 2017-02-20:
Got the crash again, but nothing helpful or new, the debug output is identical except for a few local variable values:
(gdb) info locals
a = <incomplete type>
parser = {d = 0x9ca850}
errorMessage = {static null = {<No data fields>}, 
  d = 0x7ffff6b313e0 <QArrayData::shared_null>}
background = false
w = {<QMainWindow> = {<No data fields>}, static staticMetaObject = {d = {
      superdata = 0x7ffff7874e80 <QMainWindow::staticMetaObject>, 
      stringdata = 0x545040 <qt_meta_stringdata_MainWindow>, 
      data = 0x545360 <qt_meta_data_MainWindow>, 
      static_metacall = 0x505dd0 <MainWindow::qt_static_metacall(QObject*, QMetaObject::Call, int, void**)>, relatedMetaObjects = 0x0, extradata = 0x0}}, 
  static mainWindow = 0x7fffffffde00, settingsWidget = 0x9df450, 
  kbWidgets = {<QListSpecialMethods<KbWidget*>> = {<No data fields>}, {p = {
        static shared_null = {ref = {atomic = {_q_value = -1}}, alloc = 0, 
          begin = 0, end = 0, array = {0x0}}, d = 0xf78360}, d = 0xf78360}}, 
  restoreAction = 0x9d7840, closeAction = 0x9d7860, trayIconMenu = 0x9d8240, 
  trayIcon = 0x9d8be0, ui = 0x9c9c00}

MaxHaydenChiz, 2017-02-20:
I got it running with -lasan with the GUI. Starting it and stopping it gives a few memory leaks, but none that appear to originate in ckb code. I'll see if getting the crash this way gives any additional info.

tatokis, 2017-02-20:
Do you happen to be running the nvidia proprietary graphics drivers?

frickler24, 2017-02-20:
@MaxHaydenChiz last time I ran ckb client with valgrind, just about 20kB were marked as lost. So if You encounter bigger amounts of wasted memory, it would sound strange to me.
The reason is that I suspect that an area is overwritten, that means, a too small memory area is overwritten with more data (also known as basis for buffer overflow attacks when writing viruses). That effect can't be detected by valgrind or similar tools.

The only chance is: Looking into the code, where are process objects defined (where are they stored: Stack, heap?) and what vars are around them, which are used as buffers.
Or: when the next crash occurs, try to dump the memory where the QProcess object should be. If there are readable characters or repeated structures, we get the next hints.
Last time, when I searched for a bug like this, we searched several weeks...

light2yellow, 2017-02-23:
@frickler24 if I understood it right, we could use [sentinels](https://news.ycombinator.com/item?id=11399181) to check if the data is overwritten.

MaxHaydenChiz, 2017-02-23:
Finally got it to crash while using -lasan. But I didn't learn much useful except that the "QProcess" error can happen when you run the program and then quit normally without the segfault occurring.  (Though it doesn't always happen.)

Based on thinking about how the crash always happens right after a thread exits, I think the issue is a race condition along the following lines:
While it runs the program keeps creating and eventually exiting some thread. Every time the thread exits, it immediately creates a new one. The crash always happens right after the thread exits and before it can create another one. So I think what happens is that Qt tries to do something to the threads during that interim and because there are temporarily no threads of the right type, it segfaults trying to access them. The bug doesn't happen with animations running because those threads are still around to give Qt a thread to do whatever it is trying to do.

Does this make sense?

Here is an example from the latest crash:

[Thread 0x7fffd57a9700 (LWP 23460) exited]
[New Thread 0x7fffd57a9700 (LWP 14483)]
[Thread 0x7fffd57a9700 (LWP 14483) exited]
ckb: Fatal IO error 10 (No child processes) on X server :0.
QProcess: Destroyed while process ("sh") is still running.

Thread 1 "ckb" received signal SIGSEGV, Segmentation fault.
0x00007ffff5b2c0c1 in QSocketNotifier::setEnabled(bool) ()
   from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5

The rest is also the same as the others.

MaxHaydenChiz, 2017-02-23:
re: questions, no I'm not using Nvidia drivers. This computer has Intel graphics. The leaks found by libasan are similarly small. Per my suggestion above, I concur that this isn't the kind of memory fault that valgrind will detect. lasan certainly didn't find anything of note.

MaxHaydenChiz, 2017-02-23:
Debugging it with -fsanitize=thread finds a host of issues. Data races, lock order inversions, etc.

I've got no way to tell which of these is the issue causing the crash, but there are several bugs here that are all good candidates. So I would suggest fixing the problems that this flags and then we can see if the crash still happens.

tatokis, 2017-02-23:
I'll have a look when I get home later, and try with fsanitize thread. 

Just to clarify, I asked about the graphics driver because when I'd try to run the Qt gui with libsan, it wouldn't draw a gui. 
Breaking it manually with gdb would point to the nvidia shared libraries. So I guess a VM is the only option for me, in case it's needed.

MaxHaydenChiz, 2017-02-23:
Here are some of the errors (these all happen right after the thread exits, which indicates my theory may be onto something):

[Thread 0x7fffd1421700 (LWP 27463) exited]
==================
WARNING: ThreadSanitizer: data race (pid=27440)
  Write of size 8 at 0x7d180003bf68 by main thread:
    #0 pthread_cond_destroy <null> (libtsan.so.0+0x000000027f97)
    #1 QWaitCondition::~QWaitCondition() <null> (libQt5Core.so.5+0x0000000a5f93)
    #2 __libc_start_main <null> (libc.so.6+0x00000002082f)

  Previous read of size 8 at 0x7d180003bf68 by thread T11 (mutexes: write M1682):
    #0 pthread_cond_broadcast <null> (libtsan.so.0+0x000000027e84)
    #1 QWaitCondition::wakeAll() <null> (libQt5Core.so.5+0x0000000a65b6)

  Location is heap block of size 96 at 0x7d180003bf40 allocated by thread T10:
    #0 operator new(unsigned long) <null> (libtsan.so.0+0x000000025a33)
    #1 QWaitCondition::QWaitCondition() <null> (libQt5Core.so.5+0x0000000a5d54)

  Mutex M1682 (0x7d180003bf40) created at:
    #0 pthread_mutex_init <null> (libtsan.so.0+0x0000000280b5)
    #1 QWaitCondition::QWaitCondition() <null> (libQt5Core.so.5+0x0000000a5d62)

  Thread T11 'Thread (pooled)' (tid=27463, finished) created by thread T10 at:
    #0 pthread_create <null> (libtsan.so.0+0x000000027577)
    #1 QThread::start(QThread::Priority) <null> (libQt5Core.so.5+0x0000000a4d27)

  Thread T10 'Qt HTTP thread' (tid=27462, running) created by main thread at:
    #0 pthread_create <null> (libtsan.so.0+0x000000027577)
    #1 QThread::start(QThread::Priority) <null> (libQt5Core.so.5+0x0000000a4d27)
    #2 __libc_start_main <null> (libc.so.6+0x00000002082f)

SUMMARY: ThreadSanitizer: data race ??:0 __interceptor_pthread_cond_destroy
==================
==================
WARNING: ThreadSanitizer: data race (pid=27440)
  Write of size 8 at 0x7d180003bf40 by main thread:
    #0 operator delete(void*) <null> (libtsan.so.0+0x000000025d19)
    #1 QWaitCondition::~QWaitCondition() <null> (libQt5Core.so.5+0x0000000a5faf)
    #2 __libc_start_main <null> (libc.so.6+0x00000002082f)

  Previous atomic read of size 1 at 0x7d180003bf40 by thread T11:
    #0 pthread_mutex_lock <null> (libtsan.so.0+0x000000037286)
    #1 QWaitCondition::wakeAll() <null> (libQt5Core.so.5+0x0000000a658f)

  Thread T11 'Thread (pooled)' (tid=27463, finished) created by thread T10 at:
    #0 pthread_create <null> (libtsan.so.0+0x000000027577)
    #1 QThread::start(QThread::Priority) <null> (libQt5Core.so.5+0x0000000a4d27)

SUMMARY: ThreadSanitizer: data race ??:0 operator delete(void*)
==================
==================
WARNING: ThreadSanitizer: data race (pid=27440)
  Write of size 8 at 0x7d0800070d00 by main thread:
    #0 free <null> (libtsan.so.0+0x000000025819)
    #1 QThreadPoolPrivate::reset() <null> (libQt5Core.so.5+0x0000000a1027)
    #2 __libc_start_main <null> (libc.so.6+0x00000002082f)

  Previous write of size 8 at 0x7d0800070d00 by thread T11:
    #0 malloc <null> (libtsan.so.0+0x0000000254a3)
    #1 QListData::detach_grow(int*, int) <null> (libQt5Core.so.5+0x00000010a8a2)

  Thread T11 'Thread (pooled)' (tid=27463, finished) created by thread T10 at:
    #0 pthread_create <null> (libtsan.so.0+0x000000027577)
    #1 QThread::start(QThread::Priority) <null> (libQt5Core.so.5+0x0000000a4d27)

SUMMARY: ThreadSanitizer: data race ??:0 __interceptor_free
==================
==================
WARNING: ThreadSanitizer: data race (pid=27440)
  Write of size 8 at 0x7d040000c390 by main thread:
    #0 free <null> (libtsan.so.0+0x000000025819)
    #1 QMetaCallEvent::~QMetaCallEvent() <null> (libQt5Core.so.5+0x0000002b20c2)
    #2 __libc_start_main <null> (libc.so.6+0x00000002082f)

  Previous write of size 8 at 0x7d040000c390 by thread T1:
    #0 malloc <null> (libtsan.so.0+0x0000000254a3)
    #1 QMetaObject::activate(QObject*, int, int, void**) <null> (libQt5Core.so.5+0x0000002b47c1)

  Thread T1 'QXcbEventReader' (tid=27442, running) created by main thread at:
    #0 pthread_create <null> (libtsan.so.0+0x000000027577)
    #1 QThread::start(QThread::Priority) <null> (libQt5Core.so.5+0x0000000a4d27)
    #2 __libc_start_main <null> (libc.so.6+0x00000002082f)

SUMMARY: ThreadSanitizer: data race ??:0 __interceptor_free
==================
^This last one repeats several times. So I assume that's a loop

Here is the lock order inversion error:
==================
WARNING: ThreadSanitizer: lock-order-inversion (potential deadlock) (pid=27440)
  Cycle in lock order graph: M209 (0x7d0c00005f70) => M1459 (0x7d0c000770d0) => M209

  Mutex M1459 acquired here while holding mutex M209 in main thread:
    #0 pthread_mutex_lock <null> (libtsan.so.0+0x000000037286)
    #1 <null> <null> (libdbus-1.so.3+0x00000000f91a)
    #2 KbFirmware::_checkUpdates() /home/mchiz/Downloads/ckb-next-testing/src/ckb/kbfirmware.cpp:31 (ckb+0x000000507fa7)
    #3 KbFirmware::checkUpdates() /home/mchiz/Downloads/ckb-next-testing/src/ckb/kbfirmware.h:15 (ckb+0x00000042664d)
    #4 MainWindow::timerTick() /home/mchiz/Downloads/ckb-next-testing/src/ckb/mainwindow.cpp:250 (ckb+0x000000425385)
    #5 MainWindow::qt_static_metacall(QObject*, QMetaObject::Call, int, void**) debug/moc_mainwindow.cpp:121 (ckb+0x000000548478)
    #6 QMetaObject::activate(QObject*, int, int, void**) <null> (libQt5Core.so.5+0x0000002b4d29)
    #7 __libc_start_main <null> (libc.so.6+0x00000002082f)

    Hint: use TSAN_OPTIONS=second_deadlock_stack=1 to get more informative warning message

  Mutex M209 acquired here while holding mutex M1459 in main thread:
    #0 pthread_mutex_lock <null> (libtsan.so.0+0x000000037286)
    #1 <null> <null> (libdbus-1.so.3+0x0000000274ec)

SUMMARY: ThreadSanitizer: lock-order-inversion (potential deadlock) ??:0 pthread_mutex_lock
==================

tatokis, 2017-02-23:
Can you please put the output inside markdown quotes/blocks so that it doesn't get parsed (and cause mentions on other issues)? 
Thank you

MaxHaydenChiz, 2017-02-23:
Sorry, not familiar with markdown, I kept trying to block quote it and it didn't work. You need to use three backticks before and after to get code quoting.

light2yellow, 2017-03-11:
I also experience this problem (_ckb crashed with SIGSEGV in QSocketNotifier::setEnabled()_).

Today's full log:
[_usr_bin_ckb.1000.crash.txt](https://github.com/mattanger/ckb-next/files/835791/_usr_bin_ckb.1000.crash.txt)

light2yellow, 2017-03-11:
Also see [this](https://github.com/mattanger/ckb-next/issues/99#issuecomment-283990723) comment from #99 and the one that follows it.

Aaron432, 2017-08-17:
In regard to [Issue #278](https://github.com/mattanger/ckb-next/issues/278), which was marked as duplicate of this one, I'm sending `gdb` full backtrace from one of my crashes:

https://pastebin.com/smYETUKC

Hope that helps. Please tell me if there's anything more I can do to help.

tatokis, 2018-01-21:
https://github.com/ckb-next/ckb-next/issues/2

closed 2018-01-21 by tatokis