mattanger/ckb-next #49: K65 rgb No devices connected

Wertuoz, 2017-01-30:
Sometimes it works fine, but now my keyboard is not responding.
This problem repeats in both my macs(imac and mackbook pro)
https://puu.sh/tGvWB/0ab4722308.png
https://puu.sh/tGFdB/ccebb18dbd.png
[ckb-dev-detect-report.gz](https://github.com/mattanger/ckb-next/files/738734/ckb-dev-detect-report.gz)

pid:1012
version:beta-v0.2.7
connected:

tatokis, 2017-01-30:
It appears that the dev detect script has failed completely,
can you please run ```system_profiler SPUSBDataType -detailLevel mini``` in a terminal and send the output?

Wertuoz, 2017-01-30:
[USB.txt](https://github.com/mattanger/ckb-next/files/738818/USB.txt)
Thank you for your fast reaction)
Im connect k65 to usb and then connect my mac keyboard to k65 by usb)

tatokis, 2017-01-30:
Was the K65 RGB connected to your computer when you ran that command?
It doesn't seem to be recognised at all by your operating system.

Wertuoz, 2017-01-30:
yes, rgb ligthting is working on it now

Wertuoz, 2017-01-30:
[USB.txt](https://github.com/mattanger/ckb-next/files/738843/USB.txt)
logs from another mac

tatokis, 2017-01-30:
I'm afraid this is most likely not an issue with ckb-next, as macOS doesn't seem to recognise the keyboard at all. I can't find your keyboard listed in any of your logs. It is possible that some of the output got chopped off at the end, unless you do not have anything plugged in on your USB3 port.

If I understood correctly, the device works with ckb-next and then just stops all of a sudden?

I would suggest possibly changing the poll rate to 8ms (using the switch on the back), trying different USB ports, making sure that both the power and usb cables are plugged in, and try avoiding external USB hubs.

Other than that, unfortunately I can't help any further, as I use Linux.
Maybe someone else can help or suggest something.

Wertuoz, 2017-01-30:
it works in both computers, but today is stop working(

fleischie, 2017-01-30:
Also check, whether the keyboard has enough power. Either by plugging in both USB cables into two USB 2.0 ports or the keyboard-icon-cable into a USB 3.0 port.

Especially with passive external hubs, that are only plugged into one USB 2.0 port there might not be enough power to enable the keyboard and light it up.

Wertuoz, 2017-01-30:
without drivers keyboard worked normal and symbols was typing, but i cant config rgb, now keys not responding

fleischie, 2017-01-30:
What version of Mac OS are you running and do you have another non-corsair keyboard you could test, whether it works?

tatokis, 2017-01-30:
@fleischie ckb-dev-detect has the OS version

Wertuoz, 2017-01-30:
10.11.6 el capitan, my mac keyboard works fine

fleischie, 2017-01-30:
@tatokis Sorry, seems I am not fit reading today. :confused:

mattanger, 2017-01-30:
Have do you have another keyboard that you can plug into a usb port to verify that it works? I don't think I'm quite understanding what you mean, is the keyboard working fine without the ckb-next running, or not at all?

tatokis, 2017-01-30:
@fleischie No worries, I just mentioned it for future reference :)

Wertuoz, 2017-01-30:
i have default mac usb keyboard and im writing here by this) 
Now keyboard is not working totally, only lights are bright)

Wertuoz, 2017-01-30:
Issue solved, my steps were:
run keyboard in win
corsair utility engine cant find device
remove drivers manually from admin panel
remove utility
switch poll rate to 1, connect to usb 3.0 and install cue
cue shows me my keyboard and after this i connect to mac and its works
i think, this problem starts after i switch to bios and connect to usb 3.0
magic%)

Wertuoz, 2017-01-30:
and there is small problem, when unplug keyboard from mac, and then insert back - message with device not connected repeats and only reset helps

closed 2017-06-29 by light2yellow

JeremyHi, 2017-07-19:
Still having detection error, this is my dev-detect log: 
[ckb-dev-detect-report.txt](https://github.com/mattanger/ckb-next/files/1160354/ckb-dev-detect-report.txt) 

It obviously looks like my mac (10.11 and 10.12) are detecting the keyboard, and I'm writing on it right now, but ckb is not picking it up.

JeremyHi, 2017-07-19:
Resolved: Restarted Mac without resuming windows option checked, on startup CKB worked as expected.