mattanger/ckb-next #51: Unable to open any Change color buttons

VoltaicLoki, 2017-01-30:
Hello,

I was able to get ckb working via building from git however, when I select the Change color button I get an error and the application window freezes.  I am running arch on Gnome Shell.  I have both a K70 LUX RGB keyboard and a Sabre RGB mouse.  This issue happens when I try to use the change color buttons on both devices.

The output I get from the command line is listed below.
                                                                                                                  
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile
Scanning  "/usr/bin/ckb-animations/ckb-gradient"
Scanning  "/usr/bin/ckb-animations/ckb-heat"
Scanning  "/usr/bin/ckb-animations/ckb-mviz"
Scanning  "/usr/bin/ckb-animations/ckb-pinwheel"
Scanning  "/usr/bin/ckb-animations/ckb-rain"
Scanning  "/usr/bin/ckb-animations/ckb-random"
Scanning  "/usr/bin/ckb-animations/ckb-ripple"
Scanning  "/usr/bin/ckb-animations/ckb-wave"
Downloaded new firmware list. 6 entries found.
libpng warning: iCCP: known incorrect sRGB profile

(ckb:17110): GLib-GObject-WARNING **: cannot register existing type 'GtkWidget'

(ckb:17110): GLib-GObject-WARNING **: cannot add class private field to invalid type '<invalid>'

(ckb:17110): GLib-GObject-WARNING **: cannot add private field to invalid (non-instantiatable) type '<invalid>'

(ckb:17110): GLib-GObject-CRITICAL **: g_type_add_interface_static: assertion 'G_TYPE_IS_INSTANTIATABLE (instance_type)' failed

(ckb:17110): GLib-GObject-WARNING **: cannot register existing type 'GtkBuildable'

(ckb:17110): GLib-GObject-CRITICAL **: g_type_interface_add_prerequisite: assertion 'G_TYPE_IS_INTERFACE (interface_type)' failed

(ckb:17110): GLib-CRITICAL **: g_once_init_leave: assertion 'result != 0' failed

(ckb:17110): GLib-GObject-CRITICAL **: g_type_add_interface_static: assertion 'G_TYPE_IS_INSTANTIATABLE (instance_type)' failed

(ckb:17110): GLib-GObject-CRITICAL **: g_type_register_static: assertion 'parent_type > 0' failed

(ckb:17110): GLib-GObject-WARNING **: cannot add private field to invalid (non-instantiatable) type '<invalid>'
fish: “ckb” terminated by signal SIGKILL (Forced quit)

tatokis, 2017-01-30:
Have a look at https://github.com/ccMSC/ckb/issues/461#issuecomment-259568681

light2yellow, 2017-01-30:
@VoltaicLoki this is a [known](https://bbs.archlinux.org/viewtopic.php?pid=1658017#p1658017) bug in Qt 5.7. To fix this either install Qt 5.8 or `qt5ct` package. It'll go away.

VoltaicLoki, 2017-01-30:
I just tried the suggestion.  I'm still getting the same behavior.

VoltaicLoki, 2017-01-30:
Thanks @light2yellow , using qt5ct fixed my issue.

closed 2017-01-30 by light2yellow