mattanger/ckb-next #53: K95 Platinum support

dodedodo33, 2017-02-01:
The keyboard is detected and connected to by ckb-daemon. 

The issues so far:

- The keyboard layout in ckb is incorrect, as it is the regular k95 layout where the k95 Platinum has a slightly different key-set (6 G-keys and the top rgb bar).
- More drastically: when starting the ckb client application the keyboard stops working. The animation that was already playing on the keyboard (HW profile) stays on but no key presses are registered until I quit the client and re-plug the keyboard.

dodedodo33, 2017-02-02:
Adding to this, the (testing) ckb client says there is an update available for my firmware. Which probably is a firmware for the regular k95 (LUX) as the keyboard firmware is up to date according to cue on windows.

frickler24, 2017-02-02:
@dodedodo33 Please ignore the message for the new firmware. Have a look at Issue #26. I have done an update on this, but the patch is not tested nor installed yet and your HW is not mentioned in the file.
What release number do you have currently and do you know the download URL?

dodedodo33, 2017-02-02:
@frickler24 thanks, I'll post an update on that later in the weekend, as I am currently away from home. So far the only thing done on k95 Platinum support is the addition of the device id to the K95 series in the testing branch. But as this keyboard is different in layout, firmware and maybe even protocol there is quite a bit more work left before it will function properly. I'm a bit short in time but I'd love to try some development on this toolkit. Only I have no idea where to start :stuck_out_tongue:

dodedodo33, 2017-02-05:
@frickler24 My current firmware version for the k95 Platinum is v1.07.123 I could not figure out what the download link is yet.

frickler24, 2017-02-05:
@dodedodo33  OK, Link is [K95RGBPLATINUM](http://www3.corsair.com/software/HID/K95RGBPLATINUM.zip) and veriosn of file is v1.07.123 as you mentioned above. 
I will bring it into my PR to issue #26. Because @mattanger is currently integrating it, here are the values, if I'm too late :)
Hopefully the version number is correct, because with this Firmware a completely new type is coming up.
The new line is 
`  Corsair K95PLATINUMRGB    1.07123 https://www3.corsair.com/software/HID/K95RGBPLATINUM.zip   0.2.7   FREYA_App_v1.07.123.bin         6d06b443cd85f4256d2b518b66b5c2d490d8089b678f9e34aecea97b13c5f5d4
`

tatokis, 2017-04-02:
@dodedodo33 Are you on linux? If so, can you please run ckb-dev-detect?

robryals, 2017-04-02:
I just got the K95 RBG Platinum as well and it's not being detected on my system (Mint 18).
I'm also seeing this in dmesg, not sure if it's relevant or not - 'usbhid 2-1.2:1.2: couldn't find an input interrupt endpoint' .
I've tried all of the troubleshooting tips.  The keyboard does work under Linux, just not recognized by ckb.
[ckb-dev-detect-report.gz](https://github.com/mattanger/ckb-next/files/888409/ckb-dev-detect-report.gz)

dodedodo33, 2017-04-02:
@tatokis I am, but not currently at home. I´ll try running the command when I get back tonight.

tatokis, 2017-04-02:
@dodedodo33 No need, the one posted by @robryals is enough. Thanks though :)

@robryals Is the K95 Platinum the only corsair device plugged in?

Additionally, I'd like you to install ckb-next from testing branch ( https://github.com/mattanger/ckb-next/archive/testing.zip ).
Make sure another keyboard is plugged in, as this ***will*** break input from your Corsair keyboard.

Once ckb-next from testing is up and running, open a terminal and run
```cd /dev/input/ && cat ckb0/* ; cd ckb1 && cat features fwversion model pollrate serial```
Report back with the output.

To restore functionality of your keyboard, close the GUI and run ```sudo systemctl stop ckb-daemon.service```.
You can then proceed to ```sudo systemctl disable ckb-daemon.service``` to prevent it from starting up on boot and breaking input, while ckb-next remains installed for other testing purposes.

robryals, 2017-04-02:
Yes, the K95 is the only Corsair device I have.

/dev/input/ckb1 0800F012AEFBA40457F61AABF5001943 Corsair Gaming K95 RGB PLATINUM Ke
10989
beta-v0.2.7
corsair k95 rgb bind notify fwversion fwupdate
0107
Corsair Gaming K95 RGB PLATINUM Ke
0 ms
0800F012AEFBA40457F61AABF5001943

tatokis, 2017-04-02:
Interesting, it exposes another USB interface. Not sure what it does, but I think it might be related to the hardware animations.

I can see a possible issue during init, which could be what's breaking things.
In the source code you extracted, open src/ckb-daemon/usb.h and on line 67 remove the `//` from `//#define DEBUG_USB`
https://github.com/mattanger/ckb-next/blob/master/src/ckb-daemon/usb.h#L67
Save the file, and recompile/reinstall.

If you use quickinstall, answer N to both
`Run the driver as a startup service [Y/n]?`
and `Start ckb now [Y/n]? `.

Once that's done, chek that the GUI is not running, open a terminal and run `sudo killall -9 ckb-daemon && sudo ckb-daemon`.

After it finishes starting up, give it a few seconds, and upload all of its output.

You should be able to then Ctrl+C to restore keyboard functionality again.

robryals, 2017-04-02:
This is all of the output I saw.
    ckb: Corsair RGB driver beta-v0.2.7
[I] Root controller ready at /dev/input/ckb0
[I] Connecting Corsair Gaming K95 RGB PLATINUM Ke at /dev/input/ckb1
[I] Starting input thread for /dev/input/ckb1
[E] os_usbsend (via firmware.c:15): Broken pipe
[W] _start_dev (device.c:24): Unable to load firmware version/poll rate
[I] Setup finished for /dev/input/ckb1

tatokis, 2017-04-03:
That's definitely not right, it should print the USB packets sent to the keyboard.
I've made some changes, and added further debugging.

You can either refetch testing and uncomment both 
`//#define DEBUG_USB`
`//#define DEBUG_USB_RECV`
and try again,
or you can try this precompiled binary (and hopefully it'll work).
`wget https://vps.tasossah.com/ckb-next/ckb-daemon && sudo killall ckb-daemon && sudo ./ckb-daemon

robryals, 2017-04-03:
I don't think it's ever getting to new debug lines.

    ckb: Corsair RGB driver beta-v0.2.7
[I] Root controller ready at /dev/input/ckb0
[I] Connecting Corsair Gaming K95 RGB PLATINUM Ke at /dev/input/ckb1
[I] Starting input thread for /dev/input/ckb1
[E] os_usbsend (via firmware.c:15): Broken pipe
[W] _start_dev (device.c:24): Unable to load firmware version/poll rate
[I] Setup finished for /dev/input/ckb1
[E] os_usbsend (via device_keyboard.c:38): Broken pipe
[I] Attempting reset...
[E] os_usbsend (via device_keyboard.c:38): Broken pipe
[E] os_usbsend (via device_keyboard.c:38): Broken pipe
[E] os_usbsend (via device_keyboard.c:38): Broken pipe

tatokis, 2017-04-06:
Alright, this clearly isn't going to work.
Please install wireshark `sudo apt-get install wireshark`
Load usbmon `sudo modprobe usbmon`
Run `sudo setfacl -m u:$USER:r /dev/usbmon*`

Start up wireshark.

Once it's up, under the capture label, you'll see many usbmon devices. Start hitting random keys on your K95 platinum. Note down the one that shows activity while you are typing on your K95.

After that, make sure both ckb and ckb-daemon are *NOT* running. While having another keyboard plugged in, double click on the usbmon device you noted down earlier, and then while typing *only* on the non-k95 keyboard, open a terminal and run `sudo ckb-daemon`.

Once it finishes its initialisation, Ctrl+C it.

Then stop the capture and go to file->save as, save the capture, and upload it.

robryals, 2017-04-06:
Here's the wireshark capture.

[k95platinum.zip](https://github.com/mattanger/ckb-next/files/904093/k95platinum.zip)

tatokis, 2017-04-16:
Hey guys, please try #156 and report back if it works.

robryals, 2017-04-16:
Success, thanks !

tatokis, 2017-06-07:
I have implemented some changes that might actually fix this without any ugly hacks.
Can someone please test this? https://github.com/mattanger/ckb-next/tree/k95p

dodedodo33, 2017-06-08:
@tatokis I'll be able to test this on saturday probably. Thanks for the update!

KazWolfe, 2017-06-08:
@tatokis Confirmed working here. The daemon is running and I am able to type, at least.

It also appears as though some color things are working as expected. Immediately after starting the new version, my keyboard immediately flashed into action, and WASD turned white. Similarly, keys turn white as I press them.

I can also manually set colors, add/remove effects, and a whole bunch of other things. Setting G1-G6 on the old keyboard layout will change G1-G6 on the platinum as well. G7-G18 do nothing. I can also set the brightness key color, but not the winlock color (though that may just be an artifact of the keyboard). -- it just stays white). At this time, I can also not set the macro colors.

tatokis, 2017-06-10:
@KazWolfe interesting.
Would you mind posting the output of ckb-daemon here?
On Linux, I assume systemctl status can provide that info.

And if I understood correctly, the platinum does not have the G7~G18 keys at all.
What do you mean by macro colours though?

KazWolfe, 2017-06-10:
@tatokis Will post logs when I get back to the office.

The K95P has no G7-G18, you are correct. The buttons (when assigned colors) just are ignored. I was sort of hoping they'd jump over to the top LED bar or similar, but oh well. On that note, the top LED bar doesn't work at all, but that is expected.

As for the macro colors, I'm referring to the M1-M3 buttons on the K95. I thought it would automatically map to the "profile" button/indicator, but they did not. My apologies if I used the wrong terminology.

There are still a few bugs with the keyboard dropping, but I think that's related to an existing USBHID/xhci bug with my dock. Unplugging/replugging resolves it for the most part.

KazWolfe, 2017-06-12:
@tatokis Logs from the `ckb-daemon` service:

```
-- Logs begin at Sat 2017-06-10 22:01:22 PDT. --
Jun 12 00:10:47 ubuntu ckb-daemon[24231]: [I] Removed device path /dev/input/ckb1
Jun 12 09:00:41 ubuntu ckb-daemon[24231]: [I] >>>vendor = 0x1b1c, product = 0x1b2d, path = /dev/bus/usb/003/008, syspath = /sys/devices/pci0000:00/0000:00:1d.6/0000:06:00.0/0000:07:01.0/0000:09:00.0/0000:0a:04.0/0000:0c:00.0/0000:0d:01.0/0000:0e:00.0/usb3/3-1/3-1.7/3-1.7.3
Jun 12 09:00:41 ubuntu ckb-daemon[24231]: [I] Connecting Corsair Gaming K95 RGB PLATINUM Keyboard at /dev/input/ckb1
Jun 12 09:00:41 ubuntu ckb-daemon[24231]: [I] Connecting Corsair Gaming K95 RGB PLATINUM Keyboard at /dev/input/ckb1
Jun 12 09:00:41 ubuntu ckb-daemon[24231]: [I] claiming interfaces. name=Corsair Gaming K95 RGB PLATINUM Keyboard, serial=[redacted], firmware=0107; Got >> 3<< as ep_str
Jun 12 09:00:41 ubuntu ckb-daemon[24231]: [I] claiming 3 endpoints
Jun 12 09:00:41 ubuntu ckb-daemon[24231]: [I] Starting input thread for /dev/input/ckb1
Jun 12 09:00:41 ubuntu ckb-daemon[24231]: [E] os_usbsend (via firmware.c:15): Broken pipe
Jun 12 09:00:41 ubuntu ckb-daemon[24231]: [W] _start_dev (device.c:38): Unable to load firmware version/poll rate
Jun 12 09:00:41 ubuntu ckb-daemon[24231]: [I] Setup finished for /dev/input/ckb1
```

AnatomicJC, 2017-06-13:
I bought a K95 platinium yesterday, master branch doesn't work, but I can confirm than the k95p branch works :) Many thanks @tatokis

By the way, I'm a Debian Linux user, if you need some additional debug logs, I can help.

tatokis, 2017-06-15:
@KazWolfe (or whoever does it first), could you perhaps fetch the latest code from the newdev branch, uncomment the two DEBUG lines in usb.h and compile? (./qmake-auto && make)

Then once that's done, stop the systemd service, close the ckb GUI completely, and run the new daemon manually from the source tree.
`sudo ./bin/ckb-daemon`

It will print the packets sent to and from the device, so please share the output.
Just make sure you do not run the GUI during any of this, as it will spam the console with the RGB LED packets.

frickler24, 2017-06-15:
And here we have at least a link to #220 again (as @KazWolfe wrote 
> Another quick bug I noticed is that ckb keeps thinking my keyboard can be updated to FW version 2.05 or something like that.)
I believe the the old algorithm sees the K95P as a K95RGB and find v2.05 for it.

KazWolfe, 2017-06-16:
@tatokis Output of debug branch + debug statements:

```
    ckb: Corsair RGB driver beta-v0.2.8
[I] Root controller ready at /dev/input/ckb0
[I] >>>vendor = 0x1b1c, product = 0x1b2d, path = /dev/bus/usb/003/009, syspath = /sys/devices/pci0000:00/0000:00:1d.6/0000:06:00.0/0000:07:01.0/0000:09:00.0/0000:0a:04.0/0000:0c:00.0/0000:0d:01.0/0000:0e:00.0/usb3/3-1/3-1.7/3-1.7.3
[I] Connecting Corsair Gaming K95 RGB PLATINUM Keyboard at /dev/input/ckb1
[I] Connecting Corsair Gaming K95 RGB PLATINUM Keyboard at /dev/input/ckb1
[I] claiming interfaces. name=Corsair Gaming K95 RGB PLATINUM Keyboard, serial=02004023AECB9CC556E710D9F5001946, firmware=0107; Got >> 3<< as ep_str
[I] claiming 3 endpoints
[I] Starting input thread for /dev/input/ckb1
[E] os_usbsend (via firmware.c:15): Broken pipe
[W] _start_dev (device.c:38): Unable to load firmware version/poll rate
[I] Setup finished for /dev/input/ckb1
^C
[I] Caught signal 2
[W] os_usbsend (via device_keyboard.c:66): Sent 07 04 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[W] os_usbsend (via device_keyboard.c:69): Sent 07 04 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[W] os_usbsend (via device_keyboard.c:98): Sent 07 40 1e 00 00 80 01 80 02 80 03 80 04 80 05 80 06 80 07 80 08 80 09 80 0a 80 0b 80 0c 80 0d 80 0e 80 0f 80 10 80 11 80 12 80 13 80 14 80 15 80 16 80 17 80 18 80 19 80 1a 80 1b 80 1c 80 1d 80 
[W] os_usbsend (via device_keyboard.c:98): Sent 07 40 1e 00 1e 80 1f 80 20 80 21 80 22 80 23 80 24 80 25 80 26 80 27 80 28 80 29 80 2a 80 2b 80 2c 80 2d 80 2e 80 2f 80 30 80 31 80 32 80 33 80 34 80 35 80 36 80 37 80 38 80 39 80 3a 80 3b 80 
[W] os_usbsend (via device_keyboard.c:98): Sent 07 40 1e 00 3c 80 3d 80 3e 80 3f 80 40 80 41 80 42 80 43 80 44 80 45 80 46 80 47 01 48 80 49 80 4a 80 4b 80 4c 80 4d 80 4e 80 4f 80 50 80 51 80 52 80 53 80 54 80 55 80 56 80 57 80 58 80 59 80 
[W] os_usbsend (via device_keyboard.c:98): Sent 07 40 1e 00 5a 80 5b 80 5c 80 5d 80 5e 80 5f 80 60 08 61 80 62 80 63 80 64 80 65 80 66 80 67 80 68 80 69 80 6a 80 6b 80 6c 80 6d 80 6e 80 6f 80 70 80 71 80 72 80 73 80 74 80 75 80 76 80 77 80 
[W] os_usbsend (via device_keyboard.c:98): Sent 07 40 1e 00 78 80 79 80 7a 80 7b 80 7c 80 7d 80 7e 80 7f 80 80 80 81 80 82 80 83 80 84 09 85 0a 86 0b 87 0c 88 80 89 80 8a 80 8b 80 8c 80 8d 80 8e 80 8f 80 90 80 91 80 92 80 93 80 94 80 95 80 
[W] os_usbsend (via device_keyboard.c:98): Sent 07 40 02 00 96 80 97 80 7a 80 7b 80 7c 80 7d 80 7e 80 7f 80 80 80 81 80 82 80 83 80 84 09 85 0a 86 0b 87 0c 88 80 89 80 8a 80 8b 80 8c 80 8d 80 8e 80 8f 80 90 80 91 80 92 80 93 80 94 80 95 80 
[W] os_usbsend (via device_keyboard.c:102): Sent 07 05 02 00 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[I] Disconnecting /dev/input/ckb1
[I] Removed device path /dev/input/ckb1
[I] Closing root controller
[I] Removed device path /dev/input/ckb0
```

Changes made to file:

```
index 40edad4..4410232 100644
--- a/src/ckb-daemon/usb.h
+++ b/src/ckb-daemon/usb.h
@@ -107,11 +107,11 @@
 
 ///
 /// uncomment the following Define to see USB packets sent to the device
-// #define DEBUG_USB
+#define DEBUG_USB
 
 ///
 ///uncomment to see USB packets sent from the device
-// #define DEBUG_USB_RECV
+#define DEBUG_USB_RECV
 
 ///
 /// \brief vendor_str Vendor/product string representations
```

followed by `./qmake-auto && make`

frickler24, 2017-06-17:
@KazWolfe I assume you are running linux?
@tatokis We should have a look at os_usbsend. There you find the line
`if(kb->fwversion >= 0x120 && !is_recv){` which switches between two different communication methods.
The Firmware used by 0x1b2d is 1.07, so we use an older mechnism for sending data (ioctl USBDEVFS_CONTROL instead of USBDEVFS_BULK). Maybe that's a hint.

BTW: The duplicate message about "Connecting" has gone when I merge #25.

tatokis, 2017-06-17:
@frickler24 You are a bit too late I am afraid, https://github.com/mattanger/ckb-next/blob/newdev/src/ckb-daemon/usb_linux.c#L70
and https://github.com/mattanger/ckb-next/blob/newdev/src/ckb-daemon/usb.h#L156

The Platinum is working, I am just trying to figure out why firmware.c returns broken pipe.

@KazWolfe Interesting thing is that the recv lines are not being printed.
It probably happened when frickler ended up overwriting my debugging changes, I'll have a look later and report back.

Oh, and for future reference, the daemon picks which protocol to use depending on the firmware version reported by the device on initial connection, which explains why the daemon was listening on the wrong endpoint.

The firmware version can be found without sending any packets if you run lsusb -vvv and grep the bcdDevice line.

In my case, it's 
`  idVendor           0x1b1c Corsair
  idProduct          0x1b13 
  bcdDevice            2.05`

frickler24, 2017-06-17:
@tatokis As I remember, usbrecv is used to load hardware information from the device (from getfwversion()).
If the first attempt to read the firmware and pollrate is not successful (returning -1 and skipping your debug lines), then there is no further reading.

tatokis, 2017-06-17:
@frickler24 First it checks what the firmware version reported by the OS is, and depending on that it configures the endpoints and then asks sends the init packet asking the firmware version.

It's why it compares the firmware version it replies, with the firmware the OS reported.
https://github.com/mattanger/ckb-next/blob/master/src/ckb-daemon/firmware.c#L43

frickler24, 2017-06-17:
@tatokis You are right - if you ever reach this line. The "broken pipe" comes from line 15, where the first usbrecv call is done. There you get the error and return with -1.

tatokis, 2017-06-17:
@frickler24 yeah, I only just realised that, thank you. This needs further investigation. Perhaps it doesn't like the ctrl transfer.

tatokis, 2017-06-18:
Quick update:
CUE seems to send everything using URB Interrupts.
ckb on the other hand, uses URB Controls for these functions.

I tried forcing os_usbsend and os_usbrecv to use URB_BULK transfers, as this is how interrupt transfers are implemented in usbfs.
The keyboard would reply to the packets, but the daemon would process the wrong packet as a reply instead.

For example
```
[W] os_usbsend (via firmware.c:15): Sent 0e 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[W] os_usbrecv (via firmware.c:15): Recv 00 10 00 00 00 00 00 00 da 95 40 00 00 00 00 00 70 be 0c b9 01 7f 00 00 80 4a 62 00 00 00 00 00 8f 6c 01 75 fe 7f 00 00 69 07 3a b9 01 00 00 00 21 09 00 02 00 00 01 00 f4 01 00 00 01 7f 00 00 
[E] getfwversion (firmware.c:18): Bad input header
```

Since the packet does get sent, and the reply can be seen using wireshark (instead of a broken pipe being returned), I was thinking we could try this on the K95 Platinum, and if we do get a reply then we can then investigate further.
Any suggestions?

tatokis, 2017-07-23:
Can anyone with a K95 Platinum on linux test this?
https://vps.tasossah.com/ckb-next/ckb-next-libusbtest.c

Can be compiled with `gcc -o ckb-next-libusbtest ckb-next-libusbtest.c -lusb-1.0`
And executed as such `sudo ./ckb-next-libusbtest`

Please make sure that ckb-daemon is NOT running while trying to run the above.

dodedodo33, 2017-07-24:
I've tried running it yesterday evening, got an error saying the binary could not be run... Even though I see nothing out of the ordinary during the build process (a more verbose output even seems clean). Anything I could be missing here?

dodedodo33, 2017-07-24:
Erm, never mind. It seems the issue is that it is not marked executable. As it is on a drive that is mounted with permissions mask... I'll try again when I get back home and run it from a different drive 👍

tatokis, 2017-08-02:
Any updates on this?

sgtsnookums, 2017-08-14:
I just came across this today, I can give my results from the libusbtest:

>Please do not press any keys while this is running
>Init: 0
>Hotplug: 1
>DevOpen: 1
>DrvDetach0: 0
>DrvDetach1: 0
>ConfGet: 0
>bConf: 1
>If1Claim: 0
>If2Claim: 0
>URB Interrupt Question: 0e 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
>URB Interrupt Answer: 0e 01 00 00 01 01 00 01 07 01 04 00 1c 1b 2d 1b 01 01 00 00 c0 ff 40 00 00 00 00 00 02 00 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
>InterruptRet: 0
>DrvAttach0: 0
>DrvAttach1: 0

I'm on Fedora 26, using the copr repo listed on the main project page in case that matters.

tatokis, 2017-08-14:
@sgtsnookums Thank you very much for that, I am now aware of what needs to be done for the K95 Platinum to be fully supported.

Until then, you can compile the code from the newdev branch to get your device mostly working.

Extarys, 2017-09-20:
@tatokis Any update for this amazing keyboard :)

tatokis, 2017-09-21:
@Extarys unfortunately, no.
Other than the existing basic support for it in newdev, I need to figure out how to control the lights on top, then remove the excess G keys.
And then to finish it off, since it doesn't seem to respond to URB Ctrls it'll have to wait until libusb is implemented in the daemon for Linux, not sure what can be done about Mac.

I have mostly been busy recently with two other issues, #162 and #286. Once these two are closed, I'll have to look into how to control the extra lights on the K95 Platinum.

Extarys, 2017-09-21:
Thanks for your kind reply.

That is quite sad. I'm not good enough to help I think. But I'll try to find some help.

jerryaldrichiii, 2017-09-22:
+1 to @Extarys comment. I don't feel like I'm skilled enough to help, but would love to contribute to get the K95 Platinum logo/light bar functioning.

Thanks for all your work @tatokis! Let me know if there is anything I can do to help.

Extarys, 2017-09-23:
Thanks @jerryaldrichiii I'm verry glad that I'm not alone feeling that way :P

Vagr9K, 2017-11-23:
Hey guys!

I'm thinking of picking up a K95 Platinum, but I have a couple of questions about its current state.

1) Is it possible to set the logo/light bar lighting in Windows, or it will be stuck in rainbow mode under Linux?
2) Are there any "connectivity drops" requiring system reboots/keyboard reinserts just to type?
3) Which Corsair keyboard has the best and most reliable support right now?

I'd happily buy and test it myself, but the 150-200$ price tag and inability to return (due to my location) force me to ask quite a lot of questions...

That being said, I really appreciate you guys bringing support for those devices to Linux :)
This is a really cool project!

KazWolfe, 2017-11-24:
@Vagr9K Currently:

1. The logo/light bar will be in colorful mode when ckb hasn't claimed control over the keyboard. Otherwise, it's off. There is currently still no way to set it that I am aware of.
2. I have not seen connectivity drops for typing, but I have seen lighting effect drops. These are usually resolved by restarting the daemon service.

Otherwise, the K95 works almost perfectly, given the current device capabilities.

jerryaldrichiii, 2017-11-24:
Same as @KazWolfe, the logo/lightbar cannot be set in Linux but other than that everything works as expected with only occasional restarts of the daemon necessary (once or twice every few months).

I personally prefer the effects that this repo gives me instead of the ones in Windows...just wish the logo/lightbar could be included :smile:

Vagr9K, 2017-11-24:
Thanks guys!

tatokis, 2017-11-24:
For those running linux, if anyone is willing to spend time to get the other lights working, join #ckb-next on freenode ( http://webchat.freenode.net/?channels=%23ckb-next&uio=d4 ) and we can go from there.

tatokis, 2017-11-27:
With the latest code from master, you should now be able to control the lightbar.

jerryaldrichiii, 2017-11-27:
Confirmed! Things are working now using the `ckb-next-latest-git` Arch package!

Huge thanks to @tatokis for many late nights/early mornings working with me to make the light bar work!

tatokis, 2017-11-27:
Confirmed working with the new firmware too.
Only issue remaining is the inability to respond to URB Ctrls, but since that's not an issue specific to this device, I can finally mark this as closed.

Again, thank you to everyone for your help.

closed 2017-11-27 by tatokis