mattanger/ckb-next #54: Testing: Harpoon RGB not connected macOS

Camoleopard, 2017-02-01:
So in the latest testing branch, support for the Harpoon gaming mouse was added. After testing this, I found that it wasn't showing up in CKB, I thought I should restart, and when I did, my mouse no longer worked, but it did show up in CKB now. I couldn't configure the colour because it was designed for a different mouse for testing purposes. I then went into the mouse settings in system preferences, and disconnected and reconnected my mouse, and it started working again, but it no longer showed up in CKB. Using the latest testing build.

CGA1123, 2017-05-11:
Given that 1a2dfe2 has reverted 84e195c, what is the 'proper support' that would need to be added?

tatokis, 2017-05-11:
Packet captures to see why it fails during init would be helpful. 
It's probably a minor difference in the protocol somewhere. 

The device buttons are different too, so there are additional modifications needed in the gui.

tatokis, 2017-06-10:
Could someone perhaps tell me what the firmware version for this device is?

mshade, 2017-07-04:
@tatokis I have the device; how can I find the firmware version for you?

Rested, 2017-07-17:
@light2yellow 
Here are the results of USB prober:
```
Full Speed device @ 26 (0x14100000): .............................................   Composite device: "Corsair Gaming HARPOON RGB Mouse"
    Port Information:   0x001a
           Not Captive
           Attached to Root Hub
           External Device
           Connected
           Enabled
    Number Of Endpoints (includes EP0):   
        Total Endpoints for Configuration 1 (current):   4
    Device Descriptor   
        Descriptor Version Number:   0x0200
        Device Class:   0   (Composite)
        Device Subclass:   0
        Device Protocol:   0
        Device MaxPacketSize:   64
        Device VendorID/ProductID:   0x1B1C/0x1B3C   (CORSAIR MEMORY INC.)
        Device Version Number:   0x0101
        Number of Configurations:   1
        Manufacturer String:   1 "Corsair"
        Product String:   2 "Corsair Gaming HARPOON RGB Mouse"
        Serial Number String:   3 "0801D03FAECDA40657729135F5001943"
    Configuration Descriptor (current config)   
        Length (and contents):   84
            Raw Descriptor (hex)    0000: 09 02 54 00 03 01 00 A0  96 09 04 00 00 01 03 01  
            Raw Descriptor (hex)    0010: 02 00 09 21 11 01 00 01  22 3A 00 07 05 81 03 08  
            Raw Descriptor (hex)    0020: 00 08 09 04 01 00 01 03  00 00 00 09 21 11 01 00  
            Raw Descriptor (hex)    0030: 01 22 91 00 07 05 82 03  40 00 01 09 04 02 00 01  
            Raw Descriptor (hex)    0040: 03 00 00 00 09 21 11 01  00 01 22 19 00 07 05 03  
            Raw Descriptor (hex)    0050: 03 40 00 01 
        Number of Interfaces:   3
        Configuration Value:   1
        Attributes:   0xA0 (bus-powered, remote wakeup)
        MaxPower:   300 mA
        Interface #0 - HID/Boot Interface   
            Alternate Setting   0
            Number of Endpoints   1
            Interface Class:   3   (HID)
            Interface Subclass;   1   (Boot Interface)
            Interface Protocol:   2
            HID Descriptor   
                Descriptor Version Number:   0x0111
                Country Code:   0
                Descriptor Count:   1
                Descriptor 1   
                    Type:   0x22  (Report Descriptor)
                    Length (and contents):   58
                        Raw Descriptor (hex)    0000: 05 01 09 02 A1 01 09 01  A1 00 05 09 19 01 29 05  
                        Raw Descriptor (hex)    0010: 15 00 25 01 95 05 75 01  81 02 95 01 75 03 81 01  
                        Raw Descriptor (hex)    0020: 05 01 09 30 09 31 09 38  15 81 25 7F 75 08 95 03  
                        Raw Descriptor (hex)    0030: 81 06 95 03 75 08 81 01  C0 C0 
                    Parsed Report Descriptor:   
                          Usage Page    (Generic Desktop) 
                          Usage (Mouse)    
                              Collection (Application)    
                                Usage (Pointer)    
                                    Collection (Physical)    
                                      Usage Page    (Button) 
                                      Usage Minimum...........    (1)  
                                      Usage Maximum...........    (5)  
                                      Logical Minimum.........    (0)  
                                      Logical Maximum.........    (1)  
                                      Report Count............    (5)  
                                      Report Size.............    (1)  
                                      Input...................   (Data, Variable, Absolute, No Wrap, Linear, Preferred State, No Null Position, Bitfield) 
                                      Report Count............    (1)  
                                      Report Size.............    (3)  
                                      Input...................   (Constant, Array, Absolute) 
[... 131 lines cut ...]
                    Type:   0x22  (Report Descriptor)
                    Length (and contents):   25
                        Raw Descriptor (hex)    0000: 06 C2 FF 09 04 A1 01 15  00 26 FF 00 95 40 75 08  
                        Raw Descriptor (hex)    0010: 09 02 B1 02 09 04 91 02  C0 
                    Parsed Report Descriptor:   
                          Usage Page    (Vendor defined 194) 
                          Usage 4 (0x4)    
                              Collection (Application)    
                                Logical Minimum.........    (0)  
                                Logical Maximum.........    (255)  
                                Report Count............    (64)  
                                Report Size.............    (8)  
                                Usage 2 (0x2)    
                                Feature.................   (Data, Variable, Absolute, No Wrap, Linear, Preferred State, No Null Position, Nonvolatile, Bitfield) 
                                Usage 4 (0x4)    
                                Output..................   (Data, Variable, Absolute, No Wrap, Linear, Preferred State, No Null Position, Nonvolatile, Bitfield) 
                              End Collection     
            Endpoint 0x03 - Interrupt Output   
                Address:   0x03  (OUT)
                Attributes:   0x03  (Interrupt)
                Max Packet Size:   64
                Polling Interval:   1 ms

```
Willing to work on this myself but if you could give me some pointers that would be great.

mashedkeyboard, 2017-07-23:
Just got one of these mice, would love to use ckb-next on the Linux part of my dualboot. I've got the ckb-dev-detect report below:

```
ckb-dev-detect - Sun Jul 23 10:27:33 BST 2017

Kernel: Linux quetzel 4.4.0-83-generic #106-Ubuntu SMP Mon Jun 26 17:54:43 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux

OS Version:
Distributor ID:	Ubuntu
Description:	Ubuntu 16.04.2 LTS
Release:	16.04
Codename:	xenial

USB Devices:

Bus 005 Device 009: ID 1b1c:1b3c Corsair 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1b1c Corsair
  idProduct          0x1b3c 
  bcdDevice            1.01
  iManufacturer           1 Corsair
  iProduct                2 Corsair Gaming HARPOON RGB Mouse
  iSerial                 3 0F026024AECE00A6574F1798F5001946
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           84
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              300mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      58
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               8
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 No Subclass
      bInterfaceProtocol      0 None
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     145
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 No Subclass
      bInterfaceProtocol      0 None
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      25
          Report Descriptor: (length is 25)
            Item(Global): Usage Page, data= [ 0xc2 0xff ] 65474
                            (null)
            Item(Local ): Usage, data= [ 0x04 ] 4
                            (null)
            Item(Main  ): Collection, data= [ 0x01 ] 1
                            Application
            Item(Global): Logical Minimum, data= [ 0x00 ] 0
            Item(Global): Logical Maximum, data= [ 0xff 0x00 ] 255
            Item(Global): Report Count, data= [ 0x40 ] 64
            Item(Global): Report Size, data= [ 0x08 ] 8
            Item(Local ): Usage, data= [ 0x02 ] 2
                            (null)
            Item(Main  ): Feature, data= [ 0x02 ] 2
                            Data Variable Absolute No_Wrap Linear
                            Preferred_State No_Null_Position Non_Volatile Bitfield
            Item(Local ): Usage, data= [ 0x04 ] 4
                            (null)
            Item(Main  ): Output, data= [ 0x02 ] 2
                            Data Variable Absolute No_Wrap Linear
                            Preferred_State No_Null_Position Non_Volatile Bitfield
            Item(Main  ): End Collection, data=none
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x03  EP 3 OUT
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
Device Status:     0x0000
  (Bus Powered)
```
As with @Rested, I'd be more than willing to help out with this!

tatokis, 2017-08-03:
With the introduction of  https://github.com/mattanger/ckb-next/commit/9d8889ce2567d0206184f8971100d2997e643e97 and https://github.com/mattanger/ckb-next/commit/0d9f0a66c0ebe449c4693812ca160d22a7ae459b there should be support for the Harpoon RGB on the newdev branch.
https://github.com/mattanger/ckb-next/tree/newdev

Testing would be appreciated.

Electr0Fi, 2017-08-05:
@tatokis We're slowly getting there!

tatokis, 2017-10-14:
Full support for the Harpoon should be now be in master.

closed 2017-10-14 by tatokis