mattanger/ckb-next #56: CKB Not recognizing Scimitar RGB

Mekvenner, 2017-02-03:
I have a scimitar RGB that I purchased this last christmas and installed ckb on my iMac. Initially ckb work perfectly, it recognized my mouse (after having updated the firmware with my windows laptop), and allowed me to create profiles and the like exactly as intended. The first time I restarted my computer things started to get fucky, ckb would recognize my Scimitar but whenever I clicked (left or right) the cursor would be sent to the upper left corner of the screen. Super weird problem that could potentially be fixed by a series of restarts. I got fed up with this and decided to uninstall and try a binary install, so I installed Xcode, installed QT 5.8, and used the quick install. This has gotten rid of the top left corner problem, however now CKB won't recognize the mouse at all, I can still use it to the extent of the left and right click but I can't map the number pad on the side. I've tried the Mattranger version as well as version 0.2.6 but as of now I cannot get ckb to recognize my mouse anymore.

JarrenJ, 2017-02-03:
@Mekvenner this happens to me quite often as well, what seems to help me is to quit CKB, unplug your mouse from your computer. Do a full shut down turn it back on, when it is on plugin your mouse back in and reopen CKB it should make CKB recognize your mouse again.

Mekvenner, 2017-02-03:
I have tried this in many different orders including the one mentioned above, nothing has successfully made CKB recognize my scimitar. Considering it did work at one point, is there a way to completely remove all traces of ckb from my computer to try a fresh re-install? (apart from the standard "drag it all in the trash")

light2yellow, 2017-02-04:
@Mekvenner it's all in the README. https://github.com/mattanger/ckb-next#uninstalling-1

Kodah, 2017-04-26:
Did you get this to work? I can't get my Scimitar to be detected either.

closed 2018-01-21 by tatokis