mattanger/ckb-next #57: Not an Issue but a suggestion

JarrenJ, 2017-02-03:
So, in the issue below about adding in a donation link I replied that I could work on a site for ckb-next. @mattanger , I know you said you would rather produce a stable release first; however, C++ is not my forte, but html, css, javascript, etc are. I have the basic layout of the site down just adding in more information. If you have any questions or would like to see a very early version of it just let me know. The site is still very new, I have only put like 5 hours or so into it so it isn't great but, it should be ready if you ever decide to use it.

pablobh, 2017-03-06:
I can help you with the design+ coding of the website if you need any help. Just let me know and I'll be glad to do it!

frickler24, 2017-03-07:
@pablobh thanks for your offer!
@JarrenJ How far have you come with your work on the website? Is there anything @pablobh can do?

gregf, 2017-03-08:
I can provide hosting for anything you guys need as well. Just let me know what distro/bsd you want and how much ram/disk space you need and provide me with a ssh key and I'll hook you guys up.

guenniman, 2017-03-09:
I can do the same as a Mirror of you want to, it's the least I can do to support this project

JarrenJ, 2017-03-17:
@pablobh @frickler24 @guenniman @gregf --- Hi guys, thanks for offering to help. I have not been able to work on the site right now as my job has kind of taken off for the better. My free time is very limited and what free time I do have I like to spend it with my family. That being said if one you would like to try to take over this project for the time being go for it. I will be on Spring Break in Arizona w/ family for the next week or so. When I get back I will try top rearrange my schedule to allow for my time towards this website.
~~~~~~~~~~~~~~~~~
If you do take it over right now, this is what I currently have or am working on:

- Using Wordpress to utilize a OnePage theme.

- Different sections for Branches, community, lead developers, etc. 

- If you have any additional ideas just reply them here or email me at -- j.jarren@yahoo.com -- !
~~~~~~~~~~~~~~~~~

P.S:
- Sorry it took so long to get back yo you guys, as I said above I have been quite busy.

JarrenJ, 2017-03-25:
UPDATE: I get back from Spring Break @ 1:00am my time. I have put a little more work in the site. According to my original post I said I was using WordPress. I have since scraped all of that and am currently hard coding it. I will have different sections keeping with the one page theme. If you have any ideas or would like to help out. Reply to this thread. **Thanks for the support**

frickler24, 2017-04-01:
@JarrenJ ckb-next.org seems to lead just to the github pages, so I think, you have not launched you work yet.
One topic should be documentation of the code, as you can find [here for the last docs](https://frickler24.de/doxygen/)

JarrenJ, 2017-04-01:
I was unaware that there was already a domain name registered under CKB Next. I have not published my work yet because it is still in early stages of development.

JarrenJ, 2017-04-07:
@frickler24 and anyone else. I have the main shell of website done. If you would like to see it reply back to me and if I am at my computer I will launch it using a program called ngrok.

If anyone is still interested in helping it would be most appreciated.

pablobh, 2017-04-07:
I'm still here and wanting to help. Just let me know if you guys need anything.

frickler24, 2017-04-08:
My topic is the documentation. I am interested in building a (CI-) path to keep the documentation of the master and the testing branch up-to-date.

JarrenJ, 2017-04-09:
How is the website related to the master and testing branch?

light2yellow, 2017-04-09:
@JarrenJ I suppose we would like to host the documentation on the website and have the access to both master and testing branch variants.

pablobh, 2017-04-17:
Any updates on this topic?

JarrenJ, 2017-05-13:
@mattanger , @frickler24 , @light2yellow The website is finished. I finished it yesterday. The only thing that remains is adding a PHP function to take information that has been submitted and make a file with it. However, I am not very good with PHP took me all of 1 hour just to get a Database up lol.

--- I would like some help documenting it to github. However, I am not very savvy with Github either. I would appreciate some help in that regard.

--Thx.

JarrenJ, 2017-05-21:
Well someone respond with any info.... IE: i want to see site; yes i still want site; etc. If I don't get any responses I will just use it for a different project or a personal site.

closed 2017-05-24 by JarrenJ