mattanger/ckb-next #59: Setting the autorepeat delay/rate

tdy, 2017-02-04:
Just wondering if there's a ckb counterpart to something like `xset r rate 200 30`. Not sure if I overlooked it or if it's not implemented.

frickler24, 2017-02-05:
@tdy I had a look into the daemon, but currently there is no function to control the autorepeat rate.

tatokis, 2017-02-05:
Doesn't X still have the upper hand? Ckb just creates a virtual input 
device using the functions from the kernel.
I am sure I can control the repeat rate using the configuration in 
unity-control-center.

frickler24, 2017-02-05:
@tatokis @tdy Oh, you are right! Too much hacking today...
In KDE it is controlled by the config also.

tdy, 2017-02-06:
Ah sorry, I should've given more context. Currently I control it in X with `xset` in .xinitrc. It's just that any time I launch the ckb gui, my autorepeat settings get reset. That's why I was wondering if there was some kind of setting inside ckb that was causing it.

tdy, 2017-02-06:
Also noticed the same issue with my xmodmap settings getting reset, but I noticed there were modifier settings in ckb.

frickler24, 2017-02-06:
@tdy Interesting, because there is some code for auto-repeat handling, but just for macOS. for linux I could find just one line corresponding to auto repeat: 
```
input_linux.c, uinputopen():
...        // Eanble autorepeat
        ioctl(fd, UI_SET_EVBIT, EV_REP);
...
```
with a look into a [short tutorial](http://gkos.com/www2/manuals/Linux/dash_jan_tip.pdf) there seems to be no error.

closed 2018-01-21 by tatokis