mattanger/ckb-next #61: CKB-next doesn't remember settings after restart

MaxHaydenChiz, 2017-02-04:
I'm using Ubuntu with a K95 RGB and upgraded to CKB-next to try to fix a problem that CKB was having, but it only partially fixed my issue.

After quitting CKB-next and restarting it (or rebooting or logging out and back in), it always sets the keyboard back to the CKB demo profile. Unlike with the original CKB, it still remembers my custom profile, but I have to select it manually and I can't seem to get rid of or change the demo or HwProfile1 (they just come back like they were before). Furthermore, the custom profile loses all the animation settings and only remembers key colors.

What do I need to do to figure out why this is happening and help get it fixed? I would think it would be a permission issue on some configuration file or something else that is keeping it from saving changes, but I don't know enough about the software to know what to look for.

tatokis, 2017-02-04:
First of all, close the ckb-next gui.
Once you do that, open a terminal, install strace, and run ```strace -e open ckb 2>&1 | grep "\.config"```.
That should launch ckb. Once it opens, try deleting the defaults that won't go away. And then close ckb from the menu again.

Once that is done, please attach the output of the strace command.

MaxHaydenChiz, 2017-02-04:
open("/home/mchiz/.config/dconf/user", O_RDONLY) = 9
open("/home/mchiz/.config/ibus/bus/04871c9232d50bef7879d71453eb7df5-unix-0", O_RDONLY|O_CLOEXEC) = 13
open("/home/mchiz/.config/ibus/bus/04871c9232d50bef7879d71453eb7df5-unix-0", O_RDONLY) = 16
open("/home/mchiz/.config/ckb/ckb.conf", O_RDONLY|O_CLOEXEC) = 19
open("/home/mchiz/.config/ibus/bus/04871c9232d50bef7879d71453eb7df5-unix-0", O_RDONLY) = 32

MaxHaydenChiz, 2017-02-04:
So, based on that output, I saw what the problem was. The ckb config folder and ckb.conf were owned by root. Once I changed the owner and the group to my user, everything works fine. No clue why the install set the permissions like that though. Thanks for the help.

tatokis, 2017-02-04:
You probably ran ckb as root originally, and it created the config directory with root as the owner.

I'll reopen this for now, as I'd like the GUI to be able to warn users if it fails to write the config to disk, and possibly give a reason as to why it failed.

frickler24, 2017-02-05:
I'll have a look at it.

light2yellow, 2017-03-01:
@tatokis ckb-next prints something like "ckb-next must not run as root" and exits. See https://github.com/light2yellow/ckb-next/commit/4d372d8f3a765b6e248ee3864d36cf3e61f76e46. It is impossible to launch it as root.

light2yellow, 2017-03-03:
Yeah, it's here https://github.com/mattanger/ckb-next/blob/master/src/ckb/main.cpp#L165
If we suppose that GUI was launched as root then there's a bug in main.cpp. Otherwise we need a repro from @MaxHaydenChiz.

frickler24, 2017-03-03:
Or it is just a backup file recovered with wrong UID or a too wide find --exec chmod or or or...
We should close the issue when #90 is merged into testing.

tatokis, 2018-01-21:
https://github.com/ckb-next/ckb-next/issues/4

closed 2018-01-21 by tatokis