mattanger/ckb-next #62: Can't compile any builds (newdev or ckb-next)

shibatan, 2017-02-04:
Mac OS X 10.10.5, K70 LUX, QT 5.6, Xcodes command line tools installed. Already connected to a Windows PC and updated the firmware. 
After hours of reading and troubleshooting, I still cannot get any version to build. The .pkg version of the old ckb is the only thing that actually installed on my machine, and that doesn't help since my LUX is not supported by it, and thus the device is not recognized by it. I am still able to type, however.
At the end of compiling using the quickinstall script, I always get an error displayed in the Terminal, and a ckb.app _IS_ made, but it's "damaged of incomplete". 

<details>
<summary>I will paste what the errors are that I am getting:</summary>
In file included from main.cpp:1:
In file included from ./mainwindow.h:4:
In file included from ../../../../Qt5.6.2/5.6/clang_64/lib/QtGui.framework/Headers/QCloseEvent:1:
In file included from ../../../../Qt5.6.2/5.6/clang_64/lib/QtGui.framework/Headers/qevent.h:37:
In file included from /Users/<username removed>/Qt5.6.2/5.6/clang_64/lib/QtGui.framework/Headers/qwindowdefs.h:37:
In file included from /Users/<username removed>/Qt5.6.2/5.6/clang_64/lib/QtCore.framework/Headers/qglobal.h:1110:
/Users/<username removed>/Qt5.6.2/5.6/clang_64/lib/QtCore.framework/Headers/qatomic.h:155:5: error: constexpr constructor never produces a constant expression [-Winvalid-constexpr]
    QAtomicInt(int value = 0) Q_DECL_NOTHROW : QAtomicInteger<int>(value) {}
    ^
/Users/<username removed>/Qt5.6.2/5.6/clang_64/lib/QtCore.framework/Headers/qatomic.h:155:48: note: non-literal type 'QAtomicInteger<int>' cannot be used in a constant expression
    QAtomicInt(int value = 0) Q_DECL_NOTHROW : QAtomicInteger<int>(value) {}
                                               ^
In file included from mainwindow.cpp:1:
In file included from ./ckbsettings.h:4:
In file included from ../../../../Qt5.6.2/5.6/clang_64/lib/QtCore.framework/Headers/QSettings:1:
In file included from ../../../../Qt5.6.2/5.6/clang_64/lib/QtCore.framework/Headers/qsettings.h:37:
In file included from /Users/<username removed>/Qt5.6.2/5.6/clang_64/lib/QtCore.framework/Headers/qobject.h:40:
In file included from /Users/<username removed>/Qt5.6.2/5.6/clang_64/lib/QtCore.framework/Headers/qobjectdefs.h:41:
In file included from /Users/<username removed>/Qt5.6.2/5.6/clang_64/lib/QtCore.framework/Headers/qnamespace.h:37:
In file included from /Users/<username removed>/Qt5.6.2/5.6/clang_64/lib/QtCore.framework/Headers/qglobal.h:1110:
/Users/<username removed>/Qt5.6.2/5.6/clang_64/lib/QtCore.framework/Headers/qatomic.h:155:5: error: constexpr constructor never produces a constant expression [-Winvalid-constexpr]
    QAtomicInt(int value = 0) Q_DECL_NOTHROW : QAtomicInteger<int>(value) {}
    ^
/Users/<username removed>/Qt5.6.2/5.6/clang_64/lib/QtCore.framework/Headers/qatomic.h:155:48: note: non-literal type 'QAtomicInteger<int>' cannot be used in a constant expression
    QAtomicInt(int value = 0) Q_DECL_NOTHROW : QAtomicInteger<int>(value) {}
                                               ^
In file included from main.cpp:1:
In file included from ./mainwindow.h:9:
In file included from ./kbwidget.h:7:
In file included from ./kb.h:7:
In file included from ./kbprofile.h:9:
In file included from ./kbmode.h:6:
In file included from ./kbbind.h:10:
./keyaction.h:164:44: warning: unused parameter 'rhs' [-Wunused-parameter]
    inline void operator=(const KeyAction& rhs) {}
                                           ^
./keyaction.h:165:39: warning: unused parameter 'rhs' [-Wunused-parameter]
    inline KeyAction(const KeyAction& rhs) : QObject() {}
                                      ^
2 warnings and 1 error generated.
In file included from kbwidget.cpp:2:
In file included from ../../../../Qt5.6.2/5.6/clang_64/lib/QtWidgets.framework/Headers/QFileDialog:1:
In file included from ../../../../Qt5.6.2/5.6/clang_64/lib/QtWidgets.framework/Headers/qfiledialog.h:37:
In file included from /Users/<username removed>/Qt5.6.2/5.6/clang_64/lib/QtCore.framework/Headers/qdir.h:37:
In file included from /Users/<username removed>/Qt5.6.2/5.6/clang_64/lib/QtCore.framework/Headers/qstring.h:41:
In file included from /Users/<username removed>/Qt5.6.2/5.6/clang_64/lib/QtCore.framework/Headers/qchar.h:37:
In file included from /Users/<username removed>/Qt5.6.2/5.6/clang_64/lib/QtCore.framework/Headers/qglobal.h:1110:
/Users/<username removed>/Qt5.6.2/5.6/clang_64/lib/QtCore.framework/Headers/qatomic.h:155:5: error: constexpr constructor never produces a constant expression [-Winvalid-constexpr]
    QAtomicInt(int value = 0) Q_DECL_NOTHROW : QAtomicInteger<int>(value) {}
    ^
/Users/<username removed>/Qt5.6.2/5.6/clang_64/lib/QtCore.framework/Headers/qatomic.h:155:48: note: non-literal type 'QAtomicInteger<int>' cannot be used in a constant expression
    QAtomicInt(int value = 0) Q_DECL_NOTHROW : QAtomicInteger<int>(value) {}
                                               ^
In file included from mainwindow.cpp:2:
In file included from ./kbmanager.h:8:
In file included from ./kb.h:7:
In file included from ./kbprofile.h:9:
In file included from ./kbmode.h:6:
In file included from ./kbbind.h:10:
./keyaction.h:164:44: warning: unused parameter 'rhs' [-Wunused-parameter]
    inline void operator=(const KeyAction& rhs) {}
                                           ^
./keyaction.h:165:39: warning: unused parameter 'rhs' [-Wunused-parameter]
    inline KeyAction(const KeyAction& rhs) : QObject() {}
                                      ^
make[2]: *** [release/main.o] Error 1
make[2]: *** Waiting for unfinished jobs....
In file included from colorbutton.cpp:1:
In file included from ./colorbutton.h:4:
In file included from ../../../../Qt5.6.2/5.6/clang_64/lib/QtWidgets.framework/Headers/QPushButton:1:
In file included from ../../../../Qt5.6.2/5.6/clang_64/lib/QtWidgets.framework/Headers/qpushbutton.h:37:
In file included from /Users/<username removed>/Qt5.6.2/5.6/clang_64/lib/QtWidgets.framework/Headers/qabstractbutton.h:37:
In file included from /Users/<username removed>/Qt5.6.2/5.6/clang_64/lib/QtGui.framework/Headers/qicon.h:37:
In file included from /Users/<username removed>/Qt5.6.2/5.6/clang_64/lib/QtCore.framework/Headers/qglobal.h:1110:
/Users/<username removed>/Qt5.6.2/5.6/clang_64/lib/QtCore.framework/Headers/qatomic.h:155:5: error: constexpr constructor never produces a constant expression [-Winvalid-constexpr]
    QAtomicInt(int value = 0) Q_DECL_NOTHROW : QAtomicInteger<int>(value) {}
    ^
/Users/<username removed>/Qt5.6.2/5.6/clang_64/lib/QtCore.framework/Headers/qatomic.h:155:48: note: non-literal type 'QAtomicInteger<int>' cannot be used in a constant expression
    QAtomicInt(int value = 0) Q_DECL_NOTHROW : QAtomicInteger<int>(value) {}
                                               ^
In file included from kbwidget.cpp:7:
In file included from ./fwupgradedialog.h:7:
In file included from ./kb.h:7:
In file included from ./kbprofile.h:9:
In file included from ./kbmode.h:6:
In file included from ./kbbind.h:10:
./keyaction.h:164:44: warning: unused parameter 'rhs' [-Wunused-parameter]
    inline void operator=(const KeyAction& rhs) {}
                                           ^
./keyaction.h:165:39: warning: unused parameter 'rhs' [-Wunused-parameter]
    inline KeyAction(const KeyAction& rhs) : QObject() {}
                                      ^
1 error generated.
mainwindow.cpp:257:64: warning: unused parameter 'reason' [-Wunused-parameter]
void MainWindow::iconClicked(QSystemTrayIcon::ActivationReason reason){
                                                               ^
make[2]: *** [release/colorbutton.o] Error 1
3 warnings and 1 error generated.
kbwidget.cpp:168:91: warning: unused parameter 'previous' [-Wunused-parameter]
void KbWidget::on_modesList_currentItemChanged(QListWidgetItem *current, QListWidgetItem *previous){
                                                                                          ^
make[2]: *** [release/mainwindow.o] Error 1
3 warnings and 1 error generated.
make[2]: *** [release/kbwidget.o] Error 1
make[1]: *** [release] Error 2
make: *** [sub-src-ckb-make_first] Error 2
</details>

ANY help at all is appreciated. Thank you.

mattanger, 2017-02-05:
Hi @shibatan I think is an issue with your version of clang. Can you make sure that you have the most recent version of Xcode installed and that any old versions of Xcode are completely uninstalled. This: https://forum.qt.io/topic/74118/compilation-error-when-building-shared-library-under-mac-os/7 may be relevant to your situation.

light2yellow, 2017-02-05:
What **exactly** do you get in the terminal while running `./quickinstall`? What branch are you compiling from?

max-l-weaver, 2017-02-05:
I got this too. Running OSX 10.12.3 with the latest xcode and qt. This was specifically when trying to run the build file from the downloaded source code (downloaded as zip). I ended up just cloning the repo, running the quickinstall and viola, it worked!

shibatan, 2017-02-05:
@mattanger I do not have the most recent of Xcode installed as it requires OSX 10.12 and I am on 10.10.5. I believe that my version of Xcode is the last one available before OSX 10.11 became the requirement. I shall check on these things and do some research and then report back.
Thank you so much for replying. 

@max-l-weaver Cloning the repo? I'm new to this community and coding as whole, so forgive me. What exactly worked for you?

shibatan, 2017-02-06:
IT WORKS!
Things that I did: 
1. Modified the file in Qt found at _ /users/user/Qt/5.6/clang_64/mkspecs/qdevice.pri _ and added the line: !host_build:QMAKE_MAC_SDK = macosx
as perhttp://stackoverflow.com/questions/26320677/error-could-not-resolve-sdk-path-for-macosx10-8 
2. Downloaded Xcode 6.4 from here: https://developer.apple.com/download/more/
3. Clone the repo (as per @max-l-weaver 's suggestion) and running the quickinstall through Terminal.

mattanger, 2017-02-06:
Fantastic. Glad that you got it working!

closed 2017-02-06 by mattanger

clodal, 2017-02-07:
Had this issue an hour ago. Used @shibatan's solution and it fixed it.

Here are some notes that might be helpful.
1. qdevice.pri wasn't in the location specified, I had to search within the mkspecs folder to get it
2. Xcode 6.4 is not required. I used Xcode 8.2.1 and it work fine.
3. Command to run quickinstall through the terminal can be found in the repo's [BUILD.md](https://github.com/mattanger/ckb-next/blob/master/BUILD.md)

shibatan, 2017-02-07:
@clodal Glad my that struggle helped someone else! I used 6.4 because of my version of OSX, as I was previously using 5 and was getting the errors. Since I made all three changes at once, I don't know how much my version of Xcode mattered, but I had to mention that.

light2yellow, 2017-05-06:
I had the very same issue few days ago, choosing the SDK in XCode helps, this is now documented in the README under `macOS Installation`.