mattanger/ckb-next #66: Provide releases pre-built along with release notes

linjer, 2017-02-06:
It helps with adoption, upgrading, and transparency when we have built releases ready for download along with release notes. 

I've been able to download and build manually on my local, but it wasn't entirely clear what version we're on and there could be potential environment issues with QT, depending on user.

mattanger, 2017-02-06:
Hi @linjer ckb-next is currently not stable and it is under active development. Once it is stable, we plan to make a release.

light2yellow, 2017-02-06:
@linjer yes, this was taken into account and both Linux and macOS versions will ship with README and BUILD as a documentation. Furthermore, README restructure is planned because it is
* a bit outdated
* too long and needs to be separated

As for ETA for a new release: as always in FOSS, when it's ready. When it's ready? When we fix some critical bugs that won't make into the next release in any way. What are those bugs? We'll create a milestone and specifically mark those.

closed 2017-04-25 by tatokis