mattanger/ckb-next #67: On waking from sleep: "no devices connected" (OSX)

shibatan, 2017-02-07:
K70 RGB LUX, Mac OS 10.10.5
Upon waking from sleep, the keyboard lights shine white rather than the set RGB configuration, and ckb does not recognize the device.
Upon restarting, the issue was fixed, but it does continue to happen every time the computer is woken from sleep with the keyboard connected.

batchpert, 2017-02-07:
I had a similar issue two weeks ago, but its solved now. When the correct lights come on can you type on it? #35 was my issue number and a simple reboot of my computer has seemed to fix the problems i was getting

shibatan, 2017-02-08:
I can type on it the whole time, regardless of whether or not the lights are correct. 
Like I said, rebooting is a temporary fix for me, but I felt like I should bring it to attention.

clodal, 2017-02-08:
Was just about to open this issue until i saw this. Thanks @shibatan. Same here lost all my ckb settings after waking from sleep on a macOS 10.12.3. Even after a reboot, i can't get the settings back.

light2yellow, 2017-05-03:
The issue with a vanished profile might be related to #83.

Regarding lost connection between the keyboard and the daemon upon wake-up - cannot really say anything about it, as I have never used different "sleep" features. Being a Linux user I was taught the hard way that using any sleeping features will almost always lead to the most exotic problems popping out left and right, so I had to deal with it and now make do without any. Maybe somebody else can share his experience on the healthiness of sleeping features and the impact on the daemon supplemented with logs.

frickler24, 2017-05-03:
@shibatan If you like, take the PR (it is just one line added) and try on your mac.
Please post the result, because we do not know the path on your system for the QSettings directory.

light2yellow, 2017-05-03:
Well, I'm almost sure that the config file on macOS is also under `~/.config/ckb/ckb.conf`.

light2yellow, 2017-05-06:
I was wrong. Just ran the app on macOS and got
```
Path  to settings is "/Users/username/Library/Preferences/com.ckb.ckb.plist"
```

luceat-lux-vestra, 2017-05-19:
It happens when connect to USB Hub in my case. So, I solved this problem by connect directly to motherboard's USB port.

I hope this will help.

tatokis, 2018-01-21:
https://github.com/ckb-next/ckb-next/issues/5

closed 2018-01-21 by tatokis