mattanger/ckb-next #68: Scimitar Pro RGB and K95 RGB Platinum support? (Ubuntu 16.10)

Empyrian, 2017-02-07:
Hello,

I've just got these 2 devices, and wanted to use them but neither of them are showing up in the CKB interface.
Keyboard lights up and works, haven't tried the media buttons and what not yet.
Mice is completely dead except for the optical red light.

Both devices has been updated (today) to their latest firmware and works on Windows.

tatokis, 2017-02-07:
We do not have support for the K95 RGB Platinum yet. #53

I am not sure regarding the mouse, could you please run the ckb-dev-detect script and attach the output?

Empyrian, 2017-02-07:
Hello,

Sure thing.
[ckb-dev-detect-report.gz](https://github.com/mattanger/ckb-next/files/758966/ckb-dev-detect-report.gz)

tatokis, 2017-02-07:
@Empyrian regarding the keyboard, unfortunately you will have to wait.

The interesting thing is that the mouse isn't working without ckb-next.
It seems to be recognised and added as an input device according to dmesg.

I have added the USB ID for your mouse on the testing branch.

Please try compiling ckb-next from the testing branch and report back on if it is recognised or not.
However, please unplug the K95 RGB Platinum as it will interfere (and possibly leave it in an unconfigured state, meaning the OS won't be able to receive input from it).

You can download the code that's in testing from here https://github.com/mattanger/ckb-next/archive/testing.zip

After testing this, in order to restore keyboard functionality you will have to close the ckb-next GUI and then stop the daemon. ```sudo systemctl stop ckb-daemon```
and to prevent it from starting again: ```sudo systemctl disable ckb-daemon```

Empyrian, 2017-02-07:
Hello,

I cloned the testing branch and installed it, the mouse started to light up and got recognised.

DamnWidget, 2017-02-17:
If the mouse is the Scimitar Pro RGB it does not works on Linux as it returns a non able to parse remote descriptor and the `usbhid` driver just fails to initialize it with 
```
hid-generic 0003:1B1C:1B3E.000X: item 0 1 0 8 parsing failed
hid-generic: probe of 0003:1B1C:1B3E.000X7 failed with error -22
```

Fortunately I wrote a driver to fix it, unfortunately it did not make it into the Kernel, if you know how, you can patch your kernel yourself and just rebuild the `hid` sub system, it will create a new `hid_corsair_mouse.ko` module for you.

The patches can be found here: https://github.com/DamnWidget/kernel/tree/master/drivers/hid

If you use Arch you can just follow this instructions: https://github.com/DamnWidget/archlinux-packages/tree/master/core/linux

If you use Manjaro, you can clone my `core-packages` fork in https://github.com/DamnWidget/packages-core`, then checkout the `linux49.4.9.7-1-Corsair` branch cd into `linux49` and run `makepkg -sci` in there, it will probably take a while.

tatokis, 2017-02-17:
If I am not mistaken, as I don't have a corsair mouse, the ckb daemon directly talks to the mouse and creates a virtual device, completely overriding usbhid. 

If this does not occur, then we might need to modify ckb-next to be able to read the mouse properly.

mattanger, 2017-02-17:
That is what it does. The mouse **should** get claimed by the driver if the id is listed, but after briefly looking at the patches @DamnWidgets there appear to be some issues with the mouse and how it reports.

DamnWidget, 2017-02-17:
Well, not really overriding the `usbhid`. Ckb is an user space driver that uses the kernel services provided by the `uhid` transport level driver (that also uses the `usbhid` driver). What is happening is that Ckb is querying the device firmware endpoints directly so it probably gets loaded as a generic usb mouse by the `uhid` driver and it registers a new device into the uhid core.

The mouse will respond but its real capabilities are hidden for the underlying HID system as it's remote descriptor is unknown.

tatokis, 2017-02-17:
RGB colour did work at least, so we should probably rename this thread and keep it exclusively to track scimitar pro progress.

tatokis, 2017-02-17:
@DamnWidget Considering the devices still work after setting the 0x4 usbhid quirk, I am almost sure we are overriding at least the part of usbhid that handles input.
```
T:  Bus=08 Lev=01 Prnt=01 Port=01 Cnt=01 Dev#=  2 Spd=12  MxCh= 0
D:  Ver= 2.00 Cls=00(>ifc ) Sub=00 Prot=00 MxPS=64 #Cfgs=  1
P:  Vendor=1b1c ProdID=1b13 Rev=02.05
S:  Manufacturer=Corsair
S:  Product=Corsair K70 RGB Gaming Keyboard 
S:  SerialNumber=0803400BAE3B94065371B971F5001947
C:  #Ifs= 3 Cfg#= 1 Atr=a0 MxPwr=500mA
I:  If#= 0 Alt= 0 #EPs= 1 Cls=03(HID  ) Sub=01 Prot=01 Driver=usbfs
I:  If#= 1 Alt= 0 #EPs= 1 Cls=03(HID  ) Sub=00 Prot=00 Driver=usbfs
I:  If#= 2 Alt= 0 #EPs= 1 Cls=03(HID  ) Sub=00 Prot=00 Driver=usbfs
```

Unless I have it all wrong, in which case please correct me.

DamnWidget, 2017-02-17:
The `HID_QUIRK_IGNORE` hides a device preventing it from being claimed, but you are claming it from the user space. In the other hand the `usbfs` driver uses `usbcore` directly so probably it will not uses any service from `usbhid` but I am not sure.

In any case the report descriptor of the device remains unknown as the usbcore driver is not able to parse it neither.

DamnWidget, 2017-02-17:
By the way, I just updated to the last commit in testing and this is what is working:
Left Button
Right Button
Middle Button
Wheel Up
Wheel Down
Button 4  (DPI/RGB switch)
Button 5 (DPI/RGB switch)

None of the lateral buttons work at all. With my driver trough usbhid everything works.

tatokis, 2017-02-17:
@DamnWidget Did you by any chance have any issues compiling from testing?

DamnWidget, 2017-02-17:
Everything compiles smoothly

Empyrian, 2017-02-17:
@DamnWidget Yeah, the mouse worked since @tatokis added the dev id.

DamnWidget, 2017-02-17:
The only thing that I noticed is that using ckb I can not go to the red mode {snipper) while with mine you can.

tatokis, 2017-02-17:
Okay, I am very confused right now.
Please explain exactly which part of the mouse doesn't work when using ckb-next.

@DamnWidget that's most likely because that feature isn't implemented in ckb-next (as it switches the mouse to software mode).

DamnWidget, 2017-02-17:
Looks like CKB defines other configurations for the DPI in the Performance tab
[image]
In hardware mode the ranges goes from yellow, green, blue, white, red with CKB there is 6 modes as you can see in the screen but for some reason the red one can not be selected (probably  is fine, with hardware you can select 5 DPI profiles but I don't understand why the last one can not be selected).

By the way, de max DPI for this device is 16000 DPI

Uthule, 2017-03-23:
Thank you very much, this is working for me in Opensuse. :)

I wonder when this will appear in the testing branch? Just used the link tatokis provided.

I have an issue tho, I also have the keyboard K65 (which I don't need it to be handled by ckb-next) whenever I am running the program itself (doesn't happen by only having on the daemon) if I move the keyboard a bit, just putting it back on the table for example, forces a system logout... Only happens if I have ckb-next running. 

Is there a way to stop ckb-next for handling a given decide? Otherwise be aware of this bug. I suppose I need to open a new issue for it.

Uthule, 2017-03-23:
After some searching for the K65 logout issue, and not finding anything, I ordered a new USB cable for the keyboard. Still it's odd that only happens when ckb-next is running, so far no issues just with the daemon on.

tatokis, 2017-04-02:
@Uthule I have absolutely no idea what could be causing. Maybe strace can help?
I'd suggest opening a new issue with any of your findings.

Regarding the Scimitar Pro, it seems to be in a functional state. If you experience any further issues, please open a new issue with exact details of what needs to be modified.

Regarding the K95 Platinum RGB, more info in #53

closed 2017-04-02 by tatokis