mattanger/ckb-next #69: where to download pkg

wdh1221cn, 2017-02-10:
I looked up every on the site, but I still could not find any link to down the pkg. Plz some one tell me where to find a pkg...

btw, I m using ckb beta-v0.2.6 which can not find my key board(k65 lux RGB). I hope I can find the right answer here.

gbl

mattanger, 2017-02-10:
Hi, @wdh1221cn right now there are no binary packages available. We haven't reached a point where we're ready to release any packages. If you look at  https://github.com/mattanger/ckb-next/blob/master/README.md you can find instructions to install ckb-next manually. The contents section of the README will point you in the right direction for your particular operating system.

light2yellow, 2017-02-10:
@wdh1221cn see https://github.com/mattanger/ckb-next/blob/master/README.md#current-status

wdh1221cn, 2017-02-10:
Oh... you guys are awesome, those replies are damn quick.
tks all, I will try to install it manually, cheers.

downloading qt5, starting now.......

can't stop saying that you are awesome.

closed 2017-02-10 by light2yellow