mattanger/ckb-next #70: App Crash when Try to choose a Color

baensch, 2017-02-10:
Hello Togehter,
i have an issue when i try to choose an color for the back light. i marked it at the screenshot.
System:
Core i7 6700k
Asus Board
GTX 970
Arch linux 64 bit
ckb-git version from the aur
[image]

Best regards
baensch

tatokis, 2017-02-10:
Have a look at https://github.com/mattanger/ckb-next#linux (where it references GLib errors)

baensch, 2017-02-10:
i cant find config files from qt. i only find some config files from the qt creator.

tatokis, 2017-02-12:
Please close the GUI, open a terminal, run ```ckb``` and then  click the change color button.
After that, upload the output from the terminal, so we can see what went wrong.

Nik-U, 2017-02-15:
I also experience this issue on Debian unstable with a K95 RGB building from master. My log is attached. After clicking the "Change color..." button, the application freezes but does not crash. The problem persists even when I delete ~/.config/ckb/ckb.conf. I don't see any files with "qt" in the name in my home directory. I have Qt version 5.7.1 installed. 

[ckb.txt](https://github.com/mattanger/ckb-next/files/775836/ckb.txt)

baensch, 2017-02-15:
Sorry for my Late reply but here is my terminal output:
` nils  ~  ckb
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile
Scanning  "/opt/ckb/ckb-animations/ckb-gradient"
Scanning  "/opt/ckb/ckb-animations/ckb-pinwheel"
Scanning  "/opt/ckb/ckb-animations/ckb-rain"
Scanning  "/opt/ckb/ckb-animations/ckb-random"
Scanning  "/opt/ckb/ckb-animations/ckb-ripple"
Scanning  "/opt/ckb/ckb-animations/ckb-wave"
Starting  "/opt/ckb/ckb-animations/ckb-gradient"
Downloaded new firmware list. 6 entries found.

(ckb:1559): GLib-GObject-WARNING **: cannot register existing type 'GtkWidget'
(ckb:1559): GLib-GObject-WARNING **: cannot add class private field to invalid type '<invalid>'
(ckb:1559): GLib-GObject-WARNING **: cannot add private field to invalid (non-instantiatable) type '<invalid>'
(ckb:1559): GLib-GObject-CRITICAL **: g_type_add_interface_static: assertion 'G_TYPE_IS_INSTANTIATABLE (instance_type)' failed
(ckb:1559): GLib-GObject-WARNING **: cannot register existing type 'GtkBuildable'
(ckb:1559): GLib-GObject-CRITICAL **: g_type_interface_add_prerequisite: assertion 'G_TYPE_IS_INTERFACE (interface_type)' failed
(ckb:1559): GLib-CRITICAL **: g_once_init_leave: assertion 'result != 0' failed
(ckb:1559): GLib-GObject-CRITICAL **: g_type_add_interface_static: assertion 'G_TYPE_IS_INSTANTIATABLE (instance_type)' failed
(ckb:1559): GLib-GObject-CRITICAL **: g_type_register_static: assertion 'parent_type > 0' failed
(ckb:1559): GLib-GObject-WARNING **: cannot add private field to invalid (non-instantiatable) type '<invalid>'
`

fleischie, 2017-02-16:
I suppose this is a known issue with Gtk/Qt toolkits. Check the following issues:

- https://github.com/ccMSC/ckb/issues/461#issuecomment-259568681
- https://github.com/mattanger/ckb-next/issues/51

baensch, 2017-02-17:
Hey Thanks for the Tips but i tested all of theese things and i have still the same error.

tatokis, 2017-02-17:
Perharps try this https://github.com/mati75/qt5ct ?

light2yellow, 2017-02-17:
@baensch okay, let's start from the beginning.

First error message?
* `GLib-GObject-WARNING **: cannot register existing type 'GtkWidget'`

First link in the Google Search?
* https://bbs.archlinux.org/viewtopic.php?id=214147&p=2

This is specifically an Arch Linux thread, moreover, it is marked as SOLVED. Read through it thoroughly and find your solution.

Eventually, this is a Qt issue, ckb-next has nothing to do with it.

light2yellow, 2017-03-09:
Closing this as unrelated directly to ckb-next, resolved on other threads and forums and missing any response from the author.

closed 2017-03-09 by light2yellow