mattanger/ckb-next #73: K70 LUX RGB won't connect to MacOS even after doing everything said to do

jack380, 2017-02-11:
Just got this keyboard knowing I would have to download the software, while downloading quick install script, my mac never prompted me to install system wide and CKB still shows no devices connected. I installed Xcode and QT. Thanks for the help guys.

light2yellow, 2017-02-11:
Instructions, did you follow them?

> The easiest way to build the driver is with the `quickinstall` script, which is present in the ckb-master folder. Double-click on `quickinstall` and it will compile the app for you, then ask if you'd like to install it system-wide. If the build fails for any reason, or if you'd like to compile manually, see BUILD.md.

You can try opening a terminal inside the ckb-next folder where you cloned the source and launch the script with `./quickinstall`, then posting the output of the script. Also you can always use instructions in `BUILD.md`.

xuanhung1017, 2017-02-12:
I cant find LaunchDaemons in Library

light2yellow, 2017-02-12:
@xuanhung1017 are you sure you used `master` branch and not `testing` or `cmake` or whatever?
A short recap of the console commands you must use in order to install this:
```zsh
git clone https://github.com/mattanger/ckb-next.git --single-branch ckb-master
cd ckb-master
./quickinstall
```
If this fails for some reason, provide the output of those from your terminal.

xuanhung1017, 2017-02-12:
It's working now. Thanks

jack380, 2017-02-12:
Hey i uninstalled everything and am trying to do the quick install script, do I run it in terminal or qt?

light2yellow, 2017-02-12:
@jack380 sorry, I closed the issue but *you* was the original author. Qt is just the library, install it from the website and forget about it (just for now). Commands above must me executed in the terminal, one at a time. If you don't know how to use the terminal, you're free to use the Google search.

jack380, 2017-02-12:
@light2yellow thanks for the help but when I run quick install for ckb master in terminal i get this : exit;
Johns-MacBook-Pro:~ johnmarshall$ /Users/johnmarshall/Desktop/quickinstall ; exit;
Command line tools are already installed.
There is nothing to do.

Press enter to exit.

light2yellow, 2017-02-12:
@jack380 
1. What made you append `; exit;`?
2. You write

> my mac never prompted me to install system wide

Is that still true? That's not the expected behaviour.
3. Why don't you try three commands above? To make sure they will work do them *outside* of the currently downloaded ckb-next source. For example:
```
cd ~/Documents
git clone https://github.com/mattanger/ckb-next.git --single-branch ckb-master
cd ckb-master
./quickinstall
cd
rm -rf ~/Documents/ckb-master
```

jack380, 2017-02-12:
When you say try three commands above what do you mean? when running the example command you gave I got this : Johns-MacBook-Pro:~ johnmarshall$ cd ~/Documents
Johns-MacBook-Pro:Documents johnmarshall$ git clone https://github.com/mattanger/ckb-next.git --single-branch ckb-master
Cloning into 'ckb-master'...
remote: Counting objects: 4948, done.
remote: Compressing objects: 100% (12/12), done.
remote: Total 4948 (delta 10), reused 9 (delta 8), pack-reused 4928
Receiving objects: 100% (4948/4948), 2.29 MiB | 3.91 MiB/s, done.
Resolving deltas: 100% (3504/3504), done.
Johns-MacBook-Pro:Documents johnmarshall$ cd ckb-master
Johns-MacBook-Pro:ckb-master johnmarshall$ ./quickinstall
Command line tools are already installed.
Preparing build files...
Cannot find feature spec_pre.prf
Error processing project file: ./ckb.pro

Press enter to exit.
Johns-MacBook-Pro:ckb-master johnmarshall$ rm -rf ~/Documents/ckb-master

light2yellow, 2017-02-12:
@jack380 

> Cannot find feature spec_pre.prf

I'm afraid I will not be able to help you. This is most probably not a ckb-next issue. I used the Google search and there's no solution for this issue. Nor can I reproduce it on 10.12.3. All you can do is to:
* either reinstall Qt5
* or try Homebrew's Qt5 package
or do both. Or you can actually follow the instructions in `BUILD.md`, which is your best bet.
You should also think about you PATH variable, whether you modified it or whether you [accidentally] modified some Qt libraries manually.

light2yellow, 2017-02-17:
@jack380 
As an example, here's another issue: https://github.com/nylas/nylas-mail/issues/3278
The guy there eventually resolved everything, but the issue had nothing to do with the app itself. His setup just got broken somehow. Your case is very similar by its nature.

Did you try anything from above?

tatokis, 2017-04-03:
It's been 2 months now with no response.
Closing due to inactivity.

closed 2017-04-03 by tatokis