mattanger/ckb-next #74: import/export

dkuida, 2017-02-12:
I consider creating the support for that, but before I start wanted to make sure if there is a magor blocker for that to work ? besides the tie needed I mean

light2yellow, 2017-02-12:
As @frickler24 already said in ccMSC/ckb#574, the devs will appreciate any contribution as long as it's maintainable.

> if there is a magor blocker for that to work

I'm not able to answer this, maybe someone else could.

frickler24, 2017-02-14:
@dkuida I do not understand what you ment with import/export. Can you please explain your approach?

fleischie, 2017-02-16:
@frickler24 I guess he means to write an import/export procedure for color/animation profiles.

@dkuida If you can provide a PR, that is based off of the latest master branch, that works as expected and ideally on mac and linux as well, then go ahead. :+1:

light2yellow, 2017-02-16:
@fleischie I think it'd much better if @dkuida bases his PR against testing branch. This is written in the Readme and is generally a more healthy strategy because new features can not play well with the other new features.

fleischie, 2017-02-16:
@light2yellow Yes, you may be right. I guess I was still under the impression, that the testing branch was not up to date as much. Anyway: Edited to say it's the testing branch and not the master branch. :smiley: :+1:

Extarys, 2017-09-09:
Is there any word about this?
Also I cannot seem to find where the files are stored to import and export manually config files :D

tatokis, 2018-01-22:
https://github.com/ckb-next/ckb-next/issues/6

closed 2018-01-22 by tatokis