mattanger/ckb-next #75: How to install?

TommieBoyNL, 2017-02-14:


TommieBoyNL, 2017-02-14:
I'm sorry if i don't know how to install it, but please someone help??

light2yellow, 2017-02-14:
Did you read the `README.md`?

TommieBoyNL, 2017-02-14:
yes i wrote the readme.md. I installed qt and ran the quickinstall file. When i opened it terminal said:
Project ERROR: Could not resolve SDK Path for 'macosx.' Then i tried it to just run ckb.pro in qt creator self and it said: error while building/deploying project ckb (kit: Desktop Qt 5.8.0 clang 64bit)
What do i do now?

light2yellow, 2017-02-14:
> yes i *wrote* the readme.md.

I guess you wanted to write

> yes i *read* the readme.md.

Now, on the topic.

Why didn't you provide any information about the error in your first message? We, random supporters of this project, neither have any magical abilities to guess what has happened, nor want we to pry every word out of you. We can't help you if you don't want to help yourself. From the error occurred I suppose you are using macOS. We need to know the version of the macOS. We need to know what branch were you using while installing ckb-next. Even better - provide a version of ckb-next (using `git describe`). If Qt is involved, provide the version of Qt (done by Qt Creator this time, but one can use `qmake -v`) and the way Qt was installed (for example, with the official binary installer from Qt website or using a package manager, i. e. Homebrew).

Did you try searching the issue with the Google Search or between "Issues" in this project? Let's suppose you *did* search for `Could not resolve SDK Path for 'macosx.'` with the help of the former. What were your results?

Now consider searching within the latter. There's at least one very similar issue already solved here. What were your results after trying the steps mentioned in that issue? Did you follow the links mentioned in that issue and tried other solutions?

People supporting open-source projects spend their time and energy trying to resolve other people's problems. For free. And this is absolutely fine. But they cannot be helpful and, what's more important, productive without other person being helpful as well and willing to resolve his *own* issue. Which makes me feel sad and confused.

Anyway, an issue template will be created as soon as possible to prevent such kind of "issues" in the future.

syropian, 2017-03-20:
Curious, why isn't a binary currently provided? Wouldn't that reduce the chance that someone is incorrectly compiling it?

closed 2017-04-03 by tatokis