mattanger/ckb-next #76: I cant find the download ckb.pkg.

bryanchoi1994, 2017-02-14:
i dont know what should i download.
which one should i download please help me

chicks-net, 2017-02-14:
This project is still being actively developed.  Building distribution packages is planned after development stabilizes.

bryanchoi1994, 2017-02-14:
so should i use the old one ? ccMSC that one ? 
but my K65 RAPIDFIRE cannot be detected in that one 
I dont know how to fix it

chicks-net, 2017-02-14:
See https://github.com/mattanger/ckb-next#current-status  It is currently a manual process.

ericmalloff, 2017-02-16:
So what is the latest version of the build?  It looks as if 0.2.4 was last uploaded in Feb 2016 has this project been abandoned? 

I followed the instructions and installed release 0.2.4 (https://github.com/mattanger/ckb-next/releases) but the driver still does not seem to be working. I am not even sure if it replaced the old CKB driver I had installed. 

Looking forward to getting this up and running.

mattanger, 2017-02-16:
The release versions are not accurate. There are no update to date binaries
currently. Please follow the manual install instructions in the `readme.md`.
You can see the current status here: https://github.com/mattanger/
ckb-next#current-status

mattanger, 2017-02-17:
Again, please take a look at the `readme.md` for more information on how to install manually. If any issues building are encountered, please open a new issue.

closed 2017-02-17 by mattanger