mattanger/ckb-next #78: Mouse sensitivity after uninstall

ArvidAAA, 2017-02-17:
My mouse sensitivity is really weird now even after uninstall. Moving the sensitivity slider in Sierra is doing like nothing and the mouse can't like even move anymore. Any help on this would be appreciated.
Thanks

light2yellow, 2017-02-17:
Save this as `uninstall.sh`:
```sh
#!/usr/bin/env bash
echo "Stopping ckb-next app"
/Applications/ckb-next.app/Contents/MacOS/ckb-next --close >/dev/null 2>&1
/Applications/ckb.app/Contents/MacOS/ckb --close >/dev/null 2>&1
echo "Removing ckb-next app"
sudo rm -rf /Applications/ckb-next.app >/dev/null 2>&1
sudo rm -rf /Applications/ckb.app >/dev/null 2>&1
echo "Stopping daemon"
sudo launchctl unload /Library/LaunchDaemons/ckb-next-daemon.plist >/dev/null 2>&1
sudo launchctl unload /Library/LaunchDaemons/com.ckb.daemon.plist >/dev/null 2>&1
echo "Removing daemon's plist"
sudo rm -rf /Library/LaunchDaemons/ckb-next-daemon.plist >/dev/null 2>&1
sudo rm -rf /Library/LaunchDaemons/com.ckb.daemon.plist >/dev/null 2>&1
```
Then execute in the Terminal (one command at a time):
```sh
chmod +x uninstall.sh
sudo ./uninstall.sh
```
This removes any trace of ckb/ckb-next if there is any.
Try to move your mouse around again.

ArvidAAA, 2017-02-18:
Did not work I'm reinstalling Mac OS :)))))) Good program 👎

light2yellow, 2017-02-18:
MacOS is not Windows, you don't have to reinstall it every third day. If you have any further problems regarding ckb-next after the reinstall, open another issue.

closed 2017-02-18 by light2yellow

tatokis, 2017-02-18:
@ArvidAAA I am sorry to hear that, but thank you for taking the time to help make this "Good program 👎" even better.
I am sure if anyone in the future has this issue, they will appreciate it.

I hope you have to deal with software development and bug trackers in your life.
Best wishes.