mattanger/ckb-next #79: make debug broken on Linux

MaxHaydenChiz, 2017-02-18:
There was a request for a GDB backtrace on another issue, so I tried to build a debug version to provide the information. Unfortunately, with the latest source (as of now), `make debug` appears broken. I get the following error:
make[1]: Entering directory '/home/mchiz/Downloads/ckb-next-master/src/ckb-ripple'
make[1]: *** No rule to make target 'debug'.  Stop.
make[1]: Leaving directory '/home/mchiz/Downloads/ckb-next-master/src/ckb-ripple'
Makefile:393: recipe for target 'sub-src-ckb-ripple-debug' failed
make: *** [sub-src-ckb-ripple-debug] Error 2

tatokis, 2017-02-18:
I am almost positive I fixed this issue in the testing branch.
Please try compiling ckb-next from there.

MaxHaydenChiz, 2017-02-18:
It works with the testing branch. Thanks.

closed 2017-02-18 by MaxHaydenChiz