mattanger/ckb-next #82: Segfaults

light2yellow, 2017-02-19:
While trying to test the Linux part on Arch Linux of my CMake branch [cmake-bustling](https://github.com/light2yellow/ckb-next) I'm stuck with segfaults. Related logs are [published as gist](https://gist.github.com/light2yellow/93790d98eca69b6619e613a56c967247).
If you would like to investigate, checkout that branch, run 
```bash
mkdir build && cd build
cmake .. && make -j$(nproc --all)
sudo make install
```
To uninstall:
```bash
sudo ../linux/uninstall.sh
```
Current `testing` here builds and works as expected.

frickler24, 2017-02-19:
@light2yellow I believe it is not the malloc itself, but it is the destination
`urbs[i].buffer = malloc(urbs[i].buffer_length);`
If you have a keyboard which is not detected correctly (e.g. not enough endpoints), then you get this error.
pls have a look at pullreq #64 where this sigsegv was the cause also.

light2yellow, 2017-02-19:
Thanks for the hint. Apparently, I was using `0x4` quirks, will retry compiling that the other day, maybe that was the issue (although qmake-based project did work correctly). If it won't compile, I'll try with the patch from #64 applied.

light2yellow, 2017-02-26:
So far I cannot replicate this on Ubuntu 16.10 with nothing being changed.

light2yellow, 2017-02-26:
Fixed by #89, which has nothing to do with the way which building system is used. Which means the segfault was magically omitted in qmake.

`ldd` on Arch Linux:
```
	linux-vdso.so.1 (0x00007fffee922000)
	libudev.so.1 => /usr/lib/libudev.so.1 (0x00007f22f7507000)
	libpthread.so.0 => /usr/lib/libpthread.so.0 (0x00007f22f711f000)
	libc.so.6 => /usr/lib/libc.so.6 (0x00007f22f6d81000)
	libresolv.so.2 => /usr/lib/libresolv.so.2 (0x00007f22f6b6a000)
	libcap.so.2 => /usr/lib/libcap.so.2 (0x00007f22f6966000)
	librt.so.1 => /usr/lib/librt.so.1 (0x00007f22f675e000)
	libm.so.6 => /usr/lib/libm.so.6 (0x00007f22f645a000)
	/lib64/ld-linux-x86-64.so.2 (0x00007f22f733c000)
```

`ldd` on Ubuntu 16.10:
```
	linux-vdso.so.1 =>  (0x00007ffe39398000)
	libudev.so.1 => /lib/x86_64-linux-gnu/libudev.so.1 (0x00007f9652209000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007f9651feb000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f9651c24000)
	librt.so.1 => /lib/x86_64-linux-gnu/librt.so.1 (0x00007f9651a1c000)
	/lib64/ld-linux-x86-64.so.2 (0x00005629cdf77000)
```

closed 2017-02-26 by light2yellow