mattanger/ckb-next #83: Missing default keyboard profiles

tbcrawford, 2017-02-23:
My color profiles were available last time I was logged into my computer. This morning when I turned my computer on, the "Default ckb" profile was missing from ckb-next. I was wondering where these profiles get stored so that I may be able to look into this myself some if anyone has this knowledge. I am on macOS.

I also wanted to report this as a possible issue of the profile(s) not being found properly.

frickler24, 2017-02-23:
@tbcrawford Would it be an idea to put this change as a new CLI parameter ("--restore-std-profiles")? 
Do you want to implement this change completely (incl help, documentation, README.md-update, ...) as a pullrequest?
What do our maintainers think about it?

tatokis, 2017-02-23:
From what I can see, commenting the code above will cause ckb-next to reload the demo profile from the resources and write it to disk on every launch, meaning it will overwrite any local modifications to it.
I am afraid this is not a solution.
I also don't think adding a CLI parameter to restore the profile is a good idea.

Instead, we need to find out the root cause for the profile corruption, fix it, and then once the import/export system is implemented, we can just provide the default one as a download.

tbcrawford, 2017-02-23:
I think having some sort of "restore standard profiles" ability would be nice to have. Probably at first in the form of CLI, and then maybe it could be transferred to the GUI as a simple button. I am not as familiar with the code here as I would like to be to implement all of that myself at the moment, and I don't believe I have the time to get that invested as I'm working on my final semester of grad school. I may poke around at it and see what I can understand in the future however.

light2yellow, 2017-02-23:
I will support @tatokis and Eric Raymond:
> The net benefit of having anything that can be called a software development methodology is in inverse proportion to the quality of your developers – and possibly inverse-squared.

We demonstrate even more incompetence than we have now by creating more crutches. Don't take it personally.

And yes, the button "Unfuck plz" is a good idea. Very good idea.

frickler24, 2017-02-27:
@tbcrawford any idea, how you lost your profile information?
If not, we should close this issue.

tbcrawford, 2017-02-27:
@frickler24 I wish I had some clue, but I do not. I just recall this happening before for some reason too. You may close this.

light2yellow, 2017-04-04:
This

frickler24, 2017-04-08:
I'm wondering that all three issues are related to macOS. Is there a special treatment for the mechnism used to handle the profile info? It is a very simple qt mechanism, which does an abstraction on linux conf-files, windows registry and a mac function (I do not know the last one).  So are there known bugs for this function?

light2yellow, 2017-05-03:
Here's a fixed bug from Qt tracker: [QSetting setValue("", float) does not work for all Qt apps running on iOS 10 and macOS Sierra](https://bugreports.qt.io/browse/QTBUG-56124) and here's another one being marked as duplicate, reported a bit later: [QSettings serialization problems on Mac](https://bugreports.qt.io/browse/QTBUG-56565). Please, take a look. It might be related to our problem. It was fixed (or at least the developers say so) in `5.7.1` and `5.8`.

frickler24, 2017-05-03:
Maybe #174 can help to gather more info?

closed 2018-01-21 by tatokis