mattanger/ckb-next #84: Doxygen documentation

frickler24, 2017-02-25:
Beginning at #58 I did some commenting to the linux usb implementation. 
#58 is very long and a special issue, so I opened this one for docu discussion about the usb interface.

The documents collected here are generated by doxygen.

Because doxygen takes all files in the dircetory of ckb-daemon and I have just in some of the files manually added comments, I opened a second doxygen-documentation.
In this second part only files with usb concerning comments are isolated.

Both docus - each in html or pdf format - [can be found here](https://doxygen.frickler24.de/index.html). If our ckb-next webpage has launched, it is only one directory tree to copy and the documents are available there.

fleischie, 2017-03-01:
Hey @frickler24, if I was to document the macOS-specific portions of the daemon, where would I start?

I have never used doxygen and am not familiar with the syntax, which may be easily fixed by reading the appropriate documentation. Do you have a branch with the comments already commited, so I could expand your work? Alternatively should I only document the macOS-portions and send you a patch/incomplete docs/etc.?

I may have some time this week. :smile:

frickler24, 2017-03-01:
Hey @fleischie That would be great!
You can find my work in [testing+usb-related-commenting Branch](https://github.com/frickler24/ckb-next/tree/testing%2Busb-related-commenting)
Here you can find a good [manual](http://www.stack.nl/~dimitri/doxygen/manual/).
The installation of doxygen is easy, but not neccessary. If you just want to add some comments (especially in usb_macos.c and maybe in the common usb.[ch]), then you can send me your patches or sace them in a separate branch.
If you want to see what you have done, then install the doxygen scripts (and laTex if you want to generate PDF) and use the Doxygen config file or JustUSBfiles file in src/ckb-daemon of my branch mentioned above.

closed 2018-01-21 by tatokis