mattanger/ckb-next #92: Error 2 when trying to use ./quickinstall

Zyrophr, 2017-02-26:
Hi all ! 
I'm currently under Kali, and i can't use my STRAFE RGB, it does nothing when i plug it in :(
So i came here because i thought i would find a way to use my only keyboard ! 
I tried hard, really, to make it work... and i can't. I don't know what to do, can you please help me ? :) 

here is what happens when i enter ./quickinstall : 

> `*********@********:~/Téléchargements/ckb-next-master$ ./quickinstall 
> Preparing build files...
> Compiling binaries...
> (This can take a while, please be patient)
> cd src/ckb-daemon/ && ( test -e Makefile || /usr/lib/x86_64-linux-gnu/qt5/bin/qmake /home/zyrophr/Téléchargements/ckb-next-master/src/ckb-daemon/ckb-daemon.pro -o Makefile ) && make -f Makefile 
> cd src/ckb/ && ( test -e Makefile || /usr/lib/x86_64-linux-gnu/qt5/bin/qmake /home/zyrophr/Téléchargements/ckb-next-master/src/ckb/ckb.pro -o Makefile ) && make -f Makefile 
> cd src/ckb-wave/ && ( test -e Makefile || /usr/lib/x86_64-linux-gnu/qt5/bin/qmake /home/zyrophr/Téléchargements/ckb-next-master/src/ckb-wave/ckb-wave.pro -o Makefile ) && make -f Makefile 
> cd src/ckb-ripple/ && ( test -e Makefile || /usr/lib/x86_64-linux-gnu/qt5/bin/qmake /home/zyrophr/Téléchargements/ckb-next-master/src/ckb-ripple/ckb-ripple.pro -o Makefile ) && make -f Makefile 
> make[1] : on entre dans le répertoire « /home/zyrophr/Téléchargements/ckb-next-master/src/ckb-daemon »
> make -f Makefile.Debug
> make[1] : on entre dans le répertoire « /home/zyrophr/Téléchargements/ckb-next-master/src/ckb-wave »
> gcc -c -m64 -pipe -std=c99 -O2 -fPIC  -I. -I/usr/lib/x86_64-linux-gnu/qt5/mkspecs/linux-g++-64 -o main.o main.c
> make[1] : on entre dans le répertoire « /home/zyrophr/Téléchargements/ckb-next-master/src/ckb-ripple »
> gcc -c -m64 -pipe -std=c99 -O2 -fPIC  -I. -I/usr/lib/x86_64-linux-gnu/qt5/mkspecs/linux-g++-64 -o main.o main.c
> make[2] : on entre dans le répertoire « /home/zyrophr/Téléchargements/ckb-next-master/src/ckb-daemon »
> gcc -c -std=gnu99 -Wno-unused-parameter -Werror=all -g -fPIC -DCKB_VERSION_STR=\"`cat /home/zyrophr/Téléchargements/ckb-next-master/src/ckb-daemon/../../VERSION | tr -d \n`\" -I. -I/usr/lib/x86_64-linux-gnu/qt5/mkspecs/linux-g++-64 -o debug/device.o device.c
> make[1] : on entre dans le répertoire « /home/zyrophr/Téléchargements/ckb-next-master/src/ckb »
> make -f Makefile.Release
> make[2] : on entre dans le répertoire « /home/zyrophr/Téléchargements/ckb-next-master/src/ckb »
> make[2] : on quitte le répertoire « /home/zyrophr/Téléchargements/ckb-next-master/src/ckb »
> Makefile:38 : la recette pour la cible « release » a échouée
> make[1] : on quitte le répertoire « /home/zyrophr/Téléchargements/ckb-next-master/src/ckb »
> Makefile:75 : la recette pour la cible « sub-src-ckb-make_first » a échouée
> gcc -c -std=gnu99 -Wno-unused-parameter -Werror=all -g -fPIC -DCKB_VERSION_STR=\"`cat /home/zyrophr/Téléchargements/ckb-next-master/src/ckb-daemon/../../VERSION | tr -d \n`\" -I. -I/usr/lib/x86_64-linux-gnu/qt5/mkspecs/linux-g++-64 -o debug/devnode.o devnode.c
> gcc -c -std=gnu99 -Wno-unused-parameter -Werror=all -g -fPIC -DCKB_VERSION_STR=\"`cat /home/zyrophr/Téléchargements/ckb-next-master/src/ckb-daemon/../../VERSION | tr -d \n`\" -I. -I/usr/lib/x86_64-linux-gnu/qt5/mkspecs/linux-g++-64 -o debug/input_linux.o input_linux.c
> gcc -c -std=gnu99 -Wno-unused-parameter -Werror=all -g -fPIC -DCKB_VERSION_STR=\"`cat /home/zyrophr/Téléchargements/ckb-next-master/src/ckb-daemon/../../VERSION | tr -d \n`\" -I. -I/usr/lib/x86_64-linux-gnu/qt5/mkspecs/linux-g++-64 -o debug/input_mac.o input_mac.c
> gcc -c -std=gnu99 -Wno-unused-parameter -Werror=all -g -fPIC -DCKB_VERSION_STR=\"`cat /home/zyrophr/Téléchargements/ckb-next-master/src/ckb-daemon/../../VERSION | tr -d \n`\" -I. -I/usr/lib/x86_64-linux-gnu/qt5/mkspecs/linux-g++-64 -o debug/input.o input.c
> gcc -c -std=gnu99 -Wno-unused-parameter -Werror=all -g -fPIC -DCKB_VERSION_STR=\"`cat /home/zyrophr/Téléchargements/ckb-next-master/src/ckb-daemon/../../VERSION | tr -d \n`\" -I. -I/usr/lib/x86_64-linux-gnu/qt5/mkspecs/linux-g++-64 -o debug/main.o main.c
> gcc -c -std=gnu99 -Wno-unused-parameter -Werror=all -g -fPIC -DCKB_VERSION_STR=\"`cat /home/zyrophr/Téléchargements/ckb-next-master/src/ckb-daemon/../../VERSION | tr -d \n`\" -I. -I/usr/lib/x86_64-linux-gnu/qt5/mkspecs/linux-g++-64 -o debug/notify.o notify.c
> g++ -m64 -Wl,-O1 -o ../../bin/ckb-animations/ckb-wave main.o    
> gcc -c -std=gnu99 -Wno-unused-parameter -Werror=all -g -fPIC -DCKB_VERSION_STR=\"`cat /home/zyrophr/Téléchargements/ckb-next-master/src/ckb-daemon/../../VERSION | tr -d \n`\" -I. -I/usr/lib/x86_64-linux-gnu/qt5/mkspecs/linux-g++-64 -o debug/usb_linux.o usb_linux.c
> g++ -m64 -Wl,-O1 -o ../../bin/ckb-animations/ckb-ripple main.o    
> gcc -c -std=gnu99 -Wno-unused-parameter -Werror=all -g -fPIC -DCKB_VERSION_STR=\"`cat /home/zyrophr/Téléchargements/ckb-next-master/src/ckb-daemon/../../VERSION | tr -d \n`\" -I. -I/usr/lib/x86_64-linux-gnu/qt5/mkspecs/linux-g++-64 -o debug/usb_mac.o usb_mac.c
> make[1] : on quitte le répertoire « /home/zyrophr/Téléchargements/ckb-next-master/src/ckb-wave »
> gcc -c -std=gnu99 -Wno-unused-parameter -Werror=all -g -fPIC -DCKB_VERSION_STR=\"`cat /home/zyrophr/Téléchargements/ckb-next-master/src/ckb-daemon/../../VERSION | tr -d \n`\" -I. -I/usr/lib/x86_64-linux-gnu/qt5/mkspecs/linux-g++-64 -o debug/usb.o usb.c
> gcc -c -std=gnu99 -Wno-unused-parameter -Werror=all -g -fPIC -DCKB_VERSION_STR=\"`cat /home/zyrophr/Téléchargements/ckb-next-master/src/ckb-daemon/../../VERSION | tr -d \n`\" -I. -I/usr/lib/x86_64-linux-gnu/qt5/mkspecs/linux-g++-64 -o debug/firmware.o firmware.c
> make[1] : on quitte le répertoire « /home/zyrophr/Téléchargements/ckb-next-master/src/ckb-ripple »
> gcc -c -std=gnu99 -Wno-unused-parameter -Werror=all -g -fPIC -DCKB_VERSION_STR=\"`cat /home/zyrophr/Téléchargements/ckb-next-master/src/ckb-daemon/../../VERSION | tr -d \n`\" -I. -I/usr/lib/x86_64-linux-gnu/qt5/mkspecs/linux-g++-64 -o debug/profile.o profile.c
> gcc -c -std=gnu99 -Wno-unused-parameter -Werror=all -g -fPIC -DCKB_VERSION_STR=\"`cat /home/zyrophr/Téléchargements/ckb-next-master/src/ckb-daemon/../../VERSION | tr -d \n`\" -I. -I/usr/lib/x86_64-linux-gnu/qt5/mkspecs/linux-g++-64 -o debug/extra_mac.o extra_mac.c
> gcc -c -std=gnu99 -Wno-unused-parameter -Werror=all -g -fPIC -DCKB_VERSION_STR=\"`cat /home/zyrophr/Téléchargements/ckb-next-master/src/ckb-daemon/../../VERSION | tr -d \n`\" -I. -I/usr/lib/x86_64-linux-gnu/qt5/mkspecs/linux-g++-64 -o debug/keymap.o keymap.c
> gcc -c -std=gnu99 -Wno-unused-parameter -Werror=all -g -fPIC -DCKB_VERSION_STR=\"`cat /home/zyrophr/Téléchargements/ckb-next-master/src/ckb-daemon/../../VERSION | tr -d \n`\" -I. -I/usr/lib/x86_64-linux-gnu/qt5/mkspecs/linux-g++-64 -o debug/command.o command.c
> gcc -c -std=gnu99 -Wno-unused-parameter -Werror=all -g -fPIC -DCKB_VERSION_STR=\"`cat /home/zyrophr/Téléchargements/ckb-next-master/src/ckb-daemon/../../VERSION | tr -d \n`\" -I. -I/usr/lib/x86_64-linux-gnu/qt5/mkspecs/linux-g++-64 -o debug/device_vtable.o device_vtable.c
> gcc -c -std=gnu99 -Wno-unused-parameter -Werror=all -g -fPIC -DCKB_VERSION_STR=\"`cat /home/zyrophr/Téléchargements/ckb-next-master/src/ckb-daemon/../../VERSION | tr -d \n`\" -I. -I/usr/lib/x86_64-linux-gnu/qt5/mkspecs/linux-g++-64 -o debug/device_keyboard.o device_keyboard.c
> gcc -c -std=gnu99 -Wno-unused-parameter -Werror=all -g -fPIC -DCKB_VERSION_STR=\"`cat /home/zyrophr/Téléchargements/ckb-next-master/src/ckb-daemon/../../VERSION | tr -d \n`\" -I. -I/usr/lib/x86_64-linux-gnu/qt5/mkspecs/linux-g++-64 -o debug/device_mouse.o device_mouse.c
> gcc -c -std=gnu99 -Wno-unused-parameter -Werror=all -g -fPIC -DCKB_VERSION_STR=\"`cat /home/zyrophr/Téléchargements/ckb-next-master/src/ckb-daemon/../../VERSION | tr -d \n`\" -I. -I/usr/lib/x86_64-linux-gnu/qt5/mkspecs/linux-g++-64 -o debug/led_keyboard.o led_keyboard.c
> gcc -c -std=gnu99 -Wno-unused-parameter -Werror=all -g -fPIC -DCKB_VERSION_STR=\"`cat /home/zyrophr/Téléchargements/ckb-next-master/src/ckb-daemon/../../VERSION | tr -d \n`\" -I. -I/usr/lib/x86_64-linux-gnu/qt5/mkspecs/linux-g++-64 -o debug/led.o led.c
> gcc -c -std=gnu99 -Wno-unused-parameter -Werror=all -g -fPIC -DCKB_VERSION_STR=\"`cat /home/zyrophr/Téléchargements/ckb-next-master/src/ckb-daemon/../../VERSION | tr -d \n`\" -I. -I/usr/lib/x86_64-linux-gnu/qt5/mkspecs/linux-g++-64 -o debug/led_mouse.o led_mouse.c
[... cut ...]
> make[1] : on quitte le répertoire « /home/zyrophr/Téléchargements/ckb-next-master/src/ckb-daemon »
> make[2]: ***  Aucune règle pour fabriquer la cible « /usr/lib/x86_64-linux-gnu/qt5/bin/uic », nécessaire pour « ui_mainwindow.h ». Arrêt.
> make[1]: *** [release] Erreur 2
> make: *** [sub-src-ckb-make_first] Erreur 2
> make: *** Attente des tâches non terminées....
> 
> Press enter to exit.
> `

Sorry, part of the text is in french :( I'm stuck, and i really need my keyboard to work !

Thanks for taking the time,
Z

tatokis, 2017-02-26:
Just a blind guess, but your downloads folder contains special 
characters.
Try moving ckb-next-master to your home folder and running quickinstall 
there.

If it doesn't help, run it like this LANG=C ./quickinstall in order to 
show the errors in English.

light2yellow, 2017-02-26:
Please upload `src/ckb/Makefile` here. Do this as a file upload. If you have `Makefile.Release` and/or `Makefile.Debug`, upload them as well.

light2yellow, 2017-02-26:
Also, give us the output of
```
qmake --version
```

Zyrophr, 2017-02-26:
`QMake version 2.01a
Using Qt version 4.8.7 in /usr/lib/x86_64-linux-gnu
`

[Makefile.Debug.gz](https://github.com/mattanger/ckb-next/files/802299/Makefile.Debug.gz)
[Makefile.gz](https://github.com/mattanger/ckb-next/files/802300/Makefile.gz)
[Makefile.Release.gz](https://github.com/mattanger/ckb-next/files/802301/Makefile.Release.gz)

I'll try to run it in my home folder :)

light2yellow, 2017-02-26:
@Zyrophr did you read the installation instructions?

Zyrophr, 2017-02-26:
@light2yellow Yup. Even the BUILD.md :/

light2yellow, 2017-02-26:
@Zyrophr and you are sure you did the "Preparation" step?

Zyrophr, 2017-02-26:
@light2yellow I tried to install everything needed (Qt5 , libudev, zlib, gcc, g++, and glibc) and it seemed to work. Every time i was doing the apt-get install "package". I'll check right now...

Zyrophr, 2017-02-26:
@light2yellow yeah, just checked, i've everything... :/

light2yellow, 2017-02-26:
@Zyrophr what does ` qtchooser -list-versions` give?

Zyrophr, 2017-02-26:
@light2yellow 
```

> 4
> 5
> default
> qt4-x86_64-linux-gnu
> qt4
> qt5-x86_64-linux-gnu
> qt5

```

light2yellow, 2017-02-26:
Run `QT_SELECT=5 ./quickinstall`

Zyrophr, 2017-02-26:
@light2yellow same :(

```

> QT_SELECT=5 LANG=C ./quickinstall
> Preparing build files...
> Compiling binaries...
> (This can take a while, please be patient)
> cd src/ckb-daemon/ && ( test -e Makefile || /usr/lib/x86_64-linux-gnu/qt5/bin/qmake /home/zyrophr/Téléchargements/ckb-next-master/src/ckb-daemon/ckb-daemon.pro -o Makefile ) && make -f Makefile 
> cd src/ckb/ && ( test -e Makefile || /usr/lib/x86_64-linux-gnu/qt5/bin/qmake /home/zyrophr/Téléchargements/ckb-next-master/src/ckb/ckb.pro -o Makefile ) && make -f Makefile 
> cd src/ckb-wave/ && ( test -e Makefile || /usr/lib/x86_64-linux-gnu/qt5/bin/qmake /home/zyrophr/Téléchargements/ckb-next-master/src/ckb-wave/ckb-wave.pro -o Makefile ) && make -f Makefile 
> make[1]: Entering directory '/home/zyrophr/Téléchargements/ckb-next-master/src/ckb-wave'
> gcc -c -m64 -pipe -std=c99 -O2 -fPIC  -I. -I/usr/lib/x86_64-linux-gnu/qt5/mkspecs/linux-g++-64 -o main.o main.c
> make[1]: Entering directory '/home/zyrophr/Téléchargements/ckb-next-master/src/ckb-daemon'
> make -f Makefile.Debug
> make[2]: Entering directory '/home/zyrophr/Téléchargements/ckb-next-master/src/ckb-daemon'
> gcc -c -std=gnu99 -Wno-unused-parameter -Werror=all -g -fPIC -DCKB_VERSION_STR=\"`cat /home/zyrophr/Téléchargements/ckb-next-master/src/ckb-daemon/../../VERSION | tr -d \n`\" -I. -I/usr/lib/x86_64-linux-gnu/qt5/mkspecs/linux-g++-64 -o debug/device.o device.c
> cd src/ckb-ripple/ && ( test -e Makefile || /usr/lib/x86_64-linux-gnu/qt5/bin/qmake /home/zyrophr/Téléchargements/ckb-next-master/src/ckb-ripple/ckb-ripple.pro -o Makefile ) && make -f Makefile 
> make[1]: Entering directory '/home/zyrophr/Téléchargements/ckb-next-master/src/ckb'
> make -f Makefile.Release
> make[2]: Entering directory '/home/zyrophr/Téléchargements/ckb-next-master/src/ckb'
> make[2]: Leaving directory '/home/zyrophr/Téléchargements/ckb-next-master/src/ckb'
> Makefile:38: recipe for target 'release' failed
> make[1]: Leaving directory '/home/zyrophr/Téléchargements/ckb-next-master/src/ckb'
> Makefile:75: recipe for target 'sub-src-ckb-make_first' failed
> gcc -c -std=gnu99 -Wno-unused-parameter -Werror=all -g -fPIC -DCKB_VERSION_STR=\"`cat /home/zyrophr/Téléchargements/ckb-next-master/src/ckb-daemon/../../VERSION | tr -d \n`\" -I. -I/usr/lib/x86_64-linux-gnu/qt5/mkspecs/linux-g++-64 -o debug/devnode.o devnode.c
> make[1]: Entering directory '/home/zyrophr/Téléchargements/ckb-next-master/src/ckb-ripple'
> gcc -c -m64 -pipe -std=c99 -O2 -fPIC  -I. -I/usr/lib/x86_64-linux-gnu/qt5/mkspecs/linux-g++-64 -o main.o main.c
> gcc -c -std=gnu99 -Wno-unused-parameter -Werror=all -g -fPIC -DCKB_VERSION_STR=\"`cat /home/zyrophr/Téléchargements/ckb-next-master/src/ckb-daemon/../../VERSION | tr -d \n`\" -I. -I/usr/lib/x86_64-linux-gnu/qt5/mkspecs/linux-g++-64 -o debug/input_linux.o input_linux.c
> gcc -c -std=gnu99 -Wno-unused-parameter -Werror=all -g -fPIC -DCKB_VERSION_STR=\"`cat /home/zyrophr/Téléchargements/ckb-next-master/src/ckb-daemon/../../VERSION | tr -d \n`\" -I. -I/usr/lib/x86_64-linux-gnu/qt5/mkspecs/linux-g++-64 -o debug/input_mac.o input_mac.c
> gcc -c -std=gnu99 -Wno-unused-parameter -Werror=all -g -fPIC -DCKB_VERSION_STR=\"`cat /home/zyrophr/Téléchargements/ckb-next-master/src/ckb-daemon/../../VERSION | tr -d \n`\" -I. -I/usr/lib/x86_64-linux-gnu/qt5/mkspecs/linux-g++-64 -o debug/input.o input.c
> g++ -m64 -Wl,-O1 -o ../../bin/ckb-animations/ckb-wave main.o    
> gcc -c -std=gnu99 -Wno-unused-parameter -Werror=all -g -fPIC -DCKB_VERSION_STR=\"`cat /home/zyrophr/Téléchargements/ckb-next-master/src/ckb-daemon/../../VERSION | tr -d \n`\" -I. -I/usr/lib/x86_64-linux-gnu/qt5/mkspecs/linux-g++-64 -o debug/main.o main.c
> gcc -c -std=gnu99 -Wno-unused-parameter -Werror=all -g -fPIC -DCKB_VERSION_STR=\"`cat /home/zyrophr/Téléchargements/ckb-next-master/src/ckb-daemon/../../VERSION | tr -d \n`\" -I. -I/usr/lib/x86_64-linux-gnu/qt5/mkspecs/linux-g++-64 -o debug/notify.o notify.c
> g++ -m64 -Wl,-O1 -o ../../bin/ckb-animations/ckb-ripple main.o    
> make[1]: Leaving directory '/home/zyrophr/Téléchargements/ckb-next-master/src/ckb-wave'
> gcc -c -std=gnu99 -Wno-unused-parameter -Werror=all -g -fPIC -DCKB_VERSION_STR=\"`cat /home/zyrophr/Téléchargements/ckb-next-master/src/ckb-daemon/../../VERSION | tr -d \n`\" -I. -I/usr/lib/x86_64-linux-gnu/qt5/mkspecs/linux-g++-64 -o debug/usb_linux.o usb_linux.c
> make[1]: Leaving directory '/home/zyrophr/Téléchargements/ckb-next-master/src/ckb-ripple'
> gcc -c -std=gnu99 -Wno-unused-parameter -Werror=all -g -fPIC -DCKB_VERSION_STR=\"`cat /home/zyrophr/Téléchargements/ckb-next-master/src/ckb-daemon/../../VERSION | tr -d \n`\" -I. -I/usr/lib/x86_64-linux-gnu/qt5/mkspecs/linux-g++-64 -o debug/usb_mac.o usb_mac.c
> gcc -c -std=gnu99 -Wno-unused-parameter -Werror=all -g -fPIC -DCKB_VERSION_STR=\"`cat /home/zyrophr/Téléchargements/ckb-next-master/src/ckb-daemon/../../VERSION | tr -d \n`\" -I. -I/usr/lib/x86_64-linux-gnu/qt5/mkspecs/linux-g++-64 -o debug/usb.o usb.c
> gcc -c -std=gnu99 -Wno-unused-parameter -Werror=all -g -fPIC -DCKB_VERSION_STR=\"`cat /home/zyrophr/Téléchargements/ckb-next-master/src/ckb-daemon/../../VERSION | tr -d \n`\" -I. -I/usr/lib/x86_64-linux-gnu/qt5/mkspecs/linux-g++-64 -o debug/firmware.o firmware.c
> gcc -c -std=gnu99 -Wno-unused-parameter -Werror=all -g -fPIC -DCKB_VERSION_STR=\"`cat /home/zyrophr/Téléchargements/ckb-next-master/src/ckb-daemon/../../VERSION | tr -d \n`\" -I. -I/usr/lib/x86_64-linux-gnu/qt5/mkspecs/linux-g++-64 -o debug/profile.o profile.c
[... cut ...] i.o debug/profile_mouse.o   -lpthread -ludev 
> make[2]: Leaving directory '/home/zyrophr/Téléchargements/ckb-next-master/src/ckb-daemon'
> make[1]: Leaving directory '/home/zyrophr/Téléchargements/ckb-next-master/src/ckb-daemon'
> make[2]: *** No rule to make target '/usr/lib/x86_64-linux-gnu/qt5/bin/uic', needed by 'ui_mainwindow.h'.  Stop.
> make[1]: *** [release] Error 2
> make: *** [sub-src-ckb-make_first] Error 2
> make: *** Waiting for unfinished jobs....
> 
> Press enter to exit.
> 

```

light2yellow, 2017-02-26:
@Zyrophr okay, try this:

`ln -s /etc/xdg/qtchooser/5.conf ~/.config/qtchooser/default.conf`

Then paste the output of `qmake --version` again. If it says `version 5.x.x`, try `./quickinstall`.

Zyrophr, 2017-02-26:
@light2yellow hmm, i think i made something wrong here...

```

> LANG=C sudo ln -s /etc/xdg/qtchooser/5.conf ~/.config/qtchooser/default.conf
> ln: failed to create symbolic link '/home/zyrophr/.config/qtchooser/default.conf': No such file or directory
> 

```

tatokis, 2017-02-26:
```mkdir -p ~/.config/qtchooser/``` and try again.
Also, don't use sudo.

Zyrophr, 2017-02-26:
@tatokis @light2yellow 

```
> QMake version 2.01a
> Using Qt version 4.8.7 in /usr/lib/x86_64-linux-gnu
```
also... I think i installed QT on Root, and i'm now on another session, could it came from that ? (i already had the error 2 on root)

light2yellow, 2017-02-26:
@Zyrophr this is definitely not the issue with `ckb-next`. I downloaded Kali Linux iso and launched it in VirtualBox:

[image]

As you can see, no issues. What I did was installing all the packages from the Ubuntu version of "Preparation" except libappindicator, cloning the repo and running the quickinstall. I did this from `root`, as you can expect on a fresh Live ISO.

light2yellow, 2017-02-26:
So, your best option is to do this from `root`, like the system proposes.

Zyrophr, 2017-02-26:
@light2yellow oh well, it worked using Ubuntu command... I love you <3 ! 
Thanks @tatokis and @light2yellow , i was getting mad ! And, "bravo" for this super app, works like a charm atm ! 
Sorry for taking your time :(

closed 2017-02-26 by Zyrophr